espressif/esp-hosted #3: esp-hosted not working with RPi3B+ & ESP-WROOVER-KIT

multigiorgiplex, 2020-06-29:
Hi all,

I'm having issues running the esp-hosted firmware via SDIO. The syntomphs are a bunch of SDIO initialization errors on the host with a lack of debug output on the esp. In [this archive](https://github.com/espressif/esp-hosted/files/4847854/esp-hosted.issue.zip) there are logs of the both sides.

I suppose that the issue resides on the ESP firmware, because the SDIO errors that the kernel spits out are triggered when I enable the ESP board, which is being keeped idle until that moment via pulling the EN pin to GND.
Nothing seems to change when I run the `rpi_init.sh` script on the host.

Yes I double-checked the connections.
I also tried to use the HCI via UART connections, and the result is similar, no output from the ESP `monitor` and timeout on the host side.

MangeshMalusare, 2020-06-30:
Hi,
Thanks for the feedback. This looks like a setup problem. Please share your setup image with us. We will take a close look at it.
Meanwhile, please try following and let us know if that helps you.

- As per our experience, most of the time such issues are caused due to jumper wires. Please make sure to use smaller (in length) and good quality jumper wires. 

- Please make sure to power ESP32 and Raspberry Pi separately with a power supply that provide sufficient power. ESP32 can be powered through PC using micro-USB cable. In case if you are using USB hub to power up, ESP32 and/or RPi, make sure that hub is externally powered.

Thanks

multigiorgiplex, 2020-06-30:
Thanks for the reply,

it was a setup problem, I removed the unreliable jumper cables by soldering wire traces on a perfboard.
[image]

Now the system works as expected.

Thanks for your time.

closed 2020-06-30 by multigiorgiplex