espressif/esp-hosted #4: No Rx from Raspberry Pi 3B+

htruong, 2020-07-12:
I have tripled-checked all the connections and made sure I used the shortest cables possible. This is my prototype board with a LOLIN D32 pro connected to a RPi3B+, and the ESP32 is powered from a hub -- [LOLIN32 Schematics](https://docs.wemos.cc/en/latest/_static/files/sch_d32_pro_v2.0.0.pdf) :

[image]

It looks like the kernel module is loaded successfully and I could see that the device is recognized successfully:

```
[  131.834466] esp32: loading out-of-tree module taints kernel.
[  131.839066] esp_probe: ESP network device detected
[  132.060134] Features supported are:
[  132.060145]   * WLAN
[  132.060150]   * BT/BLE
[  132.060155]     - HCI over SDIO
[  132.060159]     - BT/BLE dual mode
[  132.085714] read_packet: Failed to read data
[  132.279197] esp_sdio: probe of mmc1:0001:2 failed with error -22
```

Whenever I executed `ap_scan_list.py`, I got:

```
pi@experimentalpi:~/esp-hosted/host/linux/host_control $ python ap_scan_list.py 
Failed to get AP scan list
```

But I saw in the serial monitor that it did scan and tried to bring back the result for me:

```
data ready 
I (4239771) NETWORK_ADAPTER: serial_read_data

I (4239771) slave_commands: APSTA mode set in scan handler
I (4241371) slave_commands: Total APs scanned = 30
I (4241371) slave_commands: details of AP no 0
I (4241371) slave_commands: SSID                <REDACTED>
I (4241371) slave_commands: RSSI                -55
I (4241381) slave_commands: Channel             4
I (4241381) slave_commands: BSSID               <REDACTED>
I (4241391) slave_commands: Auth mode           4

...

I (4242231) slave_commands: details of AP no 29
I (4242231) slave_commands: SSID                <REDACTED>
I (4242241) slave_commands: RSSI                -95
I (4242241) slave_commands: Channel             11
I (4242251) slave_commands: BSSID               <REDACTED>
I (4242251) slave_commands: Auth mode           3

I (4242261) NETWORK_ADAPTER: serial_write_data 1487
```

But then the Pi's `dmesg` shows:

```
[  133.848011] Under-voltage detected! (0x00050005)
[  138.008242] Voltage normalised (0x00000000)
[  206.120861] esp_serial_open on device 0
[  206.120932] esp_serial_open on device 0
[  206.121030] Tx:02 d2 0f 00 08 00 00 00 01 07 00 63 6f 6e 74 72  ...........contr
[  206.121033] Tx:6f 6c 02 02 00 08 10                             ol.....
[  208.649340] read_packet: Failed to read data
```

So it looked like that I did get the TX across, but I never received anything from the ESP32 board. I wonder if you could kindly help in with what's wrong with my setup or how to debug the issue.

MangeshMalusare, 2020-07-13:
Hi, Please check SDIO pull up requirements from https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html.

We use ESP32 Wrover kit in our testing. Required pull ups are already provided with this kit. 
Since you are not using wrover kit, you need to setup pull ups as mentioned in above link.

Hope this helps.

Thanks.

htruong, 2020-07-13:
I tried that, it did not help entirely but I could see the command succeeding from time to time now (rarely). Thanks for the suggestions. I think I can resolve this soon -- it's just the reliability of this setup.

closed 2020-07-13 by htruong