espressif/esp-hosted #5: Bluetooth over UART not working with Pi Compute Module

kian99, 2020-07-14:
Hi,

I'm looking to setup the Bluetooth interface to a Rasperry Pi Compute Module 3+ via UART but I'm running into some transmission/reception issues. The Pi is on a Compute Module IO dev board and the ESP is an ESP32-WROOM-32. The ESP is externally powered. 

**Hardware**:
I've connected the TX,RX,CTS,RTS and GND lines as described in the [setup](https://github.com/espressif/esp-hosted/blob/master/docs/Setup.md) guide (Ensuring that Tx->RX and CTS->RTS etc.). My jumper wires are quite long but I was hoping since I am not using a high speed interface like SDIO that this wouldn't pose a problem.

**Software**
Because I am using the compute module there is no built in Bluetooth device so I've left out the 
`dtoverlay=disable-bt`
line from _/boot/config.txt_. The rest of the steps were followed and serial0 is available in /dev/ and works if I connect the Pi's RX and TX in a loopback and test with minicom using 
`minicom -D /dev/serial0`

The Pi is running Raspian Buster (Kernel 4.19.118-v7+) and has Bluez already installed but I have downloaded the source from www.kernel.org/pub/linux/bluetooth/bluez-5.50.tar.xz.
````
bluetoothctl -v
bluetoothctl: 5.50
````

Then for the driver configuration I've had a look inside the rpi_init.sh script and the GPIO pins for CTS and RTS are the same for the compute module so I don't think there is any problem there. An output of 
`sudo raspi-gpio get`
shows 
````
GPIO 14: level=1 fsel=4 alt=0 func=TXD0
GPIO 15: level=1 fsel=4 alt=0 func=RXD0
GPIO 16: level=1 fsel=7 alt=3 func=CTS0
GPIO 17: level=0 fsel=7 alt=3 func=RTS0
````

On running the script with the btuart parameter the dmesg log is 
````
[   61.920535] Bluetooth: Core ver 2.22
[   61.920603] NET: Registered protocol family 31
[   61.920607] Bluetooth: HCI device and connection manager initialized
[   61.920625] Bluetooth: HCI socket layer initialized
[   61.920636] Bluetooth: L2CAP socket layer initialized
[   61.920661] Bluetooth: SCO socket layer initialized
[   66.783412] esp32: loading out-of-tree module taints kernel.
[   66.784931] esp_probe: ESP network device detected
[   66.998140] Features supported are:
[   67.217193] esp_sdio: probe of mmc1:0001:2 failed with error -22
````
I'm not sure why nothing appears after supported features, is there any reason for that? Otherwise however it looks fine.
Then on the ESP's side, the log shows
````
I (547) cpu_start: Starting scheduler on PRO CPU.
I (550) NETWORK_ADAPTER: Supported features are:
I (550) NETWORK_ADAPTER: - WLAN
I (550) NETWORK_ADAPTER: - BT/BLE
I (560) NETWORK_ADAPTER:    - HCI Over SDIO
I (560) NETWORK_ADAPTER:    - BT/BLE dual mode
I (590) wifi:wifi driver task: 3ffd93f8, prio:23, stack:6656, core=0
I (590) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (590) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (610) wifi:wifi firmware version: ee95d06
I (610) wifi:config NVS flash: enabled
I (610) wifi:config nano formating: disabled
I (610) wifi:Init dynamic tx buffer num: 32
I (620) wifi:Init data frame dynamic rx buffer num: 32
I (620) wifi:Init management frame dynamic rx buffer num: 32
I (630) wifi:Init management short buffer num: 32
I (630) wifi:Init static rx buffer size: 1600
I (630) wifi:Init static rx buffer num: 10
I (640) wifi:Init dynamic rx buffer num: 32
I (730) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (730) wifi:mode : sta (c4:4f:33:0d:f7:79)
I (730) BTDM_INIT: BT controller compile version [c1cbe45]
I (730) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1240) NETWORK_ADAPTER: Initial set up done
I (66820) NETWORK_ADAPTER: Start Data Path
````
All seems good here.

Then running 
`sudo hciattach -s 115200 /dev/serial0 any 115200 flow`
It works and the output of _hciconfig_ is 
````
hci0:	Type: Primary  Bus: UART
	BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
	DOWN INIT RUNNING 
	RX bytes:0 acl:0 sco:0 events:0 errors:0
	TX bytes:8 acl:0 sco:0 commands:2 errors:0
````

but now checking dmesg again there are several command timeouts at the end of the output.
````
[   66.784931] esp_probe: ESP network device detected
[   66.998140] Features supported are:
[   67.217193] esp_sdio: probe of mmc1:0001:2 failed with error -22
[  208.688467] uart-pl011 3f201000.serial: no DMA platform data
[  208.717392] Bluetooth: HCI UART driver ver 2.3
[  208.717408] Bluetooth: HCI UART protocol H4 registered
[  208.717495] Bluetooth: HCI UART protocol Three-wire (H5) registered
[  208.717712] Bluetooth: HCI UART protocol Broadcom registered
[  209.008937] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[  209.008946] Bluetooth: BNEP filters: protocol multicast
[  209.008960] Bluetooth: BNEP socket layer initialized
[  210.796693] Bluetooth: hci0: command 0x1003 tx timeout
[  212.876690] Bluetooth: hci0: command 0x1001 tx timeout

````
And if I try to run the example gatt server in the Bluez source folder I get
````
pi@raspberrypi:~/BluetoothBluez/bluez-5.50/test $ sudo ./example-gatt-server
GattManager1 interface not found
````
I've tried some of the suggestions in [this link](https://stackoverflow.com/questions/16151360/use-bluez-stack-as-a-peripheral-advertiser) for correctly installing Bluez but I don't think that is the issue. Other commands like 
`sudo hciconfig hci0 up`
also result in more tx errors in dmesg.

Other things I've checked,
````
pi@raspberrypi:~ $ sudo rfkill list
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
````
and hcidump only shows 
````
$ hcidump -X -t
HCI sniffer - Bluetooth packet analyzer ver 5.50
device: hci0 snap_len: 1500 filter: 0xffffffff
````

I've checked the wiring a couple of times and tried swapping a few of the pins in case I swapped things around but no dice. I will get my hands on some shorter jumpers soon and test again. Thank you kindly for any help.

MangeshMalusare, 2020-07-15:
Hi Kian99,
Long jumper wires may create problem. Would it be possible for you to try short wires?

> I'm not sure why nothing appears after supported features, is there any reason for that? Otherwise however it looks fine.
It is okay if you do not see anything on loading kernel module. The kernel module is specifically for SDIO operations. 

Ensure that you are using same baud rate on host and esp32. To check and modify baud rate at esp32, do following:
1. cd esp/esp_driver/network_adapter
2. make menuconfig
3. navigate to 'Component config' -> 'Bluetooth' -> 'Bluetooth controller'
4. Ensure that HCI mode is 'UART(H4)'
5. Go to 'HCI UART(H4) Options' and update baud rate. 

Hope this help!

kian99, 2020-07-16:
Hi,

Checking 'Component config' -> 'Bluetooth' -> 'Bluetooth controller' I see it was set to VHCI rather than UART.
Changing it and checking the UART options I see that I can choose which UART peripheral to use and the baud rate but I can't choose UART0.

On the dev kit that I'm using only the UART0 is fully brought out to the header pins including the CTS0 and RTS0 pins while UART2 has only the Rx and Tx pins and UART1 is not available. Since I can't see any option to disable flow control I will assume that it needs the flow pins to work correctly. It seems likely this was the issue but since I don't have access to another dev kit with all the pins I won't be able to test. I was semi-successful in getting the SDIO to work but the jumpers were giving me intermittent connection problems so I think I will move the BT over there alongside the Wifi when I get things soldered down.

Thank you for the help however, I'm pretty confident that was the issue.

closed 2020-07-16 by kian99