espressif/esp-hosted #6: Where is the "pre-provided hosted mode firmware binary"?

ShonkyCH, 2020-08-13:
To try and avoid having to set up a build environment initially, where can I find the pre-compiled ESP32 binary mentioned in https://github.com/espressif/esp-hosted/blob/master/docs/Setup.md ?

Doesn't appear in the repo and I've not been able to find it. Where should I look please?

Thanks

ShonkyCH, 2020-08-18:
Ok, well I seem to have built a working version that appears to run. I've not got it working yet but suspect some interface or config issues.

Either way I think it would be could to include a pre-built binary somewhere. The documentation is a bit light there and has a a couple of confusing mistakes. e.g. it refers to running "git mv" on an ESP32.

MangeshMalusare, 2020-08-18:
Hi,
We will soon include pre-built binaries.

w.r.t kernel module compilation, we have provided a convenience script 'rpi_init.sh' which compiles and loads kernel module.
The script can be found in 'host/linux/host_control' directory.  Please refer 'Getting started' guide for more information.

Thanks,
Mangesh

ShonkyCH, 2020-08-18:
Ah, right you are. Thanks. Not mentioned on the hardware setup page which I was reading first and it jumps straight into it working with the module loaded.

It's a bit disjointed. Hardware Setup & Compilation talks about cloning the repo and getting the kernel headers and last step is, 'here's it working'. So you need to jump across to the Getting Started page in the middle to do the kernel module compile.

MangeshMalusare, 2020-08-18:
Thanks for your inputs! We will try to make it more organised and understandable in future versions.

MangeshMalusare, 2020-09-28:
Prebuilt binaries are available now.
https://github.com/espressif/esp-hosted/releases

MangeshMalusare, 2020-11-03:
Hi, can you please help close this issue?

ameyinamdar, 2020-11-03:
Closing the issue as pre-built binaries are now available.

closed 2020-11-03 by ameyinamdar