espressif/esp-hosted #7: mmc-bcm2835 3f300000.mmc: card claims to support voltages below defined range

nkhs, 2020-09-08:
[image]

nkhs, 2020-09-08:
I am using ESP32F-Kit, Raspberry Pi 3b

ameyinamdar, 2020-09-08:
@nkhs I think ESP32F-kit may not have required pull-ups on SDIO pins. Can you please check the schematic of ESP32F-Kit?

nkhs, 2020-09-08:
[ESP-32F开发板原理图.pdf](https://github.com/espressif/esp-hosted/files/5190376/ESP-32F.pdf)
This is my ESP32-F data sheet.

ameyinamdar, 2020-09-09:
@nkhs There are no pull-ups on the SDIO pins (https://github.com/espressif/esp-hosted/blob/master/docs/Setup.md#hardware-setupconnections). Also, it looks like some of these pins are actually used to connect to some other peripherals on the board. Can you please use ESP32-WROVER-Kit instead as described in the link specified here?

closed 2020-09-10 by nkhs