espressif/esp-hosted #8: ESP32 can not work in softAP mode

jianwei-bai, 2020-09-18:
1. When I let ESP32  work in softAP mode, Can't succeed. The log is as follows:  How can I solve this issues? Thanks!
Rspi:
[image]

pi@raspberrypi:~/esp-hosted/host/linux/host_control $ python softap_config.py 'xyz' 'xyz123456' 1 3 --max_conn=4 --ssid_hidden=0 --bw=1
SoftAP MAC Address 30:ae:a4:c9:f2:ad
Traceback (most recent call last):
  File "softap_config.py", line 61, in <module>
    softap_config = commands.wifi_set_softap_config(args.ssid, args.password, args.channel_id, args.encrp_mthd, args.max_conn, args.ssid_hidden, args.bw)
  File "/home/pi/esp-hosted/host/linux/host_control/host_commands/commands.py", line 180, in wifi_set_softap_config
    if ecn < esp_hosted_config_pb2.EncryptionMode.Type_WPA3_PSK:
AttributeError: 'module' object has no attribute 'EncryptionMode'

ESP32:
[image]

receive data from transport layer
receive data from transport layer
data ready 
receive data from transport layer
I (176440) slave_commands: get AP mac address
I (176450) slave_commands: mac [30:ae:a4:c9:f2:ad] 
data ready 
receive data from transport layer
I (676910) slave_commands: get AP mac address
I (676910) slave_commands: mac [30:ae:a4:c9:f2:ad]

jianwei-bai, 2020-09-18:
1.When I configure ESP32 in station mode.
python station_connect.py 'moduletest' 'test1234'   This command works. But adding WPA3 will fail. What is the problem and how can I solve it, thank you!
Raspi:
[image]

ESP32:
[image]

2.Sometimes all commands are disabled on RASPi and no data is received on ESP32.When I reset ESP32, it works.Is the connection between ESP32 and Raspi unstable, or is it something else? It happens all the time.Looking forward to your reply.
[image]
[image]

MangeshMalusare, 2020-09-18:
Thanks for reporting your issue. We will check both the issues and get back soon.

Meanwhile, can you please confirm if the AP in second issue support WPA3?