espressif/esp-hosted #9: Transport module not found

petersrob, 2020-09-18:
I followed all the instructions to flash the  wrover kit via SDIO, and also built the host driver ad received the expected result in dmesg. I then went to connect the wrover kit to my wifi, and received the following error. (I have received this error on test.py, station_connect.py, and ap_scan_list.py)

pi@raspberrypi:~/esp/esp-hosted/host/linux/host_control $ python station_connect.py 'test' '123abc'
Traceback (most recent call last):
  File "station_connect.py", line 15, in <module>
    from host_commands import commands
  File "/home/pi/esp/esp-hosted/host/linux/host_control/host_commands/commands.py", line 15, in <module>
    import transport
ModuleNotFoundError: No module named 'transport'

petersrob, 2020-09-20:
Update, I have moved the transport.py to the same directory, and now it says there is no module named utils.py. there is no file with that name in this repo.

MangeshMalusare, 2020-09-21:
Thanks for letting us know of your issue. This appears to some software setup issue. The scripts are expected to run as is. File movements are not required. We will check and get back.

MangeshMalusare, 2020-09-21:
Can you please confirm python version you are using? The current solution is compatible with python 2.7. In case if you are using python 3.0, please revert to v2.7. Python 3.0 support will be added in near future.

petersrob, 2020-09-22:
Here is everything I put in (from rpi_init.sh on, since board is already flashed), and am running python on 2.7.16. This is being run on a compute module 3+. 

[image]

MangeshMalusare, 2020-09-28:
Can you please confirm if all the steps from below link are followed while preparing your setup?
https://github.com/espressif/esp-hosted/blob/master/docs/Linux_based_host/Setup.md#python-setup

callum133, 2020-11-02:
Hi, 

I am getting the same error "No module named utils"