espressif/esp-hosted #10: GattManager1 interface not found when running Bluez example-gatt-server example

lph65724, 2020-09-28:
I followed the instructions, and the wifi is working perfectly.  However, when I run example-gatt-server, the terminal shows "GattManager1 interface not found".  it seems I have missed something.  The interface I use is SDIO.  thank you

MangeshMalusare, 2020-09-28:
Thanks for reporting this issue. Can you please share complete dmesg log of this issue?

lph65724, 2020-09-29:
Hi Mangesh,
here is my dmesg log and hciconfig command outputs

[   51.473747] Bluetooth: Core ver 2.22
[   51.473820] NET: Registered protocol family 31
[   51.473826] Bluetooth: HCI device and connection manager initialized
[   51.473845] Bluetooth: HCI socket layer initialized
[   51.473854] Bluetooth: L2CAP socket layer initialized
[   51.473874] Bluetooth: SCO socket layer initialized
[   75.440976] esp32_sdio: loading out-of-tree module taints kernel.
[   75.442150] esp32: Resetpin of Host is 6
[   75.442283] esp32: Triggering ESP reset.
[   75.443971] esp_sdio: probe of mmc1:0001:1 failed with error -110
[   75.444059] esp_sdio: probe of mmc1:0001:2 failed with error -110
[   76.316593] mmc1: card 0001 removed
[   76.370469] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[   76.377789] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[   76.380965] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[   76.382877] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[   76.382958] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[   76.383038] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[   76.384551] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[   76.384632] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[   76.384713] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[   76.384876] mmc1: new SDIO card at address 0001
[   76.385187] esp_probe: ESP network device detected
[   76.605736] Features supported are:
[   76.605755]   * WLAN
[   76.605764]   * BT/BLE
[   76.605772]     - HCI over SDIO
[   76.605781]     - BT/BLE dual mode
[   76.826443] esp_sdio: probe of mmc1:0001:2 failed with error -22
[   76.931105] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   76.931115] Bluetooth: BNEP filters: protocol multicast
[   76.931134] Bluetooth: BNEP socket layer initialized

hci0:   Type: Primary  Bus: SDIO
    BD Address: 3C:71:BF:47:37:4A  ACL MTU: 1021:9  SCO MTU: 255:4
    UP RUNNING 
    RX bytes:672 acl:0 sco:0 events:82 errors:0
    TX bytes:769 acl:0 sco:0 commands:42 errors:0

thanks

lph65724, 2020-10-01:
By the way, when I run example-advertisement, it shows LEAdvertisingManager1 interface not found, too.  Any thoughts?

thank you

MangeshMalusare, 2020-10-05:
Thanks for sharing logs and further information. We will check and get back at the earliest.

mantriyogesh, 2020-12-10:
Hi @lph65724,

This issue is fixed recently. Could you please resync with github repo and let us know the observations?