espressif/esp-hosted #11: ESP32 WIFI可以作为代理让其他linux设备通过它上网吗？

chenluhui2020, 2020-10-16:
ESP32自己可以上网，海思平台如果把ESP32当做网卡后（ifconfig能看到的网卡设备），也可以借助ESP32上网。那是否可以海思平台和ESP32都能同时借助ESP32的wifi模块上网呢？（比如ESP32连上网之后，海思和ESP32各有tcp/ip栈，esp32是代理，海思利用ESP32共享此网络）

MangeshMalusare, 2020-10-21:
Thanks for your question. In ESP Hosted solution, tcp/ip stack on host gets used. TCP/IP stack on ESP32 is not used.
i.e. in your case TCP/IP stack running on HiSilicon platform will get used. 
I did not really get your use case. Could you please provide more details of it? We will see if we can provide more relevant suggestion to you.

xiaoyaoweizi, 2020-10-21:
你好，目的要实现esp32一边自身能上网，一边能起到路由分发的功能，帮助HiSi芯片上网。各自都运行各自的TCP/IP stack。我司目前有一个项目应用是这样子的：esp32长期在线，HiSi芯片摄像头根据用户通过esp32来唤醒使用。为了节省HiSi芯片多一个wifi模块，想让HiSi芯片唤醒后，通过esp32来上网，能同时让两边都能通过esp32的网卡去联网，做数据转发。

xiaoyaoweizi, 2020-10-23:
Hi,our project requires that esp32 implement an Ethernet switch function: the exchange function between esp32 socket interface, SDIO virtual 802.3 network card and WiFi 802.11 network card.

We create a socket on esp32 to communicate with the IOT platform. At the same time, the host CPU can communicate with the Internet through the SDIO virtual network card of esp32.ESP32 socket and hosted CPU need connect Internet at the same time by WiFi 802.11.

But we run ESP hosted demo and eth2wifi demo in ESP IOT solution,and found 802.11 data receive and forward by below API:
 `esp_ err_ t esp_ wifi_ internal_ reg_ rxcb(wifi_ interface_ t ifx, wifi_ rxcb_ tfn);`
`int esp_ wifi_ internal_ tx(wifi_ interface_ t wifi_ if, void *buffer, uint16_ t len);`  
The APIs implement bridging function, if these two APIs are called, the host CPU can be connected to the network, and the socket inside the esp32 can not connect to the network.

We need a solution to keep the socket within esp32 while calling the above two APIs by host CPU.

Any advices to achieve this. Is it possible to open source code of this two APIs to us,we may can add switch function on it.

MangeshMalusare, 2020-10-29:
Thanks for providing further information. As mentioned earlier, in ESP hosted solution, for incoming packets, we skip network stack on ESP32 and forward the packets to host over SDIO interface. Network stack on host processes the received packet. 
Similarly, in case of outgoing packets, network stack on ESP32 is not used.

W.r.t. API's you mentioned, unfortunately those can not be open sourced.

I will see if I can find anything to get around this in order to achieve your goal. Will update soon.

xiaoyaoweizi, 2020-10-30:
Thinks for your reply.I am looking forward to your progress.

MangeshMalusare, 2020-11-10:
Hi,
I checked this further. There is no ready solution available for your case. You will have to update ESP-hosted solution as per your requirements. You will have to modify ESP-Hosted repo and esp-idf repo as per your needs. We are not very sure how this will work out as there are 2 network stacks and single phy interface. You might face many practical issues while doing this. But you can give it a try.

Rough idea is as below. 

Host Tx: [For this path, you can use ESP-hosted repo, you might need to update the code as some places based on your requirements]
1. Host will send packet over SDIO interface. 
2, ESP module will receive that packet
3. Once received, ESP module will send that packet over Wifi interface using esp_wifi_internal_tx.

Host Rx:
This is not going to be straight forward. You will have to hook your function in rx callback of lwip stack in esp-idf repo. Basically your function will make a copy of received packet and send it over SDIO interface to host.

e.g. for WLAN STA use case
1. In esp-idf repo, handle_sta_connected function in ./components/tcpip_adapter/event_handlers.c file, it registers a call back function to receive wifi packets from wifi driver. The callback function is: tcpip_adapter_sta_input
2. In tcpip_adapter_sta_input function, you can call your funcyion which clones the received packet and sends cloned packet over SDIO interface to host. Make sure to not touch any other code or received packet so that TCP/IP stack on ESP also receives this packet.

Similarly you can handle SoftAP case.