espressif/esp-hosted #12: Not working

callum133, 2020-10-30:
Hi, I get the following message on Visual Studio Code:

[image]

I am also getting the following error on the Raspberry Pi:

[image]

Do you have any ideas or a video tutorial?

I am trying to set up an SPI connection between an ESP32 and Raspberry pi.

MangeshMalusare, 2020-11-02:
The error on Raspberry Pi appears to be kernel configuration issue.. We are not using '__sanitizer_cov_trace_pc' symbol in our code. That must be getting included implicitly.
A quick search on Google tells me that some users have fixed this issue by including below in Makefile.

KCOV_INSTRUMENT := n

I am not sure if this will fix issue for you. If this does not help, please check your kernel configuration.

callum133, 2020-11-02:
I may have resolved it via re-installing. I had the same issue as "Transport module not found #9" but I have seemed to of fixed It by copying utils into the transport folder.

I am now just getting "failed to get AP scan list".
[image]

Have you got any assistance on the VS code side? I have installed your bin files onto the ESP32 and that seems to work fine.

MangeshMalusare, 2020-11-02:
Ideally python scripts should work as is.. movement of file/directory is not required..
We are checking VS code issue. Meanwhile could you please help close other issue. It would be easier for us to track in a single place.

callum133, 2020-11-02:
Which issue is that sorry?

I have managed to get it to work on the raspberry pi. utils can't be found in the python directory, moving into the transport folder appears to fix the issue. I just need a fix for the VS code side, thanks.

mantriyogesh, 2020-11-02:
Hello !

Seems that `VS Code` issue refers a watchdog issue.
We are shortly going to update github repository with new release which includes fix for this issue.

In case you do not wish to wait more, Below is a small code change,
`diff --git a/esp/esp_driver/network_adapter/main/app_main.c b/esp/esp_driver/network_adapter/main/app_main.c
index 94f7174..53287b2 100644
--- a/esp/esp_driver/network_adapter/main/app_main.c
+++ b/esp/esp_driver/network_adapter/main/app_main.c
@@ -412,6 +412,7 @@ void recv_task(void* pvParameters)
                if (if_context && if_context->if_ops && if_context->if_ops->read) {
                        buf_handle = if_context->if_ops->read(if_handle);
                        if (!buf_handle) {
+                               **usleep(10*1000);**
                                continue;
                        }
                }`

mantriyogesh, 2020-11-02:
Hello !

Seems that `VS Code` issue refers a watchdog issue.
We are shortly going to update github repository with new release which includes fix for this issue.

In case you do not wish to wait more, Below is a small code change,
```
--- a/esp/esp_driver/network_adapter/main/app_main.c
+++ b/esp/esp_driver/network_adapter/main/app_main.c
@@ -412,6 +412,7 @@ void recv_task(void* pvParameters)
                if (if_context && if_context->if_ops && if_context->if_ops->read) {
                        buf_handle = if_context->if_ops->read(if_handle);
                        if (!buf_handle) {
+                               usleep(10*1000);
                                continue;
                        }
                }
```

callum133, 2020-11-02:
Hi,

I have updated the code but I'm now getting the following error
[image]

Would it be possible to upload the complete VS code project?

I had to make changes to the Cmakelists.txt for the project to build. I also had to include missing header and C files into the project main folder.

MangeshMalusare, 2020-11-02:
We could have.. but unfortunately we do not use VS code in our development. Let me see if we can try that..
Meanwhile please share your CMakelists.txt changes.. Also please share your current sdkconfig file.. that will help us identify any differences in build configuration. Thanks!

callum133, 2020-11-02:
Hi, I have figured out that it fails due to the fact I changed ESP_SPI_CONTROLLER from 2 to 4.

I presume if I change the handshake pin to 4 and leave ESP_SPI_CONTROLLER at 2, the handshake will stay at 4?

callum133, 2020-11-02:
In the network_adaptor folder: 

[CMakeLists.txt](https://github.com/espressif/esp-hosted/files/5475315/CMakeLists.txt)

[sdkconfig.txt](https://github.com/espressif/esp-hosted/files/5475319/sdkconfig.txt)

In the main folder:

[CMakeLists.txt](https://github.com/espressif/esp-hosted/files/5475322/CMakeLists.txt)

[Kconfig - .txt](https://github.com/espressif/esp-hosted/files/5475323/Kconfig.-.txt)

What IDE do you use for your development?

MangeshMalusare, 2020-11-02:
Thanks for the update. 4 is not a valid value for SPI controller. We will add that check to make it more informative.
Handshake PIN is not linked to SPI controller. So you can use pin 4 as handshake pin with either SPI controller.

w.r.t IDE, we don't use it.. we use VIM as editor and make for compilation.

MangeshMalusare, 2020-11-02:
So I assume that you could proceed with above change. Please let us know if you face further issues. Thanks!

callum133, 2020-11-02:
Yes, the project compiles and enables me to connect to a station through SPI.

I can't seem to get the soft ap to work though. I get the same error as seen by jianwei-bai:

Traceback (most recent call last):
File "softap_config.py", line 61, in
softap_config = commands.wifi_set_softap_config(args.ssid, args.password, args.channel_id, args.encrp_mthd, args.max_conn, args.ssid_hidden, args.bw)
File "/home/pi/esp-hosted/host/linux/host_control/host_commands/commands.py", line 180, in wifi_set_softap_config
if ecn < esp_hosted_config_pb2.EncryptionMode.Type_WPA3_PSK:
AttributeError: 'module' object has no attribute 'EncryptionMode'

I've also noticed that the blue LED doesn't work anymore when the project is loaded through VS code. It must only be used in your BIN files for the SPI.

callum133, 2020-11-02:
What do you mean sorry?

mantriyogesh, 2020-11-02:
I think what _**ajita02**_ meant is, Regarding error,
> Traceback (most recent call last):
File "softap_config.py", line 61, in
softap_config = commands.wifi_set_softap_config(args.ssid, args.password, args.channel_id, args.encrp_mthd, args.max_conn, args.ssid_hidden, args.bw)
File "/home/pi/esp-hosted/host/linux/host_control/host_commands/commands.py", line 180, in wifi_set_softap_config
if ecn < esp_hosted_config_pb2.EncryptionMode.Type_WPA3_PSK:
AttributeError: 'module' object has no attribute 'EncryptionMode'

you faced, this issue is already fixed in latest github repository.

Can you please confirm if you have similar code to,
https://github.com/espressif/esp-hosted/blob/master/host/linux/host_control/host_commands/commands.py#L211

_i.e._ code similar to below at line 211.
```
199 def wifi_set_softap_config(ssid, pwd, chnl, ecn, max_conn, ssid_hidden, bw):
200     set_softap_config = EspHostedConfigPayload()
201     set_softap_config.msg = EspHostedConfigMsgType.TypeCmdSetSoftAPConfig
202     if len(ssid) > max_ssid_len :
203         print("SSID length is more than 32 Bytes")
204         return failure
205     if len(pwd) > max_password_len :
206         print("softAP password length is more than 64 bytes")
207         return failure
208     set_softap_config.cmd_set_softap_config.ssid = str(ssid)
209     set_softap_config.cmd_set_softap_config.pwd = str(pwd)
210     set_softap_config.cmd_set_softap_config.chnl = chnl
211     if ecn < EspHostedEncryptionMode.Type_WPA3_PSK:                                                                                               
212         set_softap_config.cmd_set_softap_config.ecn = ecn
213     else:
214         set_softap_config.cmd_set_softap_config.ecn = EspHostedEncryptionMode.Type_WPA2_PSK
215         print("Asked Encryption method is not supported in SoftAP mode, Setting Encryption method as WPA2_PSK")
216     set_softap_config.cmd_set_softap_config.max_conn = max_conn
217     set_softap_config.cmd_set_softap_config.ssid_hidden = ssid_hidden
218     set_softap_config.cmd_set_softap_config.bw = bw
219     protodata = set_softap_config.SerializeToString()
220     tp = Transport_pserial(interface)

```

callum133, 2020-11-02:
Hi,

Seems to work great now, thank you. Is there anything set up for me to send files from my windows PC to the raspberry pi through the ESP32?

mantriyogesh, 2020-11-02:
I am afraid if this directly is not supported.
Although I think with a simple socket programming code you could easily do that.

mantriyogesh, 2020-11-03:
Could you please close the bug, if the issue is resolved for you?

callum133, 2020-11-04:
Hi,

I have tried to add an extra bit of code into the main and it is causing the same issue. I removed the addition for the bug fix and the code seems to compile and work without it!

        // receive data from transport layer
                if (if_context && if_context->if_ops && if_context->if_ops->read) {
                        buf_handle = if_context->if_ops->read(if_handle);
                        if (!buf_handle) {
                              //usleep(10*1000);
                                continue;
            }

Starting from line 574 this code works enabling me to change the AP to a fixed SSID after the ESP32 is programmed.

static esp_err_t initialise_wifi(void)
{
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    esp_err_t result = esp_wifi_init_internal(&cfg);
    if (result != ESP_OK) {
        ESP_LOGE(TAG,"Init internal failed");
        return result;
    }

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));

    wifi_config_t softap_config = {
.ap = {
//.authmode = AP_AUTH,
.ssid = AP_SSID,
.password = AP_PASSWORD,
.max_connection = AP_MAX_CONN,
.channel = AP_CHANNEL,
.ssid_hidden = 0,

},
};
ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP,&softap_config));
ESP_ERROR_CHECK(esp_wifi_start());

Code that doesn’t work (included .authmode). I have tested this code in a different project and it works fine.
static void deinitialize_bluetooth(void)
{
#ifdef CONFIG_BTDM_CONTROLLER_HCI_MODE_VHCI
    if (vhci_send_sem) {
        /* Dummy take and give sema before deleting it */
        xSemaphoreTake(vhci_send_sem, portMAX_DELAY);
        xSemaphoreGive(vhci_send_sem);
        vSemaphoreDelete(vhci_send_sem);
        vhci_send_sem = NULL;
    }
    esp_bt_controller_disable();
    esp_bt_controller_deinit();
#endif
}
#endif

static esp_err_t initialise_wifi(void)
{
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    esp_err_t result = esp_wifi_init_internal(&cfg);
    if (result != ESP_OK) {
        ESP_LOGE(TAG,"Init internal failed");
        return result;
    }

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));

    wifi_config_t softap_config = {
.ap = {
.authmode = AP_AUTH,
.ssid = AP_SSID,
.password = AP_PASSWORD,
.max_connection = AP_MAX_CONN,
.channel = AP_CHANNEL,
.ssid_hidden = 0,

},
};
ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP,&softap_config));
ESP_ERROR_CHECK(esp_wifi_start());

I get the following error again.

[cid:image005.jpg@01D6B28B.F860F3B0]

Kind Regards

Callum Bassie
Design Engineer

T:  +44 (0)1922 613 633

[Lodar Logo - E-mail signature (PNG)]

Lodar Ltd, 60 Sandwell Street, Walsall, WS1 3EB, UK

[cid:image002.jpg@01D6B28A.ABF41760][cid:image003.jpg@01D6B28A.ABF41760]

NOTICE OF CONFIDENTIALITY - THIS MESSAGE IS INTENDED ONLY FOR THE ADDRESSEE. It may contain confidential and/or privileged material. Any liability (in negligence or otherwise) arising from any third party acting or refraining from acting, on any information contained in the email is hereby excluded. Lodar Ltd is not responsible for any personal opinions expressed in emails. If you have received this email in error please notify the sender immediately and do not disclose the content to any other person, use it for any purpose or store or copy the information in any medium. Whilst every effort has been made to ensure that attachments are virus free they must be checked before use especially those containing encrypted data. Registered Office: 60 Sandwell Street, Walsall, West Midlands, WS1 3EB. Company No: 1846936. VAT Registration No. GB 431 3011 18.

callum133, 2020-11-04:
[image]

mantriyogesh, 2020-11-04:
Hello,

Thank you for the detailed comment.

Please correct me if wrong, I could infer that you are trying to get station/softap working in **C**
If so, could you please have a look at test.c,
https://github.com/espressif/esp-hosted/blob/master/host/linux/host_control/test.c

SSID, password etc could be changed as per need in this file and `make` in directory, `host/linux/host_control`

Also could you please help us know the exact requirement so that we can help you better way.

callum133, 2020-11-04:
Hi,

I am trying to pre-program the ESP32 to operate as a specific softap on boot up. The plan is to then communicate with a PIC through SPI to send data back and forth through Wifi or Bluetooth. At the moment, I am trying to prove the concept with a Raspberry Pi before building the PIC project. I want the user to be able to reconfigure the AP once they purchase the end product, this is clearly working by running the Python scripts on the raspberry pi which is enabling me to set up an AP with a password and encryption. I can currently program the ESP32 to start as an AP with a given SSID but it won't let me enable the password and encryption. I am more than happy to use a different IDE if you feel that VS code is buggy and causing the problem.

mantriyogesh, 2020-11-04:
Hi,

I would _not_ recommend to setup softap initially like that as you might get yourself into multiple troubles like dhcp etc for softap.
What I would suggest you is, whenever the RaspberryPi  understands that basic communication is up,  setup softap with expected initial configuration.

Please see [ESP_PRIV_EVENT_INIT](https://github.com/espressif/esp-hosted/blob/master/host/linux/host_driver/esp32/spi/esp_spi.c#L172), which could be used to add your code to setup initial softap.

Now, To answer your issue of initial softap config not working, Please go through patch,
[initial_softap.patch.gz](https://github.com/espressif/esp-hosted/files/5487708/initial_softap.patch.gz)

This is hard-coded configuration, 
SSID: callum
Channel: 1
password: callum123
authmode: WIFI_AUTH_WPA_WPA2_PSK

This patch is just for your reference and out of scope for ESP-Hosted code repository.
With this your should be able to see softap with above config on ESP boot-up.
Once ESP is connected to RaspberryPi (linux), you can run `python softap_config.py ...` or test.c to change softap configuration.

Please **Note** below points ahead -
* For ESP based Inital softap - Static ip config would be needed for station (say laptop) while connecting, otherwise you would observe 'Obtaining IP Address' or similar error as DHCP would not be there.
* For RaspberryPi (Linux), it should fairly to easily setup dhcp on interface `ethap0` used for softap connection.

Please let us know if that works out for you.

callum133, 2020-11-04:
Hi, thanks for your help.

I have tried the following code. It compiles but the AP can not be found on my phone/computer.

static void deinitialize_bluetooth(void)
{
#ifdef CONFIG_BTDM_CONTROLLER_HCI_MODE_VHCI
	if (vhci_send_sem) {
		/* Dummy take and give sema before deleting it */
		xSemaphoreTake(vhci_send_sem, portMAX_DELAY);
		xSemaphoreGive(vhci_send_sem);
		vSemaphoreDelete(vhci_send_sem);
		vhci_send_sem = NULL;
	}
	esp_bt_controller_disable();
	esp_bt_controller_deinit();
#endif
}
#endif

static void initial_wifi_event_handler(void* arg, esp_event_base_t event_base,
                                    int32_t event_id, void* event_data)
{
    if (event_id == WIFI_EVENT_AP_STACONNECTED) {
        wifi_event_ap_staconnected_t* event = (wifi_event_ap_staconnected_t*) event_data;
        ESP_LOGI(TAG, "station "MACSTR" join, AID=%d",
                 MAC2STR(event->mac), event->aid);
    } else if (event_id == WIFI_EVENT_AP_STADISCONNECTED) {
        wifi_event_ap_stadisconnected_t* event = (wifi_event_ap_stadisconnected_t*) event_data;
        ESP_LOGI(TAG, "station "MACSTR" leave, AID=%d",
                 MAC2STR(event->mac), event->aid);
    }
}

static void initial_softap(void)
{
    /*ESP_ERROR_CHECK(esp_event_loop_create_default());

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
	*/

    ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &initial_wifi_event_handler, NULL));

    wifi_config_t wifi_config = {
        .ap = {
            .ssid = "Callum",
            .ssid_len = strlen("Callum"),
            .channel = 1,
            .password = "12345678910",
            .max_connection = 4,
            .authmode = WIFI_AUTH_WPA_WPA2_PSK
        },
    };

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config));
}

 static esp_err_t initialise_wifi(void)
 {
 	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	 	esp_err_t result = esp_wifi_init_internal(&cfg);
	if (result != ESP_OK) {
		ESP_LOGE(TAG,"Init internal failed");
	  		}
 		return result;
 	

	//callum hack
	initial_softap();

 	result = esp_wifi_start();
 	if (result != ESP_OK) {
 		ESP_LOGI(TAG,"Failed to start WiFi");
	 }
 }
/*static esp_err_t initialise_wifi(void)
{
	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_event_loop_create_default());
	esp_err_t result = esp_wifi_init_internal(&cfg);
	if (result != ESP_OK) {
		ESP_LOGE(TAG,"Init internal failed");
		return result;
	}

	ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));

	wifi_config_t softap_config = {
.ap = {
//.authmode = AP_AUTH,	
.ssid = AP_SSID, 
.password = AP_PASSWORD,
.max_connection = AP_MAX_CONN,
.channel = AP_CHANNEL,
.ssid_hidden = 0,

},
};
ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP,&softap_config));
ESP_ERROR_CHECK(esp_wifi_start());

	esp_wifi_set_debug_log();
	result = esp_supplicant_init();
	if (result != ESP_OK) {
		ESP_LOGE(TAG, "Failed to init supplicant (0x%x)", result);
		esp_err_t deinit_ret = esp_wifi_deinit_internal();
		if (deinit_ret != ESP_OK) {
			ESP_LOGE(TAG, "Failed to deinit Wi-Fi internal (0x%x)", deinit_ret);
			return deinit_ret;
		}
		return result;
	}
	result = esp_wifi_start();
	if (result != ESP_OK) {
		ESP_LOGI(TAG,"Failed to start WiFi");
		return result;
	}
	return result;

}*/

int event_handler(uint8_t val)
{
	switch(val) {
		case ESP_OPEN_DATA_PATH:
			ESP_EARLY_LOGI(TAG, "Start Data Path");
			datapath = 1;
			if_handle->state = ACTIVE;
			break;

		case ESP_CLOSE_DATA_PATH:
			ESP_EARLY_LOGI(TAG, "Stop Data Path");
			datapath = 0;
			if_handle->state = DEACTIVE;
			break;
	}
	return 0;
}

void app_main()
{
	esp_err_t ret;
	uint8_t capa = 0;
	print_firmware_version();

	capa = get_capabilities();

	//Initialize NVS
	ret = nvs_flash_init();

	if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
		ESP_ERROR_CHECK(nvs_flash_erase());
		ret = nvs_flash_init();
	}

	pc_pserial = protocomm_new();
	if (pc_pserial == NULL) {
		ESP_LOGE(TAG,"Failed to allocate memory for new instance of protocomm ");
		return;
	}

	if (protocomm_add_endpoint(pc_pserial, "control", data_transfer_handler, NULL) != ESP_OK) {
		ESP_LOGE(TAG, "Failed to add enpoint");
		return;
	}

	protocomm_pserial_start(pc_pserial, serial_write_data, serial_read_data);

	if_context = interface_insert_driver(event_handler);
	datapath = 1;

	if (!if_context || !if_context->if_ops) {
		ESP_LOGE(TAG, "Failed to insert driver\n");
		return;
	}

	if_handle = if_context->if_ops->init(capa);

	if (!if_handle) {
		ESP_LOGE(TAG, "Failed to initialize driver\n");
		return;
	}

	to_host_queue = xQueueCreate(TO_HOST_QUEUE_SIZE, sizeof(interface_buffer_handle_t));
	assert(to_host_queue != NULL);

	from_host_queue = xQueueCreate(FROM_HOST_QUEUE_SIZE, sizeof(interface_buffer_handle_t));
	assert(from_host_queue != NULL);

	ESP_ERROR_CHECK(ret);

	xTaskCreate(recv_task , "recv_task" , 4096 , NULL , 18 , NULL);
	xTaskCreate(send_task , "send_task" , 4096 , NULL , 18 , NULL);
	xTaskCreate(process_rx_task , "process_rx_task" , 4096 , NULL , 18 , NULL);

	tcpip_adapter_init();

	ESP_ERROR_CHECK(initialise_wifi());

#ifdef CONFIG_BT_ENABLED
	initialise_bluetooth();
#endif
	ESP_LOGI(TAG,"Initial set up done");

mantriyogesh, 2020-11-04:
Could you please first use patch as is if it works in stead of changing and trying.
Could you please cross check validity of below code ?

static esp_err_t initialise_wifi(void)
{
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
esp_err_t result = esp_wifi_init_internal(&cfg);
if (result != ESP_OK) {
ESP_LOGE(TAG,"Init internal failed");
}
**return result;**  // **<- Is this required here?**
//callum hack
initial_softap();    // **<- this will be left as dead code in that case.**

result = esp_wifi_start();
if (result != ESP_OK) {
	ESP_LOGI(TAG,"Failed to start WiFi");
 }
}

callum133, 2020-11-05:
[image]
Hi,

I get an error while applying the patch.

I have even changed the directory so it matches the project location.

mantriyogesh, 2020-11-05:
Anyway, Can you embed the code manually from patch ?
Code in patch is fairly small and easy to understand.

callum133, 2020-11-05:
I have tried and the project does not compile. 
I think there Is an error in the patch.

I have attached the C file with no extra code. Could you please add in the patch for me?

[app_main.zip](https://github.com/espressif/esp-hosted/files/5493690/app_main.zip)

callum133, 2020-11-05:
I think I have managed to get it to work.

I have noticed that I can connect to my windows PC but not my phone due to MAC address error.

I can see that the concept works though.

I have a question about Bluetooth. I have noticed that the ESP32 does not appear to be discoverable by other devices. I'm guessing this code is not set up to send files?

I have tried changing the Bluetooth settings in the Config and it doesn't appear to make any differences.

MangeshMalusare, 2020-11-05:
Hi Callum,
W.r.t BT, we have implemented a BT controller driver. Host needs to have a Bluetooth stack which can configure and control this BT controller. The services like file transfer, audio streaming etc are provided by Bluetooth stack. So to do file transfer or to put device in discoverable mode, you need to use utilities provided by a Bluetooth stack.

Hope this helps.

Thanks,
Mangesh

callum133, 2020-11-05:
Hi Mangesh.

Surely bluez is the bluetooth stack for the raspberry pi and it should let you make the ESP32 discoverable?

MangeshMalusare, 2020-11-05:
Yes, it should be able to make ESP32 discoverable..

callum133, 2020-11-05:
It doesn't seem to be working properly. I can scan and the hciconfig detects the ESP32 via SPI but if I want to use the example-gatt-server, it can't seem to find the device. Any ideas? I have tried the "make discoverable" in the top right-hand corner and it doesn't seem to be doing anything.

MangeshMalusare, 2020-11-06:
What error do you get with example-gatt-server?

callum133, 2020-11-06:
[image]
I have tried this on various versions of Bluez

MangeshMalusare, 2020-11-06:
This issue is being looked into. This is not working consistently. We hope to fix it soon. 

Thanks,
Mangesh