espressif/esp-hosted #14: SDIO doesn't work

matthias122, 2020-11-04:
I followed the instructions to Setup ESP32 Wrover kit as Wifi Frontend over SDIO with Raspberry PI. But it is not working.
SPI is working but it has to low band width.
Here are my logs from dmesg and putty. I think that dmesg looks not Ok. Could you please help.
[image]
[image]

MangeshMalusare, 2020-11-04:
Hello,
Thanks for reporting your problem.
As I can see, Raspberry Pi is not able to perform standard SDIO initialization. As per our experience such issues are related to setup. Could you please try following:

1. In most cases, such issue happens due to faulty or long jumper wires. Make sure to use good quality and shorter length jumper wires. We recommend 10cm long jumper wires. We have experienced such issues with > 10 cm long jumper wires.

2. Make sure to power ESP32 module and Raspberry Pi separately with a power supply that provide sufficient power. ESP32 module can not derive enough power through SDIO. You can use external power supply or use micro-USB cable.

3. Pull ups are required on SDIO lines. Since you have already mentioned that you are using ESP32 wrover kit which has required pull ups. So this should not be a problem here. In case if you are using any other ESP32 module, you will have to setup pull ups on SDIO lines.

Thanks,
Mangesh

matthias122, 2020-11-08:
Hello Mangesh,
I used shorter wires and I have powered the Wrover Kit with an external power supply.
But I'm still getting errors.
[image]
Could you please help
Thanks
Matthias

mantriyogesh, 2020-12-10:
Hi Matthias,

Apologies for delay, Could you please let us know if the issue still observed on latest github repo?

MangeshMalusare, 2020-12-10:
Hi Yogesh, This was resolved after using appropriate ESP-IDF version.

Hi Matthias,
Hope you have not faced any other issues. Kindly help close this if it is all right.

Thanks.

matthias122, 2020-12-11:
Hi Mangesh,
there was another issue with a resistor at a Dataline. I removed R50 on WroverKit V4.1.
Best regards
Matthias

mantriyogesh, 2021-02-11:
Thank you Matthias !

Could you please elaborate the exact change and reason behind that, that would help us if there is any further underlying problem(s).