espressif/esp-hosted #16: Can't change stm32cubeIDE target to STM32f407

alireza3137, 2020-12-03:
Hi
how can I change target to stm32f407?
in the example the MCU is STM32F469I-Discovery board.

mantriyogesh, 2020-12-04:
Hi Alireza3137,

Thank you for your query.

Yes, In ESP-Hosted project,  STM32F469I-Discovery board is used as an example MCU board.
You can use your preferred MCU. We have tried to keep the configuration at single place, so as it would be easy to port the solution to any MCU.

The STM32CubeMX 'IOC' project configuration file is at `host/stm32/proj/stm_spi_host.ioc`. This file has all hardware related information like GPIO to be used, SPI peripherals to be used.

To port ESP-Hosted solution to STM32F407,
1. Identify SPI peripherals available from target board data sheet
2. Identify GPIO to be used from target board data sheet / user manual
3. Port stm_spi_host.ioc to target board using STM32CubeMX
4. In case you opt for MCU other than ST, that also should be do-able, by translating 'IOC' configuration for that target board.
5. Connect the jumper wires as per target board's GPIO / SPI positions.

alireza3137, 2020-12-05:
@mantriyogesh 
Hi
I port ESP-Hosted as you mentioned. but I get this error in UART terminal:
Aborting at: C:/Users/Alireza/Desktop/esp-hosted-master/host/stm32/driver/spi/spi_drv.c:230

mantriyogesh, 2020-12-07:
Hi @alireza3137,

Could you please help to understand the place where it is aborting?

Place, https://github.com/espressif/esp-hosted/blob/master/host/stm32/driver/spi/spi_drv.c#L230 is not leading me into abort statement.
It would be great if you let me know short summary of changes and a snippet where the crash happening.

alireza3137, 2020-12-09:
@mantriyogesh 
Hi
Thanks for your reply;
I solved this problem then I encountered a new problem!!
I get this message in UART terminal: "ESP-Hosted for ESP32"
and in debug I understand that the code wait for receive for semaphore in line 689 in spi_drv.c :
			if (osSemaphoreWait(osSemaphore , osWaitForever) == osOK) {
				check_and_execute_spi_transaction();
			}
and I think ESP32 board did not interrupt GPIO_DATA_READY_Pin and GPIO_HANDSHAKE_Pin  to release semaphore .
I think that the firmware flashed to esp32 does not work properly.
is there any simple test so that I can understand if esp32 work fine?

my setup hardware is:
MCU: stm32f407zg
ESP: ESP32-WROOM-32

mantriyogesh, 2020-12-09:
Hi @alireza3137 ,

We have tested ESP32-WROVER-Kit and ESP32-S2 over SPI.
Ideally SPI over ESP32-WROOM-32 also should work.
I will test ESP32-WROOM-32 <--SPI --> STM32F469I and let you know the result shortly.

mantriyogesh, 2020-12-10:
Hi @alireza3137 ,

I could reproduce issue faced by you with ESP32-WROOM-32 <-- SPI --> STM32F469I.
I could learn in this exercise that the `NSS` pin not working as expected it to be. As an solution to this problem, we can manually handle NSS over GPIO. It would be good if GPIO Pin number is kept same as earlier NSS. In ESP-Hosted solution for STM32F469I, it is [pin 11](https://github.com/espressif/esp-hosted/blob/master/docs/MCU_based_host/Setup.md#hardware-connections-for-esp32).

To make WROOM work, I did following things -
1. *IOC file changes* - Refer to `host/stm32/proj/stm_spi_host_esp32s2.ioc`.
This is IOC (project configuration) file for ESP32-**S2**. NSS is manually handled in this IOC project.

2. Open this file with STM32CubeMX, you could see the `SPI CLK` is set to 40M. Now ESP-WROOM-32 supports 10M clock.
So adjust either `clock` or `prescaler` to make sure the clock is 10M.

[image]

3. Save the project.
4. Open `host/stm32/proj/stm_spi_host_esp32s2.ioc` file in editor and search `ProjectName`. Project name line should exactly be `ProjectManager.ProjectName=stm_spi_host`. Change it to `ProjectManager.ProjectName=stm_spi_host` if anything different.
For reference I have attached my changed IOC file.
[stm_spi_host_esp32s2.ioc.zip](https://github.com/espressif/esp-hosted/files/5675212/stm_spi_host_esp32s2.ioc.zip)

5. Now, follow step-by-step [ESP-Hosted documentation](https://github.com/espressif/esp-hosted/blob/master/docs/MCU_based_host/MCU_based_readme.md)  to setup STM32CubeIDE project from this changed 'IOC'.

With above steps, I could get ESP-Hosted SPI working for `ESP32-WROOM-32` and `STM32F469I`.

mantriyogesh, 2020-12-11:
We are also planning to add ESP32-WROOM-32 support shortly.
If cannot wait that long, please follow above post & let us know if anything blocks you ahead.

alireza3137, 2020-12-11:
@mantriyogesh 
many thanks you for your support and reply.
Finally I could get ESP-Hosted SPI working for my hardware. 
I was wondering if I know how much the transmission speed?

mantriyogesh, 2020-12-11:
Hi @alireza3137 ,
On MPU (RaspberryPi) counterpart, we have tested and the SPI transport speed is about 5Mbps.
however, as network stack is kept open to use on MCU, Exact speed is not measured/committed for ST.

alireza3137, 2020-12-11:
Hi @mantriyogesh 
I want make MCU (stm32f407) as a http server so that I can download SD memory content. How can I do that?

mantriyogesh, 2020-12-14:
Hi @alireza3137 ,

This is out of scope for ESP-Hosted.
But as suggestion,
1. Port network stack on ESP-Hosted. see network stack stub in ESP-Hosted. any network stack like lwip could be used.
2. HTTP server would just be application sitting over network stack
3. SD / SDIO API could be checked to see how to download and save stuff over ST
4. As a reference, you can use ST code itself, how these things implemented as examples.

alireza3137, 2020-12-15:
Hi @mantriyogesh 
Thanks for your reply.
how can port network stack on ESP-HOSTED?

alireza3137, 2020-12-15:
Hi @mantriyogesh
Thanks for your reply.
how can port lwip on ESP-HOSTED?

closed 2021-04-17 by alireza3137