espressif/esp-hosted #17: Support for beaglebone

SAtacker, 2020-12-06:
Is it possible to use without much modification in `beaglebone`? 
I have went through some of the code and other than `raspi-gpio` and `protobuf` , is there anything specific to rpi?

mantriyogesh, 2020-12-07:
Hi Shreyas,

Thanks for your question.

Yes, It should be easy do-able on beaglebone.
Hosted solution could be ported on both MPU and MCU.
With beaglebone, I assume you want to port it with Linux.

WRT your question,
'protobuf' is a portable layer by Google, which is used for packing and unpacking of data. for more details, refer [here](https://developers.google.com/protocol-buffers)

Regarding raspi-gpio, this is just handy utility to handle the GPIO mode through shell, you can do this through code as well.

Available transports for ESP-Hosted as if now are SDIO and SPI.

To port on MPU, below steps would be considered -
1. Decide which peripheral to use, SDIO or SPI
2. Identify peripheral pins
3. Identify GPIOs in addition to peripherals as per [hosted documentation for Linux based host].(https://gitlab.espressif.cn:6688/app-frameworks/esp_hosted/-/blob/master/docs/Linux_based_host/Linux_based_readme.md).  For example, reset ESP GPIO
4. Connect jumper cables to appropriate peripherals
5. Software setup as per documentation, mainly includes kernel headers, python2.7/3, protobuf library, BT source etc

SAtacker, 2020-12-07:
Thank You @mantriyogesh , i would try and get back here

closed 2020-12-07 by SAtacker