espressif/esp-hosted #18: Single ESP32 chip without Flash can work ?

Poiboss, 2021-01-11:
I had got that ESP32-WROOM or ESP32-WROVER can be used by Host-Linux to use WIFI/BT. ESP32-WROOM and ESP32-WROVER inside has a big Flash chip , So If I remove the inside Flash Chip,Could it still Work?
And i notice that the ESP32-U4WDH Chip has a built-in Flash,Can a Single ESP32-U4WDH Chip work?
Last,Thanks to provide Linux modules,So I can Use it to connect WIFI/BT in my self-design ARM-board

翻译：
基本了解了大概的情况，我想知道如果不用ESP32模组(铁皮模块，里面带一个外置的Flash芯片)，而只用ESP32单芯片，还能不能正常工作？铁皮模组对于我的板子来说太大了。
另外，我发现ESP32部分型号是有内置的Flash（比如：ESP32-U4WDH），这些型号不接外部的Flash能否正常工作？
最后，感谢提供Linux模块。

MangeshMalusare, 2021-01-12:
Thanks for your comment.

IRAM in ESP32-WROOM and ESP32-Wrover is not large enough to run ESP-Hosted solution. So Flash chip is needed.

ESP32-U4WDH has 4MB of Flash which is sufficient to run ESP-Hosted.

Hope this answers your questions.
Do let us know if you have further questions and face any problems.

beyond-space21, 2022-07-12:
Does the **ESP8285H16** chip alone work without an external Flash chip?

according to the datasheet, the ESP8286H16 chip does include Flash memory. (if I am not wrong).

[image]

And I think there is no need for any external flash chip unless I want more memory space.
is that right?