espressif/esp-hosted #20: When will the LwIP stack be supported？how does the lwip stack replace the network stub？

chunyexixiaoyu, 2021-01-19:


MangeshMalusare, 2021-01-20:
Thanks for your query. I assume it is with respect to STM32 based implementation.

There is no immediate plan to port lwIP stack to STM32. Though It may happen some time in future, we may not be able to provide ETA for this.

One needs to implement netif layer in order to support lwIP [or any network stack].
Please refer following sections of the document:
https://github.com/espressif/esp-hosted/blob/master/docs/MCU_based_host/Design.md#network-interface-layer-netif
https://github.com/espressif/esp-hosted/blob/master/docs/MCU_based_host/Design.md#network-interface-layer

Hope this helps. Please let us know if you have further queries on this.

grzegorzniemirowski, 2021-04-19:
@MangeshMalusare Hi. These links no longer work. There is no Design.md in MCU_based_host folder. I'm also interested in lwip integration.

mantriyogesh, 2021-04-20:
Hello !

Because of the recent documentation changes links are broken.
Please take a look at 
1. https://github.com/espressif/esp-hosted/blob/master/docs/MCU_based_host/MCU_based_architecture.md#14-network-interface-layer-netif
2. https://github.com/espressif/esp-hosted/blob/master/docs/MCU_based_host/MCU_based_architecture.md#15-network-stack-stub 

Also as Mangesh has told, we do want to integrate lwip, but current bandwidth is limiting us to move it to backlog.

drorgl, 2021-11-21:
I've done a POC for lwip, the results were not amazing for several reasons.
https://github.com/drorgl/stm32-lwip-esp-hosted/tree/master/stm32-lwip

mantriyogesh, 2021-11-22:
Hello @drorgl ,

Thank you for all your efforts for lwip integration.
Some of the review comments I have, which I mentioned in your new github issue created (https://github.com/espressif/esp-hosted/issues/72) -> https://github.com/espressif/esp-hosted/issues/72#issuecomment-975171888. Please take a look.