espressif/esp-hosted #22: SDIO support for MCU Host

seahcj, 2021-02-10:
May i know is there any plan to support SDIO connection between MCU host and ESP32? for esp-hosted solution.

MangeshMalusare, 2021-02-10:
Thanks for your query. 

We are considering adding support for SDIO connection between MCU host and ESP32.
But there are no immediate plans to implement this.

ameyinamdar, 2021-02-10:
The ESP32 side support is completely ready and is available in source format. This will not change from host to host. MCU side SDIO support is very specific to host MCU used and you can implement this based on specific MCU you plan to use. In future we will provide support for some standard MCU that has SDIO interface available.

seahcj, 2021-02-10:
Noted. Thanks for the information

closed 2021-02-10 by seahcj