espressif/esp-hosted #23: Linux support (other than Raspberry Pi)

Qbicz, 2021-02-11:
Hi, first of all, I think esp-hosted is a great project. Thanks!

I would like to use ESP32 as a WiFi + BLE slave for my IMX6 running and Embedded Linux. **esp-hosted** looks for me like a great solution, superior to **esp-at**, because from Linux host perspective, it makes BLE available over HCI and WiFi available over network interface, e.g. ethsta0.

I have read carefully through the documentation and started some porting. I have compiled and flashed the ESP32 SPI firmware. From what I understand, to make this project work on other Linux processor, I should cross-compile the kernel `esp32_spi.ko` module from `esp-hosted/host/linux/host_driver/esp32`, put it on my Linux host, load the module (`modprobe`), also make sure the correct SPI and GPIO pins are used.

Then, I should be able to run a python script to connect to WiFi, and to use BlueZ to control BLE.

- Question 1: Do you see any challenges for running this project on other device than Raspberry Pi? What else needs to be done?
- Question 2: Is it possible to keep WiFi connection and BLE connection both at the same time?

mantriyogesh, 2021-02-12:
Hi Filip,

We are very glad to see that you are following this solution since esp-at.
The ESP-Hosted is evolved since then and we are trying to add more feature set.

Coming to the queries,
1. Porting to another Linux:
- This should be fairly easy for any other Linux platform.
- Some areas you would want to concentrate:
  - GPIO lines and connections, according SW changes in driver module
  - rpi_init.sh - raspi-gpio changes
- We may port to other Linux platforms, but for sure not immediately
 2. Wi-Fi + BLE:
 We have tested basic tests to run BT + WiFi. This simultaneous should work fine. BLE is not tested but there should be no reason to work

willmoffat, 2021-02-12:
Hi Filip, I also tried to port this to the iMX and ran into problems with corrupt packets. Looks like the DMA engine requires aligned data.
As a workaround, we allocate an entire page and copy the header and payload there.

MangeshMalusare, 2021-02-12:
Hi Will, Thanks for your inputs. Please do let us know if you face any further issues. We would be happy to help.

Qbicz, 2021-02-12:
Hi @mantriyogesh and @MangeshMalusare, thanks for the replies, this is useful. 

@willmoffat, were you able to have the WiFi + BLE working in the end? Did you need changes on both ends (Linux module + on ESP32 side) or just in the Linux module?

Do you use SDIO or SPI solution?

willmoffat, 2021-02-13:
We're using 1-bit SDIO. No major changes to ESP code. Still testing BLE + WiFi.

Qbicz, 2021-02-15:
Thank you for sharing. I will let you know if I run into any issues.

prashantbhumkar, 2021-10-01:
Hello Mantriyogesh,

We are using imx as host and ESP32 as wifi module over SPI, We have integrated driver as a kernel module and also cross compiled using cross compiler, But when we are trying to insmod esp32_spi.ko, We have received error of No New device some time Master handle error and received kernel panic, We have checked with oscilloscope for Clk, MOSI, CS keeping spidev module enable just for testing, SPI bus working. 

were you able to have Wifi working ??

mantriyogesh, 2022-02-14:
@Qbicz ,
Could you please close the issue, if your problems are solved?