espressif/esp-hosted #26: Error reading data ESP32->Rasperry Pi

kian99, 2021-04-07:
Hi,

I'm having an issue receiving data from my ESP32 back to the Rasperry Pi. Details are below.

**Hardware:**
I'm using a custom PCB with an ESP32-WROOM-32D and a Rasperry Pi compute module so there are no jumper wires. I've tested this on two boards and I'm having the same issue on both.

I'm aware on the ESP32-WROOM there is a conflict with GPIO12 and the SDIO Dat 2 pin so I use the `espefuse.py set_flash_voltage 3.3V` command from [this link](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html#strapping-conflicts-dat2) to resolve that before flashing the ESP and adding the pull-up resistors.

All the appropriate pins are connected correctly and pulled-up where necessary.

**Software:**
Dmesg logs:
````
[  462.366336] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[  462.374053] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[  462.377798] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[  462.380860] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[  462.381003] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[  462.381143] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[  462.383748] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[  462.383889] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[  462.384029] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[  462.384312] mmc1: new SDIO card at address 0001
[  462.384868] esp_probe: ESP network device detected
[  462.598276] Features supported are:
[  462.819367] esp_sdio: probe of mmc1:0001:2 failed with error -22
[  487.468576] read_packet: Failed to read data - -84 [0]
````

Output of ifconfig -a:
````
ethap0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether 00:00:00:00:00:00  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

ethsta0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether 00:00:00:00:00:00  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wwan0: flags=4291<UP,BROADCAST,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 10.115.90.56  netmask 255.255.255.240  broadcast 10.115.90.63
        ether be:b1:64:b7:61:a3  txqueuelen 1000  (Ethernet)
        RX packets 8  bytes 1562 (1.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 19  bytes 2485 (2.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
````

So the ESP is getting recognised on the Pi as I can see from the ifconfig command and when I use the python script "ap_scan_list.py" I can see the on the ESP's serial port that it successfully scans the networks around me but nothing is returned and the Python script times out. Dmesg always returns some kind of read error. Sometimes I've also seen an error like `esp_handle_isr: CMD53 read/write error at 115` similar to what is seen in [this issue](https://github.com/espressif/esp-hosted/issues/14). Please let me know what I can try here and/or what other information I can provide. Thanks.

MangeshMalusare, 2021-04-08:
Thanks for reporting your issue.

Can you please confirm if you have setup pull ups on SDIO lines? Kindly check https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html#sd-pull-up-requirements for more details.

kian99, 2021-04-08:
Hi,

I can confirm that the pullups are correct according to that link, I have:

- 10k pull up on DAT0 (IO2)
- 10k pull up on DAT1 (IO4)
- 10k pull up on DAT2 (IO12)
- 10k pull up on DAT3 (IO13)
- 10k pull up on CMD  (IO15)
- No pull up/down on CLK (IO14)

kian99, 2021-04-14:
Hi,

I believe I have solved the issue. I am using the Pi compute module connecting pins 34-39 (SDIO1) to the ESP. 

Going off of the suggestion near the end of this link: https://www.raspberrypi.org/forums/viewtopic.php?t=275292
I managed to solve the problem. It seems that the drive strength on pins 34-39 on the Compute Module differs from those used on the Rasperry Pi 3B. Following the advice from that thread I added the following lines to the master dt-blob file under "cm3" (please note the pin numbers are changed to pins 34-39 compared to what is in the link).

````
	    pin@p34 { function = "emmc"; termination = "pull_up";    drive_strength_mA = < 16 >; }; // SD CLK
            pin@p35 { function = "emmc"; termination = "pull_up";    drive_strength_mA = < 16 >; }; // SD CMD
            pin@p36 { function = "emmc"; termination = "pull_up";    drive_strength_mA = < 16 >; }; // SD D0
            pin@p37 { function = "emmc"; termination = "pull_up";    drive_strength_mA = < 16 >; }; // SD D1
            pin@p38 { function = "emmc"; termination = "pull_up";    drive_strength_mA = < 16 >; }; // SD D2
            pin@p39 { function = "emmc"; termination = "pull_up";    drive_strength_mA = < 16 >; }; // SD D3
````

Compiling that and adding it to the /boot/ folder with the command `dtc -I dtb -O dts -o dt-blob.dts /boot/dt-blob.bin` and then adding the overlay `dtoverlay=sdio,poll_once=off,gpios_34_39` into /boot/config.txt did the trick.

I'm now able to successfully scan networks, connect to an AP and ping a website. I haven't tried making the ESP into an Access Point yet. I hope that helps anyone else running into similar issues with the Compute Module/other Linux devices.

I've run into one issue however, I did the following sequence of commands:
`python ap_scan_list.py`
`python station_connect.py 'MySSID' 'MyPWD'`
`ping 8.8.8.8`
I let this run for a while successfully.
`python ap_scan_list.py`
It then crashed here.

I did the above twice and it crashed consistently. A snippet of the output from the ESP was:
````
I (158240) slave_commands: Details of AP no 2                                   
E (158240) slave_commands: Invalid SSID length                                  
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
````

I have the full serial output from the ESP but please let me know whether I should put this into a new issue or use the current thread. Thanks.

MangeshMalusare, 2021-04-14:
Thank you for letting us know your fix.

We will try to replicate your other issue and let you know soon. Meanwhile can you please share output of first `python ap_scan_list.py` command. I would like to see if any AP in vicinity has any unusual name. Thanks!

kian99, 2021-04-14:
Here is the full output from the first `python ap_scan_list.py` command through to the point where the device rebooted. 

<details><summary>ESP32 Output</summary>
<p>

```
data ready                                                                      
I (50110) slave_commands: Station mode set in scan handler                      
I (51710) slave_commands: Total APs scanned = 10                                
I (51710) slave_commands: Details of AP no 0                                    
I (51710) slave_commands: SSID          i4 Group                                
I (51710) slave_commands: RSSI          -36                                     
I (51710) slave_commands: Channel               6                               
I (51720) slave_commands: BSSID                 88:dc:96:52:b5:f1               
I (51720) slave_commands: Auth mode             3                               
                                                                                
I (51730) slave_commands: Details of AP no 1                                    
I (51730) slave_commands: SSID          i4 Group_Guest                          
I (51740) slave_commands: RSSI          -38                                     
I (51740) slave_commands: Channel               6                               
I (51750) slave_commands: BSSID                 8e:dc:96:52:b5:f1               
I (51750) slave_commands: Auth mode             3                               
                                                                                
I (51760) slave_commands: Details of AP no 2                                    
I (51760) slave_commands: SSID          i4 Group_Guest                          
I (51770) slave_commands: RSSI          -62                                     
I (51770) slave_commands: Channel               11                              
I (51780) slave_commands: BSSID                 8e:dc:96:52:b5:4f               
I (51780) slave_commands: Auth mode             3                               
                                                                                
I (51790) slave_commands: Details of AP no 3                                    
I (51790) slave_commands: SSID          i4 Group                                
I (51800) slave_commands: RSSI          -62                                     
I (51800) slave_commands: Channel               11                              
I (51810) slave_commands: BSSID                 88:dc:96:52:b5:4f               
I (51810) slave_commands: Auth mode             3                               
                                                                                
I (51820) slave_commands: Details of AP no 4                                    
I (51820) slave_commands: SSID          i4 Group                                
I (51830) slave_commands: RSSI          -68                                     
I (51830) slave_commands: Channel               11                              
I (51830) slave_commands: BSSID                 88:dc:96:52:b5:dc               
I (51840) slave_commands: Auth mode             3                               
                                                                                
I (51840) slave_commands: Details of AP no 5                                    
I (51850) slave_commands: SSID          i4 Group_Guest                          
I (51850) slave_commands: RSSI          -68                                     
I (51860) slave_commands: Channel               11                              
I (51860) slave_commands: BSSID                 8e:dc:96:52:b5:dc               
I (51870) slave_commands: Auth mode             3                               
                                                                                
I (51870) slave_commands: Details of AP no 6                                    
I (51880) slave_commands: SSID          AC67B2557420                            
I (51880) slave_commands: RSSI          -70                                     
I (51890) slave_commands: Channel               1                               
I (51890) slave_commands: BSSID                 ac:67:b2:55:74:21               
I (51900) slave_commands: Auth mode             4                               
                                                                                
I (51900) slave_commands: Details of AP no 7                                    
I (51910) slave_commands: SSID          i4 Group_Guest                          
I (51910) slave_commands: RSSI          -84                                     
I (51920) slave_commands: Channel               1                               
I (51920) slave_commands: BSSID                 8e:dc:96:52:b5:2e               
I (51930) slave_commands: Auth mode             3                               
                                                                                
I (51930) slave_commands: Details of AP no 8                                    
I (51940) slave_commands: SSID          i4 Group                                
I (51940) slave_commands: RSSI          -84                                     
I (51940) slave_commands: Channel               1                               
I (51950) slave_commands: BSSID                 88:dc:96:52:b5:2e               
[... 151 lines cut ...]
I (1191) SDIO_SLAVE: Using SDIO interface                                       
[... cut ...]
I (1271) NETWORK_ADAPTER: Initial set up done   
```

</p>
</details>

For full clarity, on our board we also have a pull-down resistor on the ESP's EN pin to keep it off until we drive the pin high from the Pi. I've seen that the driver toggles this reset pin and then sets it as an input so on our board that resets the ESP anbut then leaves the ESP off. I have a script to change it back to an output and drive it high when I need and that's worked fine. This wouldn't affect the crashing I imagine but just in case. Thanks for the help!

MangeshMalusare, 2021-04-14:
Thanks for the detailed log. We appreciate the details.
We will go through it and come back soon.

ajita02, 2021-04-14:
Hi @kian99 , I have created a debug patch here [patch.zip](https://github.com/espressif/esp-hosted/files/6311466/patch.zip). Can you apply on lasted ESP-Hosted master and reproduce the above crash. Unzip attached patch in ESP-Hosted repo, then git apply debug.patch. Also in menuconfig enable GDB stub by  Component config -> ESP32-specific-> Invoke GDBStub. Share ESP32 side crash log. It will be helpful in further debugging.

kian99, 2021-04-15:
Hi @ajita02 

I've applied the patch on the latest commit in master and turned on GDBStub in the menuconfig. Please see the updated crash log below from the serial port of the ESP32.

Full sequence of commands from Pi:

- Toggled ESP EN pin high and held high to get ESP to boot and ifconfig and dmesg shows Pi is connected.
- `python ap_scan_list.py`
- `python station_connect.py 'i4 Group' 'removed'
- `ping 8.8.8.8` for a few seconds
- `python ap_scan_list.py`

<details><summary>Updated ESP32 Output</summary>
<p>

```
rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)                      
configsip: 0, SPIWP:0xee                                                     
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00      
mode:DIO, clock div:2                                       
load:0x3fff0018,len:4                                       
load:0x3fff001c,len:6900                                    
load:0x40078000,len:14052                                   
ho 0 tail 12 room 4                                         
load:0x40080400,len:4304                                    
entry 0x400806e8
I (71) boot: Chip Revision: 1
I (71) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot: ESP-IDF v4.0.2-442-g41efdb0b34-dirty 2nd stage bootloader          
I (41) boot: compile time 12:33:12                                              
I (42) boot: Enabling RNG early entropy source...                               
I (47) boot: SPI Speed      : 40MHz                                             
I (51) boot: SPI Mode       : DIO                                               
I (55) boot: SPI Flash Size : 2MB                                               
I (59) boot: Partition Table:                                                   
I (63) boot: ## Label            Usage          Type ST Offset   Length         
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000       
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000       
I (85) boot:  2 factory          factory app      00 00 00010000 00100000       
I (93) boot: End of partition table                                             
I (97) boot_comm: chip revision: 1, min. application chip revision: 0           
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1b420 (11
1648) map                                                                       
I (153) esp_image: segment 1: paddr=0x0002b448 vaddr=0x3ffbdb60 size=0x03600 ( 1
3824) load                                                                      
I (159) esp_image: segment 2: paddr=0x0002ea50 vaddr=0x40080000 size=0x00400 (  
1024) load                                                                      
I (160) esp_image: segment 3: paddr=0x0002ee58 vaddr=0x40080400 size=0x011b8 (  
4536) load                                                                      
I (170) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x8bbb4 (57
2340) map                                                                       
I (382) esp_image: segment 5: paddr=0x000bbbd4 vaddr=0x400815b8 size=0x1b8f0 (11
2880) load                                                                      
I (447) boot: Loaded app from partition at offset 0x10000                       
I (447) boot: Disabling RNG early entropy source...                             
I (448) cpu_start: Pro cpu up.                                                  
I (451) cpu_start: Application information:                                     
I (456) cpu_start: Project name:     network_adapter                            
I (461) cpu_start: App version:      release/v0.3-2-g80ef3f6-dirty              
I (468) cpu_start: Compile time:     Apr 15 2021 12:33:12                       
I (474) cpu_start: ELF file SHA256:  5fe3ff8b2c23a47c...                        
I (480) cpu_start: ESP-IDF:          v4.0.2-442-g41efdb0b34-dirty               
I (487) cpu_start: Single core mode                                             
I (492) heap_init: Initializing. RAM available for dynamic allocation:          
I (499) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM                       
I (505) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM                       
I (511) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM                      
I (517) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM                       
I (523) heap_init: At 3FFD1120 len 0000EEE0 (59 KiB): DRAM                      
I (529) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM                   
[... 202 lines cut ...]
000005                                                                          
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffba6ec  A13     : 0x00
000000                                                                          
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x00
00001c                                                                          
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xff
fffffa                                                                          
                                                                                
ELF file SHA256: 5fe3ff8b2c23a47c                                               
                                                                                
[... cut ...]
5630 0x400d7e88:0x3ffd5650 0x400d8797:0x3ffd56a0 0x400d8321:0x3ffd56f0 0x400d83b
a:0x3ffd5750 0x4008fb6d:0x3ffd5780                                              
                                                                                
Entering gdb stub now.                                                          
$T0b#e6
```

</p>
</details>

I ran the whole procedure twice and the crash occurred both times. The last output from the gdb stub was also the same both times.

I hope that helps.

ajita02, 2021-04-15:
Hi @kian99, Thanks for debug log. There are two problems in above crash. First its showing Invalid length error for valid SSID and second after error occurred underlying protobuf is producing a crash. As of now I can provide you a patch in short time which can fix crash problem. Resolving first problem will take some time as I never got above crash on my setup. Can you make sure, you are using latest ESP-IDF repo i.e. release/v4.0 for SDIO setup? so that we can be on same page.

kian99, 2021-04-15:
Hi @ajita02 

A fix would be appreciated but not urgent. I'll just add a check on my side to ensure I don't do a scan while connected to a network. If I disconnect first, scans happen without any crash so that's fine.

I can confirm my ESP-IDF is on release/v4.0, there are only 2 changes present. The first is the line:
`components/protocomm/src/common/protocomm_priv.h -> components/protocomm/include/common/protocomm_priv.h`

The second happened I think after I ran `git submodule update --init --recursive` to update my idf. The file /components/coap/libcoap has the line:
`Subproject commit 017528d7d96dca2318ed587b5fd3c0d57a89e49a-dirty`

where the `-dirty` part is the change.

ajita02, 2021-04-20:
Hi @kian99, Thanks for your inputs. Let me explain the issue first. ESP wifi driver was returning Hidden APs record. Hidden AP has empty SSID i.e. zero length which results in `Invalid SSID length` error. Apply this
[final.zip](https://github.com/espressif/esp-hosted/files/6344618/final.zip) in your esp-hosted repo, it will solve your crash. This crash will get fixed soon in master branch.
Let us know if any help is needed.

kian99, 2021-04-29:
Hi @ajita02,

Apologies for the delayed reply. I have applied the path and can confirm the issue has been solved! Thank you very much for the help.

closed 2021-04-29 by kian99