espressif/esp-hosted #27: It seems there is a problem in Data Ready interrupt from ESP

alireza3137, 2021-04-14:
Hi 
My question is :
In spi communication protocol is it possible that the data ready pin set HIGH while ESP has not  data packet  to send to host in case of  MCU host?

I thinks it is possible and it affects the efficiently so that the transfer rate would be drop a lot.
my hardware is:
esp32-wroom-32
stm32h743

I port LWIP to esp-hosted for MCU host and implement tcpip stack to use http application. I download a file from micro SD and the transfer rate is near 200KB/s. As you can see from wireshark image the host send 2 packet(1514 bytes) to ESP and wait for ACK packet and when ACK packet received the next packet is sent. the time for waiting for ACK packet and sending next packet(1514 bytes) is something between 11 to 15 ms. This time is very long because the ESP interrupt the host with data ready pin while ESP has not any valid data packet to send to MCU so there is a transaction between host and ESP so that the TX buffer and RX buffer are both empty and this transaction waste the time and drop transfer rate a lot.
As you can see in ping image this problem affect to ping time too so that the ping time is near 15 ms.
I think there is bug in ESP firmware which interrupt host with data ready pin while it has not any valid TX packet data.
[image]
 
[image]

alireza3137, 2021-04-18:
Is there any body who have any idea about my problem described above?

MangeshMalusare, 2021-04-19:
Thanks for reporting this issue. We will check and get back to you soon.

Thanks,
Mangesh

MangeshMalusare, 2021-04-20:
Thank for your patience.

The low level SPI protocol in ESP32 demands that a transaction should be configured all the time irrespective of whether ESP32 has valid data to be sent to host or not. So when ESP32 has a valid data, we configure another transaction and trigger data ready interrupt. Now in this case, host needs to be perform two SPI transactions to get actual data from ESP32.
1. First transaction to consume empty buffer
2. Second transaction to read actual data

This describes in general how rx works. We have added further optimizations to this part to improve throughput.
This logic is independent of the host. Please refer throughput numbers mentioned in https://github.com/espressif/esp-hosted#41-wi-fi-performance-in-shielded-environment.
The data rate you have mentioned is pretty low compared to these numbers.  Unfortunately we don't have lwip stack ported in our setup, so we won't be able to confirm through put numbers that we get. But as mentioned our ESP side design is independent of host, so we expect can similar performance on MCU as well. Of course it will vary as per MCU specification, but we don't expect that much drop. You will need to find bottleneck in your system in order to localize this issue.

About round trip time in ping, we see similar round trip time with Linux host too. It stays between 12-20ms.

Hope this helps you. Please let us know if you face further issues or have any questions or suggestions for us.

Thanks,
Mangesh

alireza3137, 2021-04-21:
Hi @MangeshMalusare 
Thanks for your reply.

I have checked the numbers mentioned in https://github.com/espressif/esp-hosted#41-wi-fi-performance-in-shielded-environment. But the result is for 20 and 40 Mhz Spi clock . I cannot change the spi clock further than 10 Mhz. How can I change spi clock?

Thanks,
Alireza

MangeshMalusare, 2021-04-22:
Hi Alireza,
20 and 40MHz in the troughput table indicates Wifi bandwidth and not SPI clock. 
We used 10MHz SPI clock on ESP32 Wrover kit while doing this throughput test.
Hope this helps.

Thanks,
Mangesh

alireza3137, 2021-04-23:
Hi Mangesh,
If I suppose SPI clock is 10MHz ideally the maximum speed of SPI is 1.25MBytes/s. So the time needed for transfer 1600 bytes(the data length which is sent in each SPI transaction) is (1600Bytes)/(1.25MBytes/s) = 1.28 ms.
in TCP protocol the server send 2 packets and wait to get ACK packet so 3 SPI transaction is needed plus 1 addition transaction with no TX valid data and RX valid data. Therefore 4 SPI transaction is needed. The time for these 4 transaction is 4*1.28 = 5.12 ms. In 4 transaction maximum data sent is 2*1600=3200 bytes. so the transfer rate is 3200 Bytes/5.12 ms = 625 KBytes/s = 5Mbps. This number is ideal and in reality the bandwidth is lower than 5Mbps. Therefore it seems the number imentioned in  https://github.com/espressif/esp-hosted#41-wi-fi-performance-in-shielded-environment is not valid. 
why we cannot increase the SPI clock to 40MHz?

Thanks,
Alireza

MangeshMalusare, 2021-04-27:
Hi Alireza,
Thanks for further analysis. As mentioned earlier, we have added optimization in this part of data path that allows us to get rid of empty transaction wherever possible. This allows us to achieve more data rate. You can refer the code for more information on this.

ESP32 wrover kit does not support SPI clock beyond 10MHz in SPI slave mode. If you need higher SPI clock, please consider switching to ESP32-S2 which supports 40MHz clock in slave mode.