espressif/esp-hosted #28: ESP32 does not probed by esp32_sdio driver

SerhiiElert, 2021-04-21:
Hi everybody,

While COVID shakes the earth I'm trying to build a soulution using ESP32-WROOM-32D module with esp-hosted software and Allwinner A10 based device.

I'm using 4bit SDIO interface. DAT2 strap option was disabled by burning eFuse -> DAT3:0 and CMD are equipped with 10k PU to 3.3V. All wires are about 6cm long.  Directly on ESP32 module 100n+47u caps are soldered -> power is checked and ok.

For complete information - v0.3 firmware was flashed and host running **Linux arm 5.4.40-armv7-x28 #139 SMP PREEMPT Tue Apr 20 16:58:47 EEST 2021 armv7l GNU/Linux** 

Here is ESP32 boot log:

[image]

Here is eFuse summary:

[image]

ESP32 module is detected by MMC controller as shown below:

[image]

But when I try to load esp32_sdio module -> I don't see any information that network adapter is detected as in your example log.
 

[image]

Could someone suggest what to check and what can cause such problem?
Kindly
Serhii

MangeshMalusare, 2021-04-27:
Hi Serhii,
Sorry for the delay. 
We will check it and get back as soon as possible.

Thanks,
Mangesh

SerhiiElert, 2021-04-27:
Thank you very much!

Serhii

MangeshMalusare, 2021-04-27:
Meanwhile can you please check below comment from other issue:
https://github.com/espressif/esp-hosted/issues/26#issuecomment-819344162

This was for RPi based platform but kindly see if you are facing similar issue on your platform.

SerhiiElert, 2021-04-27:
Thanks. I saw that thread.
It is not my case because another device connected to these SDIO lines works.

I saw with oscilloscope that host continuosly(MMC controller uses polling to detect ESP32) send requests but it looks like ESP32 is not responding.

Serhii

SerhiiElert, 2021-04-27:
Just for information - I did check v0.2 firmware -> same result.

MangeshMalusare, 2021-04-27:
Thanks for letting us know. By chance do you have another ESP32 module that you can try? Earlier we have seen such problems on faulty ESP32 modules.

SerhiiElert, 2021-04-27:
No problem, it is available in local stores. I will buy one more and test it.
I will let you know about result.

Thanks
Serhii

nis243, 2021-04-30:
Hello.
I had a hard time working with SDIO with a similar message the other day.
Environment: Raspberry Pi2, RaspberryPi A+ and ESP-WROOM-32
At that time, only EN was pulled up with 10KΩ.

The Raspberry Pi kernel was tested with 4.19, so I tried it with 4.19.127, 4.20.17, 5.0.21, 5.1.21, 5.2.21, 5.3.18, 5.4.83. However, with a similar message, it didn't work.

When I tried it with 4.18.20 and 4.17.19, it worked smoothly in the same environment.

Was the cable long? Or I don't know if it was because I didn't pull up other data lines, but only about 2Mbps came out.

The SPI interface was kernel 5.10.17 and ran at 4Mbps.
After confirming these behaviors, I am currently struggling to get them to work on my Orange PI PC. Maybe it works if you specify the interrupt pin in Device Tree.

SerhiiElert, 2021-04-30:
Hi,
Thanks for your feedback. I will try your idea.

SerhiiElert, 2021-05-01:
Hi,

Brand new ESP32 module -> same result as with the first one.

Coco1780, 2021-05-06:
I get same issue today, with RPi4 with wifi onboard.
I resolved with adding this in /boot/config.txt
```
dtoverlay=disable-wifi
dtoverlay=disable-bt
dtoverlay=sdio,poll_once=off
```

Not only "disable-bt" as written in doc.
Hope this help you.

MangeshMalusare, 2021-05-06:
Thanks Coco for the inputs. We will take a note of it and add it in documentation.

MangeshMalusare, 2021-05-06:
[0001-Debug.patch.zip](https://github.com/espressif/esp-hosted/files/6435691/0001-Debug.patch.zip)

HI Serhii,
Kindly try attached debug patch. It prints more information when your issue happens. Kindly share dmesg output.

Thanks,
Mangesh

SerhiiElert, 2021-05-06:
Hi Mangesh,

I will apply it tomorrow and post dmesg log here.

Thanks,
Serhii