espressif/esp-hosted #29: Could ESP-NOW be implemented as well?

nielsnl68, 2021-04-27:
Hi, just wondering how this could fit in.

I'm working on a esp-now project where the master is controlled by a raspberry pi, for this to work i created an ESP32 as bridge between the ESP-NOW network protocol and the PI over serial or maybe later over SPI.

Using the ESP32 for more then only as bridge as described above it could als work as network adapter as you made it happen in the project. Although in my opinion it cold just act as a normal network port a full wifi adapter i maybe easier and is all the control of SSID and pw in the hands of the PI.

At this point of time the ESP32 bridge sends every package received from ESP-NOW over serial including the senders MAC and package. 
And all messages received via serial will be send to the client with the given MAC address. 
Every new mac address will be added to the Peer list until the PI send a control message requesting to unpeer a specific mac address. 

For now this works but i have not setup this in a full scale setup yet and maybe i need to switch over to SPI in the future. 

But can we have a way to interface also with the ESP-NOW as well?

mantriyogesh, 2021-05-04:
Hello @nielsnl68,

Thank you for your query.
We will assess how easily could we place ESP-NOW over ESP-HOSTED and will get back to you.

Please let me brief you about `data` and `control` packets here,
All the `data` packets from AP or SoftAP are transparently sent to host.
Host is thought of master of the ESP (be it control or data packets).

The `control` packets are terminated at ESP using `serial` interface.
Control packets/interface is used to setup ESP's connection towards AP or establish SoftAP on ESP.
This `serial` interface would work over all available transports (SPI and SDIO till date).

By the time this ESP-NOW is not handled, I suggest you to go through `serial` interface code, which might be useful in what you are trying to achieve.

Please do let us know if anymore questions.

nielsnl68, 2021-05-04:
Sorry, i'm a little confused, but that could be just from translation. So please be patient with me. But first thanks in advanced, for willing to think about the option.

- The control packets are terminated at ESP using serial interface.
The first part seems clear the ESP handles the control package, but then you added the Serial part. Do you mean that the ESP although it handles the control packages, it is also sending those over Serial to the PI?

 - By the time this ESP-NOW is not handled, I suggest you to go through serial interface code, 
 Do you mend "At this time the ESP-NOW .... " when you mend so than that makes  makes it clear for me, As long the serial port is already in use to send the control packages as above understanding,

mantriyogesh, 2021-05-04:
@nielsnl68 ,
 
Ohh, My bad !

Serial interface here refers to [virtual serial interface](https://github.com/espressif/esp-hosted/blob/master/esp/esp_driver/network_adapter/main/app_main.c#L477).
This is `virtual serial interface` over transport (`SPI` or `SDIO`).

Please do not confuse this with a physical serial interface.

nielsnl68, 2021-05-04:
Perfect, i will wait for your update, in the main while i use the _physical serial interface_  to communicate between the ESP and the PI.

mantriyogesh, 2021-05-04:
@nielsnl68 

Did you mean `virtual serial` interface?

nielsnl68, 2021-05-04:
i mean the UART serial interface.

mantriyogesh, 2021-05-04:
Ohh. Understood.

May I know what hardware are you using as `host`? Is it MCU/MPU?
Also, what is the ESP chip set you are using?

Also, Please take a look at [feature matrix](https://github.com/espressif/esp-hosted/#15-feature-matrix) where BT/BLE is supported directly over UART.
Although in your case I see wifi is needed, in that case hosted transport (SPI/SDIO) could be used is needed.

nielsnl68, 2021-05-04:
Qua ESP it would be an ESP32 series on. But which version i did not have an decision made. I have some ESP32-CAM board laying around that i don't use so i think they will do fine for now. But any suggestions are welcome.

I want to use the Raspberry PI 4b to connect to the ESP32.  

The whole bluetooth pad i tried in the past but at that time i could not setup a good network that multiple devices could talk to each other. Without properly peering them etc. That's why i was so happy to find the ESP-NOW solution. 

The main use of the ESP-NOW is to send voice messages from the one to the other. I have seen it work and i did work well. Now i'm extending this to make an Intercom system around it. 
  
BTW. i have seen some efforts to allow the onboard wifi of the pi receiving the ESP-NOW packages but i could not get it to work, yet. and using a mode bridge is good for me for now.