espressif/esp-hosted #31: ESP32 reboots constantly after loading driver

UnstableSlayer, 2021-05-27:
I'm using raspberry pi zero.
After loading driver esp32 is stuck in reboot loop.

esp32 output:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6960
ho 0 tail 12 room 4
load:0x40078000,len:14704
load:0x40080400,len:4292
csum err:0x62!=0x2e
ets_main.c 371 
ets Jun  8 2016 00:22:57
```

UnstableSlayer, 2021-05-27:
Problem solved! I was flashing partition table with offset 0x4000 instead of 0x8000.

UnstableSlayer, 2021-05-27:
But now when i run test.py pi is failing to read data:

```
Failed to read data within 20 seconds
mac failure
Failed to read data within 20 seconds
wifi mode is failure
Failed to read data within 20 seconds
failure
Failed to read data within 20 seconds
failure
Failed to read data within 20 seconds
failure
Failed to read data within 20 seconds
failure
Failed to read data within 20 seconds
failure
Failed to read data within 20 seconds
failure
Failed to read data within 20 seconds
failure
Failed to read data within 20 seconds
failure
Failed to read data within 20 seconds
failure
Failed to read data within 20 seconds
failure
Failed to read data within 20 seconds
Power save mode is failure
Failed to read data within 20 seconds
failure
```

mantriyogesh, 2021-05-28:
Hello,
Thanks for the query.

Could you please Let us know below things?
1. Which ESP module used
2. Wires are correctly connected
3. device tree i.e. /boot/config.txt is correctly changed
4. Which is the idf version used

UnstableSlayer, 2021-05-28:
I use ESP32-WROOM-32. 
I used VSPI pins on esp32: https://i.stack.imgur.com/3NQUX.png

in config.txt: 
```
dtoverlay=dwc2
dtparam=spi=on
dtoverlay=disable-bt
```

I didn't use idf. I used esp flash download tool

mantriyogesh, 2021-05-28:
Hello !

Thank you for all the answers.

QUADWP and QUADHD are not supported pins as if now for SPI. Other four pins MOSI, MISO, SCLK and CS0 is expected to be connected. Further as mentioned in SPI documentation Additional Pins `Handshake Pin` and `Data Ready Pin` is also expected to be connected for ESP-Hosted. I hope those are connected.

`dtoverlay=dwc2` setting in your configuration is not used/tested in our environment. Could you please try without that, just to rule out the issue?

Also could you please state, which release binary of hosted did you flash?

UnstableSlayer, 2021-05-28:
Sorry, I actually moved to SDIO. Well, it doesn't work with sdio too. I flashed v0.3 firmware for sdio.
I used esptool.py this time to flash the firmware.
Esp32 output:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
```

dmesg:
```
[ 1764.806230] mmc-bcm2835 20300000.mmc: no support for card's volts
[ 1764.806268] mmc1: error -22 whilst initialising SDIO card
[ 1764.806290] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[ 1764.806308] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[ 1764.816398] mmc1: Controller never released inhibit bit(s).
[ 1764.845861] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[ 1764.845894] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[ 1764.868438] mmc1: Controller never released inhibit bit(s).
[ 1771.217652] mmc1: Controller never released inhibit bit(s).
[ 1771.217736] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[ 1883.845364] mmc1: Controller never released inhibit bit(s).
[ 1883.845417] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[ 1883.900049] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[ 1883.900080] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[ 1883.911450] mmc1: Controller never released inhibit bit(s).
[ 1883.963141] mmc1: Controller never released inhibit bit(s).
[ 1883.963200] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[ 1886.498269] mmc1: Controller never released inhibit bit(s).
[ 1886.498315] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[ 1886.553727] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[ 1886.553755] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
[ 1886.565077] mmc1: Controller never released inhibit bit(s).
[ 2200.146549] mmc-bcm2835 20300000.mmc: card claims to support voltages below defined range
[ 2200.162542] mmc1: error -110 whilst initialising SDIO card
[ 2200.175916] mmc1: Controller never released inhibit bit(s).
[ 2200.176046] mmc1: Got command interrupt 0x00030000 even though no command operation was in progress.
```

UnstableSlayer, 2021-06-15:
I bought another esp32 and now spi works! thanks for help! I wasn't able to make it work with sdio but i'll try again and let you know.

UnstableSlayer, 2021-06-15:
While using sdio, rpi can't find esps0 even though it lists it in `ls dev/`. Here is result from `python3 ap_scan_list.py`:
```
pi@raspberrypi:~/esp-hosted/host/linux/host_control/python_support $ python3 ap_scan_list.py
Traceback (most recent call last):
  File "ap_scan_list.py", line 23, in <module>
    ap_list = wifi_ap_scan_list()
  File "/home/pi/esp-hosted/host/linux/host_control/python_support/commands.py", line 401, in wifi_ap_scan_list
    tp = Transport_pserial(interface)
  File "/home/pi/esp-hosted/host/linux/host_control/python_support/transport/transport_pserial.py", line 45, in __init__
    self.f = os.open(devname, mode, flags)
OSError: [Errno 6] No such device or address: '/dev/esps0'
```

And here is `dmesg`:
```
[   31.091854] 8021q: 802.1Q VLAN Support v1.8
[   32.053457] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   32.581263] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   33.171046] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   92.475072] Bluetooth: Core ver 2.22
[   92.475262] NET: Registered protocol family 31
[   92.475278] Bluetooth: HCI device and connection manager initialized
[   92.480292] Bluetooth: HCI socket layer initialized
[   92.480330] Bluetooth: L2CAP socket layer initialized
[   92.480437] Bluetooth: SCO socket layer initialized
[  189.212007] esp32_sdio: loading out-of-tree module taints kernel.
[  189.213976] ESP32: Resetpin of Host is 6
[  189.214129] ESP32: Triggering ESP reset.
```

mantriyogesh, 2021-06-16:
Hi @UnstableSlayer,
 
Thanks a lot for testing SPI and confirming, it is working fine.

I hope [SDIO over Raspberry setup](https://github.com/espressif/esp-hosted/blob/master/docs/Linux_based_host/SDIO_setup.md) is followed for SDIO.

In addition, For SDIO with ESP32-WROOM-32, there are Pull Up requirements stated [here](https://github.com/espressif/esp-hosted/blob/master/docs/Linux_based_host/Troubleshoot.md#1-host-fails-to-detect-sdio-esp-peripheral).

Could you please confirm this?

mantriyogesh, 2022-02-19:
Hello @UnstableSlayer ,

Are the thngs working at your side now?
If yes, could you please close the bug?

UnstableSlayer, 2023-05-02:
I am sorry for the late response. I broke my raspberry pi zero and was unable to get my hands on another one. Seems like there won't be any activity so closing the issue.

closed 2023-05-02 by UnstableSlayer