espressif/esp-hosted #33: ethsta0 has no ip address

shchoi1, 2021-06-22:
Hi,
my platform is rk3399.

after insmod esp32_sdio.ko, i ran simple test application which is based on test_api.c of your c_support package.
below is the log.
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

==== test_set_wifi_mode =>
wifi mode is 0
====

==== test_get_wifi_mode =>
wifi mode is 0
====

==== test_get_available_wifi =>
Number of available APs is 7
0 th AP's ssid ...(omitted).. rssi "-55" channel "9" authentication mode "4"
1 th AP's ssid ...(omitted)..  rssi "-56" channel "5" authentication mode "3"
2 th AP's ssid ...(omitted)..  rssi "-56" channel "9" authentication mode "4"
...(omitted).. 
====

==== test_set_wifi_mode =>
wifi mode is 1
====

wifi_set_mac:190 Failed to set mac address
==== test_station_mode_set_mac_addr_of_esp =>
MAC address is not set
====

==== test_station_mode_get_mac_addr =>
Station mode: mac address c4:dd:57:aa:c3:64
====

==== test_station_mode_connect =>
Connected to AP
====

==== test_station_mode_get_info =>
AP's ssid ...(omitted)..
AP's bssid i.e. MAC address ...(omitted)..
AP's channel number 5
AP's rssi -59
AP's encryption mode 4
====

<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

and the result of ifconfig -a are as follows.

>>>>>>>>>>>>>>>>>>>>

[root@rk3399:/userdata]# ifconfig -a
ethap0    Link encap:Ethernet  HWaddr 00:00:00:00:00:00
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ethsta0   Link encap:Ethernet  HWaddr 00:00:00:00:00:00
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:2583 errors:0 dropped:2583 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:322652 (315.0 KiB)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:224 errors:0 dropped:0 overruns:0 frame:0
          TX packets:224 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:13888 (13.5 KiB)  TX bytes:13888 (13.5 KiB)

<<<<<<<<<<<<<<<<<<<<

in my opinion, if the configuration is done successfully, ethsta0 has the ip address like follows. but it didn't

wlan0     Link encap:Ethernet  HWaddr B0:F1:EC:98:7B:EC
          inet addr:192.168.20.107  Bcast:192.168.20.255  Mask:255.255.255.0
          inet6 addr: fe80::ae7a:e026:530:d35f/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:10 errors:0 dropped:3 overruns:0 frame:0
          TX packets:35 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1016 (1016.0 B)  TX bytes:4839 (4.7 KiB)

 and it seems it's network doesn't work.

do i have to something more to connect network?

shchoi1, 2021-06-22:
i ran below commands and then i could see the board got ip address.
ifconfig ethsta0 hw ethsta0 ...
ifconfig ethsta0 down
ifconfig ethsta0 up
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
ethsta0   Link encap:Ethernet  HWaddr AA:BB:DD:CC:EE:FF
          inet addr:169.254.132.239  Bcast:169.254.255.255  Mask:255.255.0.0
          inet6 addr: fe80::e8b6:5e6e:9dd6:d7a8/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:24485 errors:0 dropped:2053 overruns:0 frame:0
          TX packets:94 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:3429352 (3.2 MiB)  TX bytes:21688 (21.1 KiB)
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

but actually, connected AP doesn't provide such address. 
And of course, network of my board doesn't work.

could you give me some advice?

mantriyogesh, 2021-06-22:
Hello @shchoi1,

Thank you for your query.

This overall issue looks like some wrong configuration in network side.
Could you please check network configuration scripts for your platform for example, /etc/network/interfaces in debian.

Also, could you please check dmesg and /var/log/messages for any obvious network issues?
Next step to would be to check dhcp configuration in board and AP configuration.

At the last, suggestion to check the `station_connect.py` script which may guide sequential bash steps in case you cannot use python as is.

mantriyogesh, 2021-06-22:
Focus should be given that dhcp gets correct address while executing dhclient. Please refer to station_connect.py and commands executed in there.

shchoi1, 2021-06-22:
thank you mantriyogesh,
i will check what you suggest and share the result.

shchoi1, 2021-07-02:
this issue was caused by wrong setting of target board configuration.

closed 2021-07-02 by shchoi1

mantriyogesh, 2021-07-02:
If possible to you, could you please enlighten the board config issue?
This helps us to enrich our knowledge..

donny681, 2021-08-02:
@shchoi1  I got the same problem about getting ip,would you tell me solution?

shchoi1, 2021-08-03:
> @shchoi1 I got the same problem about getting ip,would you tell me solution?

this is what i did to get ip address.
insmod esp32_sdio.ko
mknod /dev/esps0 c 221 0
chmod 666 /dev/esps0

ifconfig ethsta0 hw ether aa:bb:dd:cc:ee:ff (your mac addr.)
ifconfig ethsta0 down
ifconfig ethsta0 up

and then, connect to AP and input passwd