espressif/esp-hosted #34: How to load ethsta0 interface on boot in Raspberry Pi?

ryanhz, 2021-06-29:
I can successfully start the interface, by

```
 $ cd host/linux/host_control/
 $ ./rpi_init.sh spi
```

then run 
```
cd host/linux/host_control/python_support/
python3 station_connect.py 'xyz' 'xyz123456'
```
To connect to wifi and create ethsta0 interface.  But how could we automatically load those on boot?

ajita02, 2021-06-29:
Hi @ryanhz , You can take help of [This issue](https://github.com/espressif/esp-hosted/issues/24). It basically helps to register station_connect script as service.

ryanhz, 2021-06-29:
@ajita02 Thanks for the reply.  I was aware of that issue. But it looks like the device vanish after reboot, so you have to run
```
./rpi_init.sh spi
```
again, but annoyedly,  it will compile the source code again.

MangeshMalusare, 2021-07-13:
Hi Ryan, 
The project can be customised as per your needs. We kept that autoload part open as many may like it and many may not.
Also, kindly note that rpi_init.sh script is convenience script. It can be customised as per your requirements. Please let us know if you face issues in achieving how you want it. We will try to provide all the needed support.

Thanks,
Mangesh

ryanhz, 2021-07-13:
@MangeshMalusare Thanks for your reply. I thought you have off-the-shelf solution.

mantriyogesh, 2021-07-14:
Hello Ryan,

As Mangesh said, we can help if you are stuck somewhere.
We may not know exact expectations and similarities across customers, so we try to keep the solution as generic as possible.

For your case, Simple script which looks similar below could be hooked as daemon or init script depending upon the distribution -

```
#!/bin/bash

sudo rm -rf /dev/esps0
sudo rmmod esp32_sdio
sleep 2
# persistent_directory could be any directory where built kernel module could be kept
sudo insmod <persistent_directory>/esp32_sdio.ko  resetpin=6
sudo mknod /dev/esps0 c 221 0
sudo chmod 666 /dev/esps0
sleep 1
#test.out is optional user application derived from host/linux/host_control/c_support
#test.py is optional user application derived from host/linux/host_control/python_support
#as per customer expectation one of these line could be uncommented
# <persistent_dir>/test.out
# <persistent_dir>/test.py
# sleep 1
```

You do not need to rebuild the module again.
As you are using raspberry Pi, Minimal things as per your expectation from `rpi_init.sh` could be fetched and hooked as init script.
You can refer to some ways to do it like [this](https://www.dexterindustries.com/howto/run-a-program-on-your-raspberry-pi-at-startup/#:~:text=The%20third%20method%20to%20run,shutdown%20or%20reboot%20the%20system).

I hope this helps.

ryanhz, 2021-07-14:
@mantriyogesh thank you so much for your reply.  Yes, I was thinking of write a script as systemd to load the adapter.