espressif/esp-hosted #35: Low bandwidth Linux Host

HikingDev, 2021-06-30:
Hi,

The target is an ESP32-WROOVER-B
The host is a Raspberry Pi 3 Model B V 1.2

ESP-IDF v4.3 was used with the master to build esp-hosted for Linux Host with SPI transport protocol.

According to the [Performance and Validation](https://github.com/espressif/esp-hosted#4-performance-and-validation) Chapter a throughput of up to roughly 8 Mbps should be possible in Station Mode.

Three tests were performed:

A hotspot on a Workstation, connecting the ESP32/Raspberry to the hotspot and downloading an ISO from [mint mirror](https://mirror.bauhuette.fh-aachen.de/linuxmint-cd/stable/20.1/linuxmint-20.1-cinnamon-64bit.iso)
A hotspot on a Workstation, connecting the ESP32/Raspberry to the hotspot and downloading an ISO directly from workstation (python webserver)
An iPhone hotspot, downloading the aforementioned iso from mint mirror

In all cases a throughput of around 800 KiB/s was achieved.

The SPI_INITIAL_CLK_MHZ 10 was for testing purposes also lowered to 1Mhz and 500 Khz.
The throughput dropped to 100 KiB/s.

Any advices how the throughput can be increased? 
The jumper cables have a length of 5 cm.
The documentation states that the SPI slave can only work with < 10 Mhz

HikingDev, 2021-06-30:
Closed through a silly calculation error

closed 2021-06-30 by HikingDev

mantriyogesh, 2021-06-30:
Hi Benjamin,

Thanks for your query.
This is very low throughput I could understand.

Also, please check Wi-Fi conditions, i.e. distance from Wi-Fi, any obvious interference near the ESP32?