espressif/esp-hosted #37: linux wlan

LSPgh, 2021-07-02:
Hi,

I have successfully got the ESP32-hosted solution working using the python scripts provided.

I can see that once connected, the interface is presented to the OS as an Ethernet adapter would be. Is there any way to configure this drive to represent itself as a WiFi adapter, that would support scanning in the same way a native one would? The ESP32 is currently being considered as the on board WiFi module, accompanying the main ARM processor running Linux.

thanks

fonix232, 2021-07-12:
No, not in the current form. As the README states, `esp-hosted` provides a 802.3 network interface - think of it as if you connected to an ESP32 "router" via Ethernet.

Hopefully with the ESP32-S2 and S3, both having native USB support, we will have some proper WiFi stack implementation, but that will be a much more complex implementation than the current work-around.

And yes, I say work-around, since the current solution basically emulates the behaviour of a routed networking device. You connect to the ESP32 via a localised 802.3 network (over SDIO or SPI), where the WiFi network is connected/bridged over to the host, without direct control. It's an incredibly convoluted way of handling WiFi management, but also much more straightforward to implement than a full WiFi stack that grants direct control of the underlying hardware to the host.

mantriyogesh, 2022-07-26:
Just to update,

With the latest release, cfg802.11 support is added. With that, one can directly configure using wpa_supplicant.

LSPgh, 2022-07-26:
That's great news!