espressif/esp-hosted #38: why not suport openwrt?it needs it more

3050311118, 2021-07-08:
raspi 3  raspi 4    they have wifi and bt 。
i think openwrt needs it more

fonix232, 2021-07-12:
There's a number of changes required for OpenWrt support - definitely doable, but unlikely to be supported in mainstream OpenWrt, since there's no commercial device using `esp-hosted`.

1. The `host_driver` package needs to be ported to the mainstream Linux kernel
This is unlikely to happen in its current shape, since the code is quite... Hacky at best.
2. The driver is using a script for initialisation
This is generally frowned upon in Linux land. It works for a custom solution, but no official support will be given, and again, this is another point against merging things. 
3. The controller also needs to be re-worked to support multiple platforms and standards

Basically, for proper OpenWrt support, the driver should set itself up automatically, and the control interface should be using standard Linux approach instead of a custom control binary. The fact that this project only provides a 802.3 interface, not 802.11, makes it harder for OpenWrt to manage it properly.

Hopefully manufacturers will pick this up - having a secondary, low-power and low cost WiFi/BLE interface for configuring, accessing the web interface of routers and such. It's a good option since in most cases it can be configured to provide a completely separate network from the main set, meaning even if you completely obliterate all the default settings, or somehow manage to create a configuration that results in network failure, you can recover without the need of a full reset - since the ESP-provided "config network" still exists.

MangeshMalusare, 2021-07-13:
Hello,
We don not support openwrt out of the box. But the solution can be easily ported to openwrt. Please refer 'https://github.com/espressif/esp-hosted/blob/master/docs/Linux_based_host/porting_guide.md' if you are interested in porting to openwrt.

Thanks,
Mangesh