espressif/esp-hosted #39: Static IP Problems

axelmoser, 2021-07-19:
I am using the ESP 32 with the RPi over SDIO.
While using in dhcp mode, everything works ok.
When i try to use a static IP (I modified the station_connect.py), the ESP gets 2 IPs, Dhcp and Static.
I need to get only one address.
I use:
        command = 'sudo ifconfig ethsta0 down'
        os.system(command)
        time.sleep(1)

        command = 'sudo ifconfig ethsta0 {0} netmask {1} hw ether {2}'.format(args.ipadd, args.ipmask, str(sta_mac))
        os.system(command)
        time.sleep(1)

        command = 'sudo ifconfig ethsta0 up'
        os.system(command)

Or should I use another commands to connect?
Thanks!

MangeshMalusare, 2021-07-19:
Hello, 
Your command looks good. If you do not want DHCP address, then you need to stop/disable dhcp client running on your RPi. 

Thanks,
Mangesh

axelmoser, 2021-07-19:
Thanks Mangesh,

The dhcpcd was running...

`sudo service dhcpcd stop`

If running in static ip mode resolved everything.
Thank you very much!