espressif/esp-hosted #40: Can not get ip

donny681, 2021-08-01:
I use esp32 spi interface to connect to rpi.The esp32 connected wifi sucessfully,but it cannot get ip from dhcp.

`
pi@raspberrypi:~/esp-hosted/host/linux/host_control/python_support $ python station_connect.py 'AC2100' 'wifiwifi'
station MAC address bc:dd:c2:d1:b3:10
Connected to AC2100
sudo ifconfig ethsta0 down
sudo ifconfig ethsta0 hw ether bc:dd:c2:d1:b3:10
sudo ifconfig ethsta0 up
Killed old client process
sudo dhclient ethsta0 -r
Internet Systems Consortium DHCP Client 4.4.1
Copyright 2004-2018 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/ethsta0/bc:dd:c2:d1:b3:10
Sending on   LPF/ethsta0/bc:dd:c2:d1:b3:10
Sending on   Socket/fallback
DHCPDISCOVER on ethsta0 to 255.255.255.255 port 67 interval 6
DHCPDISCOVER on ethsta0 to 255.255.255.255 port 67 interval 8
DHCPDISCOVER on ethsta0 to 255.255.255.255 port 67 interval 14
DHCPDISCOVER on ethsta0 to 255.255.255.255 port 67 interval 20
DHCPDISCOVER on ethsta0 to 255.255.255.255 port 67 interval 11
DHCPDISCOVER on ethsta0 to 255.255.255.255 port 67 interval 2
No DHCPOFFERS received.
No working leases in persistent database - sleeping.
sudo dhclient ethsta0 -v
Success in setting AP config
pi@raspberrypi:~/esp-hosted/host/linux/host_control/python_support $ ifcofnig
-bash: ifcofnig: command not found
pi@raspberrypi:~/esp-hosted/host/linux/host_control/python_support $ ifconfig
enxb827eb7d1286: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.121  netmask 255.255.255.0  broadcast 192.168.0.255
        inet6 fe80::b0df:5751:a4f9:6bda  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:7d:12:86  txqueuelen 1000  (Ethernet)
        RX packets 3779  bytes 367614 (358.9 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1874  bytes 247806 (241.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

ethap0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 169.254.221.117  netmask 255.255.0.0  broadcast 169.254.255.255
        inet6 fe80::4f12:90f2:618a:5c3c  prefixlen 64  scopeid 0x20<link>
        ether bc:dd:c2:d1:b3:11  txqueuelen 1000  (Ethernet)
        RX packets 608  bytes 100491 (98.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 113  bytes 22440 (21.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

ethsta0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 169.254.246.62  netmask 255.255.0.0  broadcast 169.254.255.255
        inet6 fe80::f4ce:cda0:201f:45b9  prefixlen 64  scopeid 0x20<link>
        ether bc:dd:c2:d1:b3:10  txqueuelen 1000  (Ethernet)
        RX packets 644  bytes 127307 (124.3 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 285  bytes 50627 (49.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 4  bytes 1128 (1.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 4  bytes 1128 (1.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether b8:27:eb:28:47:d3  txqueuelen 1000  (Ethernet)
[... 189 lines cut ...]
I (987276) slave_commands: connected to ap SSID:'Ziroom-13A02', password:'4001001111'
data ready 
I (1338856) slave_commands: Get station mac address
I (1338856) slave_commands: mac [bc:dd:c2:d1:b3:10] 
data ready 
I (1338866) slave_commands: Disconnecting from previously connected AP
I (1338866) wifi:state: run -> init (0)
I (1338876) wifi:pm stop, total sleep time: 315850937 us / 351606871 us

I (1338876) wifi:new:<11,0>, old:<11,2>, ap:<255,255>, sta:<11,2>, prof:1
I (1338886) slave_commands: station mode set
I (1339016) wifi:new:<1,1>, old:<11,0>, ap:<255,255>, sta:<1,1>, prof:1
I (1339986) wifi:state: init -> auth (b0)
I (1339996) wifi:state: auth -> assoc (0)
I (1339996) wifi:state: assoc -> run (10)
I (1340616) wifi:connected with AC2100, aid = 3, channel 1, 40U, bssid = 1c:fa:68:b9:99:ca
I (1340616) wifi:security: WPA2-PSK, phy: bgn, rssi: -25
I (1340616) wifi:pm start, type: 1

I (1340626) slave_commands: Connected to AP
I (1340626) slave_commands: connected to ap SSID:'AC2100', password:'wifiwifi'
I (1340646) wifi:AP's beacon interval = 102400 us, DTIM period = 1

`
Could you tell me the reason why rpi can not get ip address from DHCP?

mantriyogesh, 2021-08-01:
Hello Divter,

Thank you for your query but DHCP is not under the scope of this project.

Anyway I would suggest you to check your dhcp configuration, /etc/network/interfaces or similar script used for network configuration. I would also suggect to check dhcp server logs if you are unsure what happening.

Alternatively to localize dhcp issue, I would suggest to set up static ip first. If that is working, dhcp problem can be debugged further.

donny681, 2021-08-02:
Yes,if I set up static ip and it works.But I want to know the reason why rpi can not get wifi ip?How to fix it ?

mantriyogesh, 2022-09-24:
Sorry, this went out of focus. Were you able to get it working @donny681 ?