espressif/esp-hosted #41: Handshake problems usign MCU interface

starkeeper82, 2021-08-09:
Hello,
I got some problems using the MCU interface. In some cases the handshake pins i controlled by the ESP32 in a weired way. At the end of a transmission the handshake goes low and high while the ready pin stays at a high state.

This is the logic analyser output:
[image]

At first a normal transmission is made. Probably Host and the Slave are sending data to each other. At the end of the transission the ESP takes the handshake pin low to signalize the processing of the received data. But then it starts to get worse. The handshake pin goes up for a short period of time, signalizing the host that a transaction should be started, because the ready pin is still high.

But before the new transaction can be finished the handshake pin is going low again. I don't know what is going on inside the ESP to produce this behavior. But I know that each time this happens there is data lost, that should have been transferred from the ESP to the Host. The host is receiving a dummy frame in this case.

I even get this behaviour sometimes during initialization, when I try to read the MAC address from the ESP.

mantriyogesh, 2021-08-12:
Hello @starkeeper82,

I understood your problem.
As per your handshake pin graph in logic analyzer output,  it may cause troubles.
Could you please let us know your git commit ID?
We will try to get the problem reproduced.

starkeeper82, 2021-08-12:
Hi @mantriyogesh,
my MCU project is not located on github. As far as I was able to investigate this it seem to be a timing problem. I am using a STM32 microcontroller with DMA enabled to transfer the data via SPI. I always toggel the CS lin on the STM32 by software and do not use the NSS hardware line. I saw some comments in the code of MCU host implementation stating that using the NSS hardware line of the STM32 would cause problems.

But using the DMA on STM32 side is quite fast and might be a key to the problem. Reducing the SPI clock to ~3MHz did not change anything. Than I did some testing by replacing the DMA transfer with a software transfer and did not experience the issue again. But the transfer of each data byte is then much slower. 

Reading here:
[https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_slave.html#spi-dma-known-issues](url)

tells me that there is a known limitation with the DMA, SPI slave and ESP32 when using the SPI mode 2. The ESP-Hosted project uses SPI with DMA and mode 2. Which seems to be not so great. But I already changed the ESP code and it did not change my problem.

I am using a plain "old" ESP32. Have you tested the implementation with ESP32, ESP32-S2 or ESP32-C3 ?

mantriyogesh, 2022-02-14:
Hello @starkeeper82,

This just to notify you,
ESP32-S2 and ESP32-C3 is available.