espressif/esp-hosted #43: ESP32 Power Cycle Breaks SDIO Driver hosted on Linux

dwhitters, 2021-08-13:
# Issue
The char device registration and addition is occurring in the probe function and the deletion and unregistration is occurring in the remove function.

This results in additions when the ESP32 is detected while the driver is running, and unregistration when the ESP32 is no longer powered. Unregistering the device while a node that is connected to it is in use by a process (e.g. **cat** results in the node no longer being connected/usable, even though the ESP32 may come back and the device will again be registered with the kernel with the same number.

# Steps to Reproduce
1. Power ESP32 on
2. Build and insert the kernel module using `./rpi_init.sh sdio`
2. `cat /dev/esps0 | xxd`
3. Power off ESP32
4. Power on ESP32
    * The _/dev/esps0_ node will no longer operate.
5. Kill cat
6. `cat /dev/esps0 | xxd`
    * This will result in the following error: `cat: /dev/esps0: No such device or address`

# Potential Solution
Char device registration appears to be typical in the __init function of a device driver as seen in the following link:
https://embetronicx.com/tutorials/linux/device-drivers/device-file-creation-for-character-drivers/
This way, a device is always registered with the kernel, regardless of if an external device is connected to the driver. Nodes can then be created and attached to the kernel device; they remain usable, regardless of the presence of the external device.

When the  _esp_serial_init_ is moved to the __init function and the _esp_serial_cleanup_ is moved to the __exit function, the device node remains operable across power cycle.

mantriyogesh, 2021-08-13:
Hello @dwhitters ,

Thank you for your valuable suggestion.
We will work on this and try to this proposed solution over master.

If you already have working solution, we welcome you to submit the pull request as well.
We will try to review and merge the solution.

mantriyogesh, 2026-09-28:
This proposed change was implemented in the FG codebase some time ago, so I am closing the issue.

closed 2026-09-28 by mantriyogesh