espressif/esp-hosted #44: No SPI outcoming command by esp32.ko

neva-blyad, 2021-08-18:
Hello,
I have Linux board with ESP32 board connected via SPI.
ESP board is ESP-WROOM-32. It is directly wired to my Linux NXP chip.

File esp32_spi.c is compiled with right settings of SPI bus, number, chip select.
I have load module esp32.ko and created device in /dev:
insmod esp32.ko
mknod /dev/esps0 c 221 0
chmod 666 /dev/esps0

Then I'm trying to scan Wi-Fi using python script:
python ap_scan_list.py

And what it has output:
Failed to read data within 20 seconds
Traceback (most recent call last):
  File "/esp-hosted/esp-hosted/host/linux/host_control/python_support/ap_scan_list.py", line 23, in <module>
    ap_list = wifi_ap_scan_list()
  File "/esp-hosted/esp-hosted/host/linux/host_control/python_support/commands.py", line 412, in wifi_ap_scan_list
    get_ap_scan_list.ParseFromString(response[1])
  File "/usr/lib/python3.9/site-packages/google/protobuf/message.py", line 199, in ParseFromString
    return self.MergeFromString(serialized)
  File "/usr/lib/python3.9/site-packages/google/protobuf/internal/python_message.py", line 1136, in MergeFromString
    serialized = memoryview(serialized)
TypeError: memoryview: a bytes-like object is required, not 'str'

As you see, there is no response from ESP32. I've checked signal voltage with oscilloscope and it showed there is no SPI request from Linux side to ESP32 at all.

What I have missed?
Thank you.

ajita02, 2021-08-18:
@neva-blyad Its problem with your ESP. Your ESP seems down. It could be power problem or connecting wires problem with ESP.

neva-blyad, 2021-08-18:
> @neva-blyad It could be problem with your ESP. It may be power problem or connecting wires problem with ESP.

But Linux host and esp32.ko don't even try to send any data. I see it on oscilloscope. So it doesn't matter connected ESP or not at all.

ajita02, 2021-08-19:
@neva-blyad can you share dmesg log and `/boot/config.txt` ? also share ESP side log also for debugging.

neva-blyad, 2021-08-19:
ESP side log:

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7136
load:0x40078000,len:14284
load:0x40080400,len:3528
entry 0x40080650
I (26) boot: ESP-IDF v4.4-dev-2575-g2c49af9e75-dirty 2nd stage bootloader
I (26) boot: compile time 18:28:29
I (28) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1b598h (112024) map
I (151) esp_image: segment 1: paddr=0002b5c0 vaddr=3ffbdb60 size=03b30h ( 15152) load
I (157) esp_image: segment 2: paddr=0002f0f8 vaddr=40080000 size=00f20h (  3872) load
I (159) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=8da74h (580212) map
I (375) esp_image: segment 4: paddr=000bda9c vaddr=40080f20 size=1b0bch (110780) load
I (421) esp_image: segment 5: paddr=000d8b60 vaddr=50000000 size=00010h (    16) load
I (435) boot: Loaded app from partition at offset 0x10000
I (435) boot: Disabling RNG early entropy source...
I (446) cpu_start: Pro cpu up.
I (447) cpu_start: Single core mode
I (455) cpu_start: Pro cpu start user code
I (455) cpu_start: cpu freq: 160000000
I (455) cpu_start: Application information:
I (459) cpu_start: Project name:     network_adapter_0.3
I (465) cpu_start: App version:      release/v0.3-27-gdacac83-dirty
I (472) cpu_start: Compile time:     Aug 12 2021 19:03:40
I (478) cpu_start: ELF file SHA256:  7cb51eb27194cd7e...
I (484) cpu_start: ESP-IDF:          v4.4-dev-2575-g2c49af9e75
I (491) heap_init: Initializing. RAM available for dynamic allocation:
I (498) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
I (504) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (510) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (517) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (523) heap_init: At 3FFC6020 len 00019FE0 (103 KiB): DRAM
I (529) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (535) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (542) heap_init: At 4009BFDC len 00004024 (16 KiB): IRAM
I (549) spi_flash: detected chip: generic
I (552) spi_flash: flash io: dio
I (557) sleep: Configure to isolate all GPIO pins in sleep state
I (563) sleep: Enable automatic switching of GPIO sleep configuration
I (571) cpu_start: Starting scheduler on PRO CPU.
I (576) NETWORK_ADAPTER: *********************************************************************
I (576) NETWORK_ADAPTER:                 ESP-Hosted Firmware version :: 0.3                        
I (586) NETWORK_ADAPTER:                 Transport used :: SPI                           
I (596) NETWORK_ADAPTER: *********************************************************************
I (606) NETWORK_ADAPTER: Supported features are:
I (616) NETWORK_ADAPTER: - WLAN over SPI
I (616) NETWORK_ADAPTER: - BT/BLE
I (626) NETWORK_ADAPTER:    - HCI Over SPI
I (626) NETWORK_ADAPTER:    - BT/BLE dual mode
I (646) BTDM_INIT: BT controller compile version [bfbbe1e]
I (646) system_api: Base MAC address is not set
I (646) system_api: read default base MAC address from EFUSE
I (656) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1226) SPI_DRIVER: Using SPI interface
I (1226) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1226) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1246) NETWORK_ADAPTER: Initial set up done`

dmesg log:
`[69248.745995] device: 'spi3.1': device_add
[69248.750377] bus: 'spi': add device spi3.1
[69248.756274] ESP32 peripheral is registered to SPI bus [3],chip select [1], SPI Clock [10]
[69248.970272] device: 'ethsta0': device_add`

neva-blyad, 2021-08-19:
***Linux kernel .config:**

```
#
# Automatically generated file; DO NOT EDIT.
# Linux/arm 4.19.138 Kernel Configuration
#

#
# Compiler: arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 8.4.0 r0-6c6f99e) 8.4.0
#
CONFIG_CC_IS_GCC=y
CONFIG_GCC_VERSION=80400
CONFIG_CLANG_VERSION=0
CONFIG_CC_HAS_ASM_GOTO=y
CONFIG_IRQ_WORK=y
CONFIG_BUILDTIME_EXTABLE_SORT=y

#
# General setup
#
CONFIG_INIT_ENV_ARG_LIMIT=32
# CONFIG_COMPILE_TEST is not set
CONFIG_LOCALVERSION=""
# CONFIG_LOCALVERSION_AUTO is not set
CONFIG_BUILD_SALT=""
CONFIG_HAVE_KERNEL_GZIP=y
CONFIG_HAVE_KERNEL_LZMA=y
CONFIG_HAVE_KERNEL_XZ=y
CONFIG_HAVE_KERNEL_LZO=y
CONFIG_HAVE_KERNEL_LZ4=y
# CONFIG_KERNEL_GZIP is not set
# CONFIG_KERNEL_LZMA is not set
CONFIG_KERNEL_XZ=y
# CONFIG_KERNEL_LZO is not set
# CONFIG_KERNEL_LZ4 is not set
CONFIG_DEFAULT_HOSTNAME="(none)"
CONFIG_SWAP=y
CONFIG_SYSVIPC=y
CONFIG_SYSVIPC_SYSCTL=y
CONFIG_POSIX_MQUEUE=y
CONFIG_POSIX_MQUEUE_SYSCTL=y
# CONFIG_CROSS_MEMORY_ATTACH is not set
# CONFIG_USELIB is not set
# CONFIG_AUDIT is not set
CONFIG_HAVE_ARCH_AUDITSYSCALL=y

#
# IRQ subsystem
#
CONFIG_GENERIC_IRQ_PROBE=y
CONFIG_GENERIC_IRQ_SHOW=y
CONFIG_GENERIC_IRQ_SHOW_LEVEL=y
CONFIG_GENERIC_IRQ_EFFECTIVE_AFF_MASK=y
CONFIG_HARDIRQS_SW_RESEND=y
CONFIG_GENERIC_IRQ_CHIP=y
CONFIG_IRQ_DOMAIN=y
CONFIG_IRQ_DOMAIN_HIERARCHY=y
CONFIG_HANDLE_DOMAIN_IRQ=y
CONFIG_IRQ_FORCED_THREADING=y
CONFIG_SPARSE_IRQ=y
# CONFIG_GENERIC_IRQ_DEBUGFS is not set
CONFIG_GENERIC_IRQ_MULTI_HANDLER=y
CONFIG_ARCH_CLOCKSOURCE_DATA=y
CONFIG_GENERIC_TIME_VSYSCALL=y
CONFIG_GENERIC_CLOCKEVENTS=y
CONFIG_ARCH_HAS_TICK_BROADCAST=y
CONFIG_GENERIC_CLOCKEVENTS_BROADCAST=y

#
[... 3894 lines cut ...]
# CONFIG_TEST_USER_COPY is not set
# CONFIG_TEST_BPF is not set
# CONFIG_FIND_BIT_BENCHMARK is not set
# CONFIG_TEST_FIRMWARE is not set
# CONFIG_TEST_SYSCTL is not set
# CONFIG_TEST_UDELAY is not set
# CONFIG_TEST_STATIC_KEYS is not set
# CONFIG_TEST_KMOD is not set
# CONFIG_MEMTEST is not set
# CONFIG_BUG_ON_DATA_CORRUPTION is not set
# CONFIG_SAMPLES is not set
CONFIG_HAVE_ARCH_KGDB=y
# CONFIG_KGDB is not set
# CONFIG_UBSAN is not set
CONFIG_ARCH_HAS_DEVMEM_IS_ALLOWED=y
# CONFIG_ARM_PTDUMP_DEBUGFS is not set
# CONFIG_DEBUG_WX is not set
# CONFIG_ARM_UNWIND is not set
# CONFIG_DEBUG_USER is not set
# CONFIG_DEBUG_LL is not set
CONFIG_DEBUG_LL_INCLUDE="mach/debug-macro.S"
CONFIG_UNCOMPRESS_INCLUDE="debug/uncompress.h"
# CONFIG_PID_IN_CONTEXTIDR is not set
# CONFIG_CORESIGHT is not set
```

neva-blyad, 2021-08-19:
[config.txt](https://github.com/espressif/esp-hosted/files/7014023/config.txt)

ajita02, 2021-08-20:
@neva-blyad your dmesg log conveys that you are using SPI bus 3 and chip select 1.  Can you confirm that you have done device tree related changes in `spidev_disabler.dts` ?

neva-blyad, 2021-08-20:
I'm sure spidev driver is disabled and only esp_spi driver is used for SPI bus 3 chip select 1.
I have completely disabled CONFIG_SPI_SPIDEV configure option and rebuild kernel without spidev.

neva-blyad, 2021-08-20:
@ajita02, I've saw the driver sources and found out that /dev/esps0 device is not used by kernel module esp32_spi.ko at all. Python script ap_scan_list.py (and other python scripts) write different commands to it, I thought esp32_spi.ko should resend these commands over SPI bus to ESP32, but, obviously, it doesn't know about /dev/esps0 device file at all.
Have I missed something? How to do simple operations (scan Wi-Fi, for example) over SPI?

mantriyogesh, 2021-08-20:
Hi @neva-blyad ,

I note that you are using SPI3 and chip select 1.

Could you please have a look at SPI related instructions from [porting guide](https://gitlab.espressif.cn:6688/app-frameworks/esp_hosted/-/blob/master/docs/Linux_based_host/porting_guide.md).

Also, As you are not able to get the any signal on SPI GPIO, we should suspect the Device Tree has appropriate changes to instantiate your expected SPI bus, say SPI bus 3 ?

neva-blyad, 2021-08-20:
@mantriyogesh, thank you for your responce.
I have looked esp_spi.c and found out that it doesn't use Device Tree.
So I edited spi_dev_init() function to stick with my settings:

```
	strlcpy(esp_board.modalias, "esp_spi", sizeof(esp_board.modalias));
	esp_board.mode = SPI_MODE_2;
	esp_board.max_speed_hz = spi_clk_mhz * NUMBER_1M;
	esp_board.bus_num = 3;
	esp_board.chip_select = 1;
```
And SPI mutes with this.
Also tried "esp32_spi" modalias string. No difference, SPI mutes.
Also tries "spidev" modalias string. Then I sent some random bytes with spidev_test utility and see on osclilloscope that they really sent. Voltage levels changed during spidev_test -D /dev/spidev3.1 call. So my bus and chip select numbers are right! But esp32_spi.ko doesn't send anything.

I really wonder why it nothing to do with /dev/esps0 file as I see in sources, I though ap_scan_list.py and other Python script write to it some commands but kernel driver did not ever try to read this device file.

mantriyogesh, 2021-08-22:
Little elaborated response below !!

A sample dtsi we modified for Linux other than Raspberry Pi looks like this:
```
//PART 1
&spi_2 {
....
        status = "okay";

        spidev2: spidev@2 {           
            status = "okay";
            compatible = "linux,spidev"; 
            ...
            spi-max-frequency = <...>;
            controller-data { 
                 ...
            };  
        };  
};

//PART 2
&spidev2 {
    status = "disabled";
};
```
In above snippet, `...` represents board specific SPI config, which you can refer from device tree manuals for SPI.

**Please note**, Above snippet depends and varies upon your hardware/SoC.
Please modify referring to other nodes in your board's dtsi/dts file

Here, Essential part is, we had allowed spidev and then disabled it.
Obviously this is one way of doing it, there could be other ways possible too.

We would suggest to do this step wise.
1. Allow spidev and do not disable it (Only add PART 1)
    - Here, Run the test spidev application using your spidev node /dev/spidev**X**.**Y**, where **X** is the SPI bus index and **Y** is the SPI bus chip select.
2. Now add spidev disabler code too (PART 1 & PART 2)
    - Here you can observe `/dev/spidevX.Y` or `/dev/spi*` is **not** present.
    - Now try to load the SPI using esp32_spi.ko kernel module.
    - Basically now kernel module has clear access to SPI bus as spidev is not going to acquire it.

Please check and change other SPI config in device tree which are supported by the SoC and slave device.
_For example_, Max frequency for ESP32 SPI slave should be <= maximum **SPI slave Freq** in respective ESP32 Technical Reference Manual.
PTAL at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_slave.html. This is ESP32 varient specific SPI slave documentation. You can change device type left top combo box to see other chip-set variant to your matching ESP chip-set variant.

SPI freq should be optimized _i.e._ gradually increased from very low to high using file [spi_slave_api.c](https://github.com/espressif/esp-hosted/blob/master/esp/esp_driver/network_adapter/main/spi_slave_api.c).
Please check other SPI config (like SPI mode etc), which should be in sync with SPI master.

Regarding `/dev/esps0` file, this is created using [rpi_init.sh](https://github.com/espressif/esp-hosted/blob/master/host/linux/host_control/rpi_init.sh). We plan to handle this from within kernel module, but as if now you can create minimally required script similar to that of `rpi_init.sh`.
Please be warned, this `rpi_init.sh` has some content that is specific to Raspberry Pi. You might have to port that specific content for your Linux board.

Please do let us know if any difficulties you face.

neva-blyad, 2021-08-23:
@mantriyogesh, thank your for advanced and detailed response.
I did all steps which you've recommended. Here how my dts looks:
```
// PART 1
&ecspi4 {
	...
	cs-gpios = <&gpio2 15 GPIO_ACTIVE_LOW>,
	           <&gpio2 10 GPIO_ACTIVE_LOW>;
	status = "okay";
	...

	//spi-cpha;
	spi-cpol;

	esp32: spi@1 {
		compatible = "spidev";
		reg = <0x01>;
		irq-gpios = <&gpio1 1 GPIO_ACTIVE_LOW>;
		spi-max-frequency = <1000000>;
		status = "okay";
	};
};

// PART 2
&esp32 {
	status = "disabled";
};
```

It is true that file _/dev/spidev3.1_ appears when I commented out the PART 2 snippet. spidev_test utility works fine.
Then I added PART 2 snipped and saw that file _/dev/spidev3.1_ disappears. So code works as needed!

Then I created _/dev/esps0_ and loaded kernel module:
```
insmod esp32.ko
mknod /dev/esps0 c 221 0
chmod 666 /dev/esps0
```

Then I started Python:
```
python ap_scan_list.py
```

And nothing! Oscilloscope shows nothing, empty voltage levels. esp32.ko does send nothing. Should it read file _/dev/esps0_? (I asked you this twice because no code working with _/dev/esps0_ was found by me in esp32 sources.)

neva-blyad, 2021-08-23:
Should I enable interfaces ethsta0 and ethap0 before proceed?

mantriyogesh, 2021-08-23:
First transaction if happens, it will show _similar_ entry like below in /var/log/messages or dmesg
```
Aug 23 14:56:06 raspberrypi kernel: [14073.961927] Received INIT event from ESP32 peripheral
Aug 23 14:56:06 raspberrypi kernel: [14073.961945] EVENT: 2
Aug 23 14:56:06 raspberrypi kernel: [14073.961959] EVENT: 1
Aug 23 14:56:06 raspberrypi kernel: [14073.961971] EVENT: 0
Aug 23 14:56:06 raspberrypi kernel: [14073.961984] ESP peripheral capabilities: 0x3a
```

mantriyogesh, 2021-08-24:
Also please check with different SPI mode like 1 or 3.

neva-blyad, 2021-09-02:
@mahavirj, many thanks for your detailed and advanced response!
Sorry for my late answer.

I have checked everything I can and find out, that I have wrong SPI (and GPIO) pinout.
Model of my SoC is ESP-WROOM-32 and my board is NodeMCU ESP-32S.
These board and chip are not supported by esp-hosted firmware.
I have manually add the following in my spi_slave_api.c:
```
#define ESP_SPI_CONTROLLER 0
#define GPIO_MISO          21
#define GPIO_MOSI          22
#define GPIO_SCLK          20
#define GPIO_CS            19
    
#define DMA_CHAN ESP_SPI_CONTROLLER
```
SPI pins are chosen according to SoC datashit.
But an error appears in console output:

```
E (1216) gpio: gpio_set_pull_mode(241): GPIO number error
E (1226) spi: SPI1 already claimed by <null>.
E (1226) spi_slave: spi_slave_initialize(129): host already in use
```

Do you have any thoughts why it is occured, and how it could be fixed?

mantriyogesh, 2021-09-08:
Hello @neva-blyad 

Actually only HSPI and VSPI are available at GPIO.
I think you have got these numbers from similar documentation to this:
[image]
But be cautious the pin numbers are just for reference, actual pin names are the real GPIO used.

Why you cannot use this SPI, please refer this: https://docs.espressif.com/projects/esp-idf/en/release-v4.0/api-reference/peripherals/spi_slave.html#overview

These are reserved for other purpose.

Regarding your query, 
```
These board and chip are not supported by esp-hosted firmware.
```
Could you please brief if any technical difficulty you face while using one of these SPI peripherals ***at ESP32*** side?

In case of HSPI and VSPI, the pins which would give you optimum performace are IO_MUX pins.
Please check them in table at, https://docs.espressif.com/projects/esp-idf/en/release-v4.0/api-reference/peripherals/spi_slave.html#gpio-matrix-and-iomux

You can also freely choose other `GPIO matrix` pins as well in case above pins are not free for your use.
But SPI bus can only be used over GPIO is restricted to `HSPI` / `VSPI`

mantriyogesh, 2021-09-23:
Hello @neva-blyad 
Is there any update on this issue?

neva-blyad, 2021-09-30:
@mantriyogesh many thanks for your reply!
I've checked my pinout again and ensured that it is OK.
I think there is another hardware problem but don't know which exactly.
I think I was wrong with something so it is not your trouble and I'll close this issue.

closed 2021-09-30 by neva-blyad

mantriyogesh, 2021-09-30:
Thank you @neva-blyad.
Glad that you could fix the issue.
If it is possible, we would really like to know the problem.
We all learn from the mistakes and will help some one else facing same type of issue.
Thanks again.

neva-blyad, 2021-10-05:
Hi, @mantriyogesh again!
I am very happy because ESP32 board now works correctly!
I tried it with another Linux board, carefully rechecked all pin connections, wire the both boards with shared ground and now it can scan Wi-Fi networks. The problem was I determined HANDSHAKE_PIN and SPI_DATA_READY_PIN in file _esp_spi.h_ wrongly.
I should say thank you for all of your responses! I have learned ESP32 software & firmware more deeply.