espressif/esp-hosted #46: esp_serial_write not returning error upon packet send failure

dwhitters, 2021-08-23:
# Problem
The return value of **esp_send_packet** is ignored, and instead is used to print a kernel error "Failed to transmit data".

# Steps to Reproduce
There are a few but writing a packet with a length greater than the SDIO transport length will cause a failure.

# Proposed Solution
The return value of this function should be negative on failure as it is mapped to the linux "write" operation.

" A negative value means an error occurred; like for read, valid error values are those defined in <linux/errno.h>. " - [OReilly](https://www.oreilly.com/library/view/linux-device-drivers/0596000081/ch03s08.html)

MangeshMalusare, 2021-08-24:
Thanks for the input. We will update it.

ajita02, 2021-09-02:
Hi @dwhitters ,
Appreciate your input. Your suggestion is updated in latest master. 

Thanks,
Ajita

davewhitters, 2022-04-22:
Hi @ajita02,

What commit was this update applied in?

Thanks,
Dave

ajita02, 2022-04-25:
Hi @dwhitters,

Commit ID `c4767da6c10dad1c637644dabe6c507c2f4eed6a` fixed the issue, but recent  changes for fragmentation over same lines may have broken expected return value from `esp_serial_write` function. We are looking into this issue.

Thanks,
Ajita

mantriyogesh, 2026-09-28:
Closing this as a stale issue; please reopen it if you continue to experience the problem.

closed 2026-09-28 by mantriyogesh