espressif/esp-hosted #48: SPI Interface Ready Pin not set

starkeeper82, 2021-08-25:
During investigation of my Handsahke pin problems I stumbled across some weired ready pin behavior. In some cases the the ESP does not set the ready pin while the tx_task is waiting for a possibility to transmit.

I think there are at least two possible ways of reproducing this.

1.
In case a real packet was transmitted and the callback function "spi_post_trans_cb" will clear the ready pin. The pin will not be set again if the task "spi_transaction_tx_task" is already waiting for the semaphore "spi_sema". This is because the "tx_task" sets the ready pin before waiting for "spi_sema".

2. 
This case might be wrong but I think it can happen too. The task "spi_transaction_tx_task" is working and gets interrupted by the function "spi_post_trans_cb" (which is called by a ISR) or any other task (system is preemptive). The interrrupt in the tx-task happens between the setting of the ready pin and the wait for the semaphore "spi_sema", but before waiting for the semephore.
The ISR callback "spi_post_trans_cb" will reset the ready pin (in case a real packet was transmitted) and signalize the semaphore "spi_sema". Instead of returning to the task "spi_transaction_tx_task" FreeRTOS decides to schedule "spi_transaction_post_process_task" which was recently triggered by the end of the transmission. In this case the task "spi_transaction_tx_task" will get the semaphore "spi_sema" and it will schedule a dummy packet because the queue "spi_tx_queue" is empty (the last queue entry is still stuck in the processing of the task "spi_transaction_tx_task").
With this scenario the task "spi_transaction_tx_task" will wait for the semaphore "spi_sema" without the ready pin set, while a dummy packet is scheduled.

I wonder if the design with two tasks "spi_transaction_tx_task" and "spi_transaction_post_process_task" is really a good choice. Synchronizing both tasks to schedule the correct packets and managing the ready pin from three different locations ("spi_transaction_tx_task", "spi_transaction_post_process_task" and "spi_post_trans_cb") is quite complicated. I think of a single tx-task that waits for the end of spi transaction event, then looks (non blocking) into the tx-queue and schedules a packet or a dummy.

Any opinions on that?

starkeeper82, 2021-08-30:
I did some coding and testing and came up with a solution that fixes all issues for me. The solution has only a single task, no semphore and no dummy_queued flag. 
Commited under this id: 60d2ae58b9e34d98e37bfe1568580137c502052c

With these changes I can use the ESP32-S2 as slave together with a STM32H7 and DMA. My SPI frequency is 25Mhz and for now I do not get lock ups or such things anymore. I am running lwip on the STM32H7 and even transfer of big data chunks runs without the need for any retransmission or something like this on TCP connections.

MangeshMalusare, 2021-08-30:
Hello @starkeeper82,
Thank you for your contribution! We will check your code and get back to you!

Regards,
Mangesh

starkeeper82, 2021-09-22:
Is there any progress on this topic?

MangeshMalusare, 2021-09-22:
Hi @starkeeper82, we are evaluating your commit.. we will get back soon.. Thanks!

alireza3137, 2021-09-27:
@starkeeper82 
Hi 
What is the transfer speed in your porting LWIP with Your code?

starkeeper82, 2021-09-28:
Hi @alireza3137 
until now I do not know the exact speed of my imlpementation. But this is already a task that I will try to solve within the next weeks. As soon as I have results, I will give you an update.
I think to get some reliable results I will have to write a dedicated task on the controller because otherwise the file system or the HTTP server would have an impact on the measurements.

mantriyogesh, 2021-10-06:
Hello @starkeeper82 

thank you for all your time and the contributions to make this project better.
Could you please let us know your mail id for credits ASAP?
We are integrating your code in the mainstream.

~ Yogesh

starkeeper82, 2021-10-06:
Hi @mantriyogesh 
nice to hear that my code is assumed to be useful 👍 
I not sure what you mean by "email id", but probably the github replacement for my priavte email?

mantriyogesh, 2021-10-06:
the mail id by which the open source contribution you would like to see by.
Generally it is public domain mail id, which will be visible on web (or search engines).

starkeeper82, 2021-10-06:
Ah ok, I was thinking too complicated..
starkeeper@gmx.de

alireza3137, 2021-11-20:
Hi @starkeeper82
Have you measeared the speed of TCP protocol?