espressif/esp-hosted #50: stm32 cubemx project issue

xiongyu0523, 2021-08-31:
I follow document to replicate stm32cube project. Finally I can build a binary but it seems like the project does not have any ESP-Hosted files within it. What steps are missing?

[image]

ajita02, 2021-08-31:
Hi @xiongyu0523, by looking at your .project file it seems that you have missed 

- $ cd </path/to/esp_hosted>/host/stm32/proj
- $ bash ./prepare_project.sh </path/to/workspace_directory>

these steps for linux machine. 
please follow all steps mentioned at [this](https://github.com/espressif/esp-hosted/blob/master/docs/MCU_based_host/Getting_started.md#11-start-project-with-stm32) link for successful ESP-Hosted build.

Thanks,
Ajita

xiongyu0523, 2021-08-31:
> Hi @xiongyu0523, by looking at your .project file it seems that you have missed
>
> * $ cd </path/to/esp_hosted>/host/stm32/proj
> * $ bash ./prepare_project.sh </path/to/workspace_directory>
>
> these [...]

Yes. I have done this step (on windows use powershell) and the script told me sth like 'success... open cube project again'

ajita02, 2021-08-31:
@xiongyu0523, we will check the issue and get back to it.

Thanks,
Ajita

xiongyu0523, 2021-08-31:
Just FYI,  I'm using the latest build, I just run git pull before start to build my project. My CubeIDE is 1.6.1

xiongyu0523, 2021-09-06:
Any update? I follow the guide to setup environment again on another PC, but it looks the same. 

I can't see esp-hosted files added to my project, or we do not need them to be added into stm32cubeide project? 

I tried to build the project, it prompts 'undefined reference to `MX_FREERTOS_Init', because esp-hosted.c is not in build scope.

Should I connect hardware to start build (I don't have the exact hardware, just want to have a good build to understand code

xiongyu0523, 2021-09-06:
I can fix by manually adding 3 link folders to esp-hosted project. 

Not sure if it is related the value of `CODE_BASE` in Linked Resources section was defined to `C\Users\Neo\Downloads\esp-hosted`, should be `C:\Users\Neo\Downloads\esp-hosted`?

ajita02, 2021-09-06:
Hi @xiongyu0523 , Can you check CODE_BASE variable points to correct ESP-Hosted code base? please follow this [Link](https://github.com/espressif/esp-hosted/blob/master/docs/MCU_based_host/Getting_started.md#note).  If issue is still persist, let us know. 

Thanks,
Ajita

ajita02, 2021-09-06:
> I can fix it by manually adding 3 link folders to esp-hosted project. Then I can build the project
> 
> Not sure if it is related the value of `CODE_BASE` in Linked Resources section was started with to `C\<path-to-esp-hosted>`, should be `C:\<path-to-esp-hosted>`?

@xiongyu0523 , Thanks for input. It should be `C:\<path-to-esp-hosted>`. We will fix it.

Thanks,
Ajita

xiongyu0523, 2021-09-07:
Even I change it to `C:` after the patch run, it seems it does not work automatically. Would you please verify the whole process on Windows 10?

closed 2021-09-09 by xiongyu0523

mluisPlux, 2021-09-16:
I'm having the same issue how did you fix it?

HikingDev, 2021-10-13:
I have followed the steps and the problem still persists. 
I have followed the instructions and the paths were successfully added as well as he CODE_BASE variable fixed.
In the project Includes i can see the esp-hosted sources but MX_FREERTOS_Init and vApplicationGetIdleTaskMemory are not found in linking process.
@xiongyu0523 how did you fix it?
Why was the issue closed? On windows it still does not work @ajita02

mantriyogesh, 2021-10-13:
Hello Benjamin,
 
As original author has closed the issue, we think it is worth to retry the steps from documentation.
Could you please us know which step do you face the problem.
Also did you apply `prepare_project.bat` as suggested?

Also we encourage you to file separate issue if **you** feel the issue is different.