espressif/esp-hosted #51: Provide an OS abstraction layer for MCU host?

xiongyu0523, 2021-09-01:
To me, this project is very valuable for projects need higher throughput.. I'd like to port to other MCU and other RTOS. But currently there is almost no abstraction for OS, the platform I'm working on is Azure RTOS ThreadX, which I need to modify the `official` code a lot.. 

Do you consider to have OS abstraction layer so that we can leverage your code as much as possible? (Better if you can provide MCU abstraction as well since I want my solution can work with various of vendors)

ajita02, 2021-09-02:
Hi @xiongyu0523,
Right now, OS abstraction layer for MCU is not in project pipeline. But we will revert back to this suggestion soon. Appreciate your patience.

Thanks,
Ajita

ameyinamdar, 2021-09-02:
Hi @xiongyu0523. MCUs not only vary in terms of RTOS that they can run, but also as there is no standardisation for the peripheral drivers (SPI/SDIO/UART) that is used for transport, it becomes difficult to provide MCU side network interface driver. To make this task simple, we provide STM32 netif implementation that uses ST's XCube SDK. This may be a good starting point for supporting any other MCU. Unfortunately we don't see an easy way to address this generically.

xiongyu0523, 2021-09-03:
Few suggestions and want to have a discussion:

OS abstraction:

The code is mixing raw FreeRTOS API and cmsis-os v2 API. I'd suggest to only use cmsis-os v2 API for the first step so it can be a OS agnostic solution. Considering the complexity of cmsis-os v2 API surface, if there is no official conversion layer available, it would be significant efforts for developer to implement, 

HW abstraction:

Direct use of ST HAL driver and interrupt is making this project hard to port, at least I need to have deeper understand on what the library is going to do to finish a new device port. It is better to have one or two port surface so people can totally treat your library as a blackbox and all they need to do is replace those hardware operations. 

Integration to network stack 

Everyone use this library is going to connect their TCPIP stack, at least one end-to-end demo is appreciated