espressif/esp-hosted #54: build esp firmware issue

xiongyu0523, 2021-09-07:
I have installed esp-idf toolchain (v4.3) in visual studio code. When I try to build this project, I got below errors:

Steps:
1. open `C:\Users\yuxiong\Documents\GitHub\esp-hosted\esp\esp_driver\network_adapter` folder by vs code.
2. ESP-IDF: Set Espressif device target to ESP-S2
3. ESP-IDF: SDK Configuration editor, confirm I'm using SPI transport
4. ESP-IDF: Build your project

What is missing? Is there any dependency to esp-idf?

```
[0/1] Re-running CMake...
*************************************************************************************
                    Building ESP-Hosted Firmware version :: 0.3
*************************************************************************************
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32s2
-- Project sdkconfig file C:/Users/yuxiong/Documents/GitHub/esp-hosted/esp/esp_driver/network_adapter/sdkconfig
Loading defaults file C:/Users/yuxiong/Documents/GitHub/esp-hosted/esp/esp_driver/network_adapter/sdkconfig.defaults...
C:/Users/yuxiong/Documents/GitHub/esp-hosted/esp/esp_driver/network_adapter/sdkconfig.defaults:10 CONFIG_BTDM_CONTROLLER_MODE_BTDM was replaced with CONFIG_BTDM_CTRL_MODE_BTDM
C:/Users/yuxiong/Documents/GitHub/esp-hosted/esp/esp_driver/network_adapter/sdkconfig.defaults:11 CONFIG_BTDM_CONTROLLER_HCI_MODE_VHCI was replaced with CONFIG_BTDM_CTRL_HCI_MODE_VHCI
C:\Users\yuxiong\AppData\Local\Temp\confgen_tmp5_ign240:4 line was updated to CONFIG_SDIO_DAT2_DISABLED=n
C:\Users\yuxiong\AppData\Local\Temp\confgen_tmp5_ign240:9 line was updated to CONFIG_BT_BLUEDROID_ENABLED=n
C:\Users\yuxiong\AppData\Local\Temp\confgen_tmp5_ign240:14 line was updated to CONFIG_ESP32_WIFI_NVS_ENABLED=n
-- Could NOT find Perl (missing: PERL_EXECUTABLE) 
-- App "network_adapter" version: release/v0.3-37-g08a6de0
-- Adding linker script C:/Users/yuxiong/esp/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.ld
-- Adding linker script C:/Users/yuxiong/esp/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.api.ld
-- Adding linker script C:/Users/yuxiong/esp/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.libgcc.ld
-- Adding linker script C:/Users/yuxiong/esp/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-funcs.ld
-- Adding linker script C:/Users/yuxiong/esp/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-data.ld
-- Adding linker script C:/Users/yuxiong/esp/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.spiflash.ld
-- Adding linker script C:/Users/yuxiong/Documents/GitHub/esp-hosted/esp/esp_driver/network_adapter/build/esp-idf/esp32s2/esp32s2_out.ld
-- Adding linker script C:/Users/yuxiong/esp/esp-idf/components/esp32s2/ld/esp32s2.project.ld.in
-- Adding linker script C:/Users/yuxiong/esp/esp-idf/components/esp32s2/ld/esp32s2.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32s2 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_local_ctrl esp_netif esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb touch_element ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa    
-- Component paths: C:/Users/yuxiong/esp/esp-idf/components/app_trace C:/Users/yuxiong/esp/esp-idf/components/app_update C:/Users/yuxiong/esp/esp-idf/components/asio C:/Users/yuxiong/esp/esp-idf/components/bootloader C:/Users/yuxiong/esp/esp-idf/components/bootloader_support C:/Users/yuxiong/esp/esp-idf/components/bt C:/Users/yuxiong/esp/esp-idf/components/cbor C:/Users/yuxiong/esp/esp-idf/components/cmock C:/Users/yuxiong/esp/esp-idf/components/coap C:/Users/yuxiong/esp/esp-idf/components/console C:/Users/yuxiong/esp/esp-idf/components/cxx C:/Users/yuxiong/esp/esp-idf/components/driver C:/Users/yuxiong/esp/esp-idf/components/efuse C:/Users/yuxiong/esp/esp-idf/components/esp-tls C:/Users/yuxiong/esp/esp-idf/components/esp32s2 C:/Users/yuxiong/esp/esp-idf/components/esp_adc_cal C:/Users/yuxiong/esp/esp-idf/components/esp_common C:/Users/yuxiong/esp/esp-idf/components/esp_eth C:/Users/yuxiong/esp/esp-idf/components/esp_event C:/Users/yuxiong/esp/esp-idf/components/esp_gdbstub C:/Users/yuxiong/esp/esp-idf/components/esp_hid C:/Users/yuxiong/esp/esp-idf/components/esp_http_client C:/Users/yuxiong/esp/esp-idf/components/esp_http_server C:/Users/yuxiong/esp/esp-idf/components/esp_https_ota C:/Users/yuxiong/esp/esp-idf/components/esp_https_server C:/Users/yuxiong/esp/esp-idf/components/esp_hw_support C:/Users/yuxiong/esp/esp-idf/components/esp_ipc C:/Users/yuxiong/esp/esp-idf/components/esp_local_ctrl C:/Users/yuxiong/esp/esp-idf/components/esp_netif C:/Users/yuxiong/esp/esp-idf/components/esp_pm C:/Users/yuxiong/esp/esp-idf/components/esp_ringbuf C:/Users/yuxiong/esp/esp-idf/components/esp_rom C:/Users/yuxiong/esp/esp-idf/components/esp_serial_slave_link C:/Users/yuxiong/esp/esp-idf/components/esp_system C:/Users/yuxiong/esp/esp-idf/components/esp_timer 
C:/Users/yuxiong/esp/esp-idf/components/esp_websocket_client C:/Users/yuxiong/esp/esp-idf/components/esp_wifi C:/Users/yuxiong/esp/esp-idf/components/espcoredump C:/Users/yuxiong/esp/esp-idf/components/esptool_py C:/Users/yuxiong/esp/esp-idf/components/expat C:/Users/yuxiong/esp/esp-idf/components/fatfs C:/Users/yuxiong/esp/esp-idf/components/freemodbus C:/Users/yuxiong/esp/esp-idf/components/freertos C:/Users/yuxiong/esp/esp-idf/components/hal C:/Users/yuxiong/esp/esp-idf/components/heap C:/Users/yuxiong/esp/esp-idf/components/idf_test C:/Users/yuxiong/esp/esp-idf/components/jsmn C:/Users/yuxiong/esp/esp-idf/components/json C:/Users/yuxiong/esp/esp-idf/components/libsodium C:/Users/yuxiong/esp/esp-idf/components/log C:/Users/yuxiong/esp/esp-idf/components/lwip C:/Users/yuxiong/Documents/GitHub/esp-hosted/esp/esp_driver/network_adapter/main C:/Users/yuxiong/esp/esp-idf/components/mbedtls C:/Users/yuxiong/esp/esp-idf/components/mdns C:/Users/yuxiong/esp/esp-idf/components/mqtt C:/Users/yuxiong/esp/esp-idf/components/newlib C:/Users/yuxiong/esp/esp-idf/components/nghttp C:/Users/yuxiong/esp/esp-idf/components/nvs_flash C:/Users/yuxiong/esp/esp-idf/components/openssl C:/Users/yuxiong/esp/esp-idf/components/partition_table C:/Users/yuxiong/esp/esp-idf/components/perfmon C:/Users/yuxiong/esp/esp-idf/components/protobuf-c C:/Users/yuxiong/esp/esp-idf/components/protocomm C:/Users/yuxiong/esp/esp-idf/components/pthread C:/Users/yuxiong/esp/esp-idf/components/sdmmc C:/Users/yuxiong/esp/esp-idf/components/soc C:/Users/yuxiong/esp/esp-idf/components/spi_flash C:/Users/yuxiong/esp/esp-idf/components/spiffs C:/Users/yuxiong/esp/esp-idf/components/tcp_transport C:/Users/yuxiong/esp/esp-idf/components/tcpip_adapter C:/Users/yuxiong/esp/esp-idf/components/tinyusb C:/Users/yuxiong/esp/esp-idf/components/touch_element C:/Users/yuxiong/esp/esp-idf/components/ulp C:/Users/yuxiong/esp/esp-idf/components/unity C:/Users/yuxiong/esp/esp-idf/components/vfs C:/Users/yuxiong/esp/esp-idf/components/wear_levelling C:/Users/yuxiong/esp/esp-idf/components/wifi_provisioning C:/Users/yuxiong/esp/esp-idf/components/wpa_supplicant C:/Users/yuxiong/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/yuxiong/Documents/GitHub/esp-hosted/esp/esp_driver/network_adapter/build
[3/1007] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[422/1007] Performing configure step for 'bootloader'
-- Found Git: C:/Users/yuxiong/.espressif/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1")
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
[... cut ...]
ccache: error: Failed to create temporary file for esp-idf\libsodium\CMakeFiles\__idf_libsodium.dir\libsodium\src\libsodium\crypto_pwhash\scryptsalsa208sha256\sse\pwhash_scryptsalsa208sha256_sse.c.obj.d: No such file or directory
[803/1007] Building C object esp-idf/freemodbus/CMakeFiles/__idf_freemodbus.dir/common/esp_modbus_slave.c.obj
ninja: build stopped: subcommand failed.
The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake --build ." terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

```

xiongyu0523, 2021-09-08:
Edit error logs, it is consistent on two PCs..

```
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/protocomm_pserial.c.obj 
ccache C:\Users\Neo\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin\xtensa-esp32s2-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DPROJECT_VERSION=0.3 -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../main -IC:/Users/Neo/Downloads/esp-hosted/common/include -IC:/Users/Neo/esp/esp-idf/components/newlib/platform_include -IC:/Users/Neo/esp/esp-idf/components/freertos/include -IC:/Users/Neo/esp/esp-idf/components/freertos/port/xtensa/include -IC:/Users/Neo/esp/esp-idf/components/esp_hw_support/include -IC:/Users/Neo/esp/esp-idf/components/esp_hw_support/port/esp32s2/. -IC:/Users/Neo/esp/esp-idf/components/esp_hw_support/port/esp32s2/private_include -IC:/Users/Neo/esp/esp-idf/components/heap/include -IC:/Users/Neo/esp/esp-idf/components/log/include -IC:/Users/Neo/esp/esp-idf/components/lwip/include/apps -IC:/Users/Neo/esp/esp-idf/components/lwip/include/apps/sntp -IC:/Users/Neo/esp/esp-idf/components/lwip/lwip/src/include -IC:/Users/Neo/esp/esp-idf/components/lwip/port/esp32/include -IC:/Users/Neo/esp/esp-idf/components/lwip/port/esp32/include/arch -IC:/Users/Neo/esp/esp-idf/components/soc/include -IC:/Users/Neo/esp/esp-idf/components/soc/esp32s2/. -IC:/Users/Neo/esp/esp-idf/components/soc/esp32s2/include -IC:/Users/Neo/esp/esp-idf/components/hal/esp32s2/include -IC:/Users/Neo/esp/esp-idf/components/hal/include -IC:/Users/Neo/esp/esp-idf/components/esp_rom/include -IC:/Users/Neo/esp/esp-idf/components/esp_rom/esp32s2 -IC:/Users/Neo/esp/esp-idf/components/esp_common/include -IC:/Users/Neo/esp/esp-idf/components/esp_system/include -IC:/Users/Neo/esp/esp-idf/components/esp32s2/include -IC:/Users/Neo/esp/esp-idf/components/driver/include -IC:/Users/Neo/esp/esp-idf/components/driver/esp32s2/include -IC:/Users/Neo/esp/esp-idf/components/esp_ringbuf/include -IC:/Users/Neo/esp/esp-idf/components/efuse/include -IC:/Users/Neo/esp/esp-idf/components/efuse/esp32s2/include -IC:/Users/Neo/esp/esp-idf/components/xtensa/include -IC:/Users/Neo/esp/esp-idf/components/xtensa/esp32s2/include -IC:/Users/Neo/esp/esp-idf/components/espcoredump/include -IC:/Users/Neo/esp/esp-idf/components/esp_timer/include -IC:/Users/Neo/esp/esp-idf/components/esp_ipc/include -IC:/Users/Neo/esp/esp-idf/components/esp_pm/include -IC:/Users/Neo/esp/esp-idf/components/vfs/include -IC:/Users/Neo/esp/esp-idf/components/esp_wifi/include -IC:/Users/Neo/esp/esp-idf/components/esp_wifi/esp32s2/include -IC:/Users/Neo/esp/esp-idf/components/esp_event/include -IC:/Users/Neo/esp/esp-idf/components/esp_netif/include -IC:/Users/Neo/esp/esp-idf/components/esp_eth/include -IC:/Users/Neo/esp/esp-idf/components/tcpip_adapter/include -IC:/Users/Neo/esp/esp-idf/components/app_trace/include -IC:/Users/Neo/esp/esp-idf/components/mbedtls/port/include -IC:/Users/Neo/esp/esp-idf/components/mbedtls/mbedtls/include -IC:/Users/Neo/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -IC:/Users/Neo/esp/esp-idf/components/app_update/include -IC:/Users/Neo/esp/esp-idf/components/spi_flash/include -IC:/Users/Neo/esp/esp-idf/components/bootloader_support/include -IC:/Users/Neo/esp/esp-idf/components/nvs_flash/include -IC:/Users/Neo/esp/esp-idf/components/pthread/include -IC:/Users/Neo/esp/esp-idf/components/esp_gdbstub/include -IC:/Users/Neo/esp/esp-idf/components/esp_gdbstub/xtensa -IC:/Users/Neo/esp/esp-idf/components/esp_gdbstub/esp32s2 -IC:/Users/Neo/esp/esp-idf/components/wpa_supplicant/include -IC:/Users/Neo/esp/esp-idf/components/wpa_supplicant/port/include -IC:/Users/Neo/esp/esp-idf/components/wpa_supplicant/include/esp_supplicant -IC:/Users/Neo/esp/esp-idf/components/asio/asio/asio/include -IC:/Users/Neo/esp/esp-idf/components/asio/port/include -IC:/Users/Neo/esp/esp-idf/components/cbor/port/include -IC:/Users/Neo/esp/esp-idf/components/unity/include -IC:/Users/Neo/esp/esp-idf/components/unity/unity/src -IC:/Users/Neo/esp/esp-idf/components/cmock/CMock/src -IC:/Users/Neo/esp/esp-idf/components/coap/port/include -IC:/Users/Neo/esp/esp-idf/components/coap/port/include/coap -IC:/Users/Neo/esp/esp-idf/components/coap/libcoap/include -IC:/Users/Neo/esp/esp-idf/components/coap/libcoap/include/coap2 -IC:/Users/Neo/esp/esp-idf/components/console -IC:/Users/Neo/esp/esp-idf/components/nghttp/port/include -IC:/Users/Neo/esp/esp-idf/components/nghttp/nghttp2/lib/includes -IC:/Users/Neo/esp/esp-idf/components/esp-tls -IC:/Users/Neo/esp/esp-idf/components/esp-tls/esp-tls-crypto -IC:/Users/Neo/esp/esp-idf/components/esp_adc_cal/include -IC:/Users/Neo/esp/esp-idf/components/esp_hid/include -IC:/Users/Neo/esp/esp-idf/components/tcp_transport/include -IC:/Users/Neo/esp/esp-idf/components/esp_http_client/include -IC:/Users/Neo/esp/esp-idf/components/esp_http_server/include -IC:/Users/Neo/esp/esp-idf/components/esp_https_ota/include -IC:/Users/Neo/esp/esp-idf/components/protobuf-c/protobuf-c -IC:/Users/Neo/esp/esp-idf/components/protocomm/include/common -IC:/Users/Neo/esp/esp-idf/components/protocomm/include/security -IC:/Users/Neo/esp/esp-idf/components/protocomm/include/transports -IC:/Users/Neo/esp/esp-idf/components/mdns/include -IC:/Users/Neo/esp/esp-idf/components/esp_local_ctrl/include -IC:/Users/Neo/esp/esp-idf/components/sdmmc/include -IC:/Users/Neo/esp/esp-idf/components/esp_serial_slave_link/include -IC:/Users/Neo/esp/esp-idf/components/esp_websocket_client/include -IC:/Users/Neo/esp/esp-idf/com [... cut ...] ut-set-variable -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.3-dirty\" -DESP_PLATFORM -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/protocomm_pserial.c.obj -MF esp-idf\main\CMakeFiles\__idf_main.dir\protocomm_pserial.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/protocomm_pserial.c.obj   -c ../main/protocomm_pserial.c
../main/protocomm_pserial.c:24:10: fatal error: protocomm_priv.h: No such file or directory
 #include <protocomm_priv.h>
          ^~~~~~~~~~~~~~~~~~
```

ajita02, 2021-09-08:
> Edit error logs, it is consistent on two PCs..
>
> ```
> FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/protocomm_pserial.c.obj
> ccache [...]

Hi @xiongyu0523, you are missing this [step](https://github.com/espressif/esp-hosted/blob/master/docs/MCU_based_host/MCU_based_readme.md#22-setup). Please follow setup document to avoid errors.

Thanks,
Ajita

xiongyu0523, 2021-09-09:
Thanks. I missed this step.

closed 2021-09-09 by xiongyu0523