espressif/esp-hosted #57: Is there a mechanism to get control path notification?

xiongyu0523, 2021-09-13:
STATION_DISCONNECTED is currently implemented, while STATION_CONNECTED is actually the return of sync call.  
Is there an async notification mechanism to tell user that a device is disconnected, or a device is connected to ap?

ajita02, 2021-09-16:
Hi @xiongyu0523 , 
You can use [wifi_get_ap_config](https://github.com/espressif/esp-hosted/blob/master/host/linux/host_control/python_support/commands.py#L219) function. If station is connected to AP, it will return ssid, bssid, channel, rssi, ecn. If its not connected AP, it will print "not_connected". There is no async notification mechanism as such.

Thanks,
Ajita

xiongyu0523, 2021-09-16:
This means host must have a mechanism to poll for WIFI connection status periodically. Normally this should be a notification & callback. Can you add this feature which is essential to manage link over serial interface?