espressif/esp-hosted #61: Empty buffers on init sequence.MCU Host demo it's stuck

mluisPlux, 2021-09-23:
I'm trying out your MCU Host demo and it's stuck here:
[image]
 
I'm using a ESP32-C3 and a ST32F7 board.
I'm doing the debug using the Cube IDE and the buffers being sent are empty.
[image]
Do you know what might be happening?

mluisPlux, 2021-09-24:
I was able to pass this point but it's not being able to send/read commands. 
[image]

starkeeper82, 2021-09-28:
Have you used the correct pins? Pinning on C3 and S2 is different.

mantriyogesh, 2021-09-28:
Hello @mluisPlux ,

tx buffer should not be anytime NULL.
They can have 'dummy' transaction but always some valid allocated memory.
Can you please check if the malloc is failed / heap is insufficient? Can increase the heap in that case.

We will also try to reproduce this issue at our end.
Will update further.

mluisPlux, 2021-09-28:
@mantriyogesh @starkeeper82 Hello,
In the ESP32-C3 i use the recommended pins, in the STM32 i had to make a custom configuration for my board.
The heap at the ST is at the max size.  I'm able to receive the first message on the ST with if_type 4 and it initializes the control path. But then the HANDSHAKE and DATA_READY pins go LOW and stay that way so no more transactions are made an i can't pass the point showed on my last comment. https://github.com/espressif/esp-hosted/issues/61#issuecomment-926521137

mantriyogesh, 2021-10-06:
Hello @mluisPlux 
I had tested over commit 
08a6de0ac9db898839eb8f06a102fe6ad49b5887

C3 is working fine.
Could you please check if SPI connections are correct?

mluisPlux, 2021-10-06:
I will try this out. But that commit is related to Linux based host, i am working on a STM32H7 MCU host.

mantriyogesh, 2021-10-06:
Any commit before 54f5d2e47d0c5a837e26a613cdf590cf94d9ded9 you can use.