espressif/esp-hosted #73: sdio_reset 接口实现是不是错了

zh7152106, 2021-11-04:
while(1) {
		sdio_slave_buf_handle_t handle = NULL;

		/* Return buffers to driver */
		ret = sdio_slave_send_get_finished(&handle, 0);
		if (ret != ESP_OK)
			break;

		if (handle) {
			ret = sdio_slave_recv_load_buf(handle);
			ESP_ERROR_CHECK(ret);
		}
	}
一reset就会跑死

zh7152106, 2021-11-04:
下面的地方注释掉会解决复位问题，但我不明确是否会遗漏一些资源
       if (handle) {
		ret = sdio_slave_recv_load_buf(handle);
		ESP_ERROR_CHECK(ret);
	}

mantriyogesh, 2021-12-02:
The intention of this code is to remove any stale data pending at receiver.
In general, it is good practice to remove any pending buffer.
However if you are blocked, it is okay to comment as you have done already.

By the time, we will look into this issue.
Can you please provide ESP log and host dmesg log when you encounter this issue?

mantriyogesh, 2022-05-09:
Hello @zh7152106 
Can we please close this issue, if no more experienced?