espressif/esp-hosted #74: SPI never receives INIT event

rforro, 2021-11-04:
I'm trying to connect ESP32-WROOM-32U to Raspberry Pi 3 (5.10.63-v7+) over SPI, but without success. I've tried SDIO communication, which was possible only one way because of missing Pullups and also tried BT over UART with success. But this is not the issue because I need SPI.

I've followed the [SPI guide](https://github.com/espressif/esp-hosted/blob/master/docs/Linux_based_host/SPI_setup.md), tried 2 different ESP32 boards flashed with pre-built v0.3 and also v0.2 version but nothing. I can observe that the board is being rebooted after executing `./rpi_init.sh spi` but it never receives the INIT event from esp32.

It stays like this for ever:

```
[  106.532552] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
[  106.660189] Bluetooth: Core ver 2.22
[  106.660284] NET: Registered protocol family 31
[  106.660291] Bluetooth: HCI device and connection manager initialized
[  106.660311] Bluetooth: HCI socket layer initialized
[  106.660323] Bluetooth: L2CAP socket layer initialized
[  106.660341] Bluetooth: SCO socket layer initialized
[  133.208991] esp32_spi: loading out-of-tree module taints kernel.
[  133.210223] esp_reset, ESP32: Resetpin of Host is 6
[  133.210349] esp_reset, ESP32: Triggering ESP reset.
[  133.210977] ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]
```

You can see that ESP32 boots without an issue:

```
I (575) NETWORK_ADAPTER: *********************************************************************
I (575) NETWORK_ADAPTER:                 ESP-Hosted Firmware version :: 0.3
I (585) NETWORK_ADAPTER:                 Transport used :: SPI
I (595) NETWORK_ADAPTER: *********************************************************************
I (605) NETWORK_ADAPTER: Supported features are:
I (615) NETWORK_ADAPTER: - WLAN over SPI
I (615) NETWORK_ADAPTER: - BT/BLE
I (625) NETWORK_ADAPTER:    - HCI Over SPI
I (625) NETWORK_ADAPTER:    - BT/BLE dual mode
I (645) BTDM_INIT: BT controller compile version [4bc0a8e]
I (645) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (745) phy: phy_version: 4390, 6b3c1f2, Sep 10 2020, 15:09:07, 0, 0
I (1215) SPI_DRIVER: Using SPI interface
I (1215) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1215) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1245) wifi:wifi driver task: 3ffd6e24, prio:23, stack:6656, core=0
I (1245) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1245) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1255) wifi:wifi firmware version: 6e1f9e4
I (1255) wifi:config NVS flash: disabled
I (1265) wifi:config nano formating: disabled
I (1265) wifi:Init data frame dynamic rx buffer num: 32
I (1275) wifi:Init management frame dynamic rx buffer num: 32
I (1275) wifi:Init management short buffer num: 32
I (1285) wifi:Init dynamic tx buffer num: 32
I (1285) wifi:Init static rx buffer size: 1600
I (1295) wifi:Init static rx buffer num: 10
I (1295) wifi:Init dynamic rx buffer num: 32
I (1305) wifi:mode : null
I (1305) NETWORK_ADAPTER: Initial set up done
```

Have you encountered similar issue?

RathiSonika, 2021-11-08:
Hello @rforro 

Please verify if ESP32 EN pin is properly connected to RPi pin no. 31 or not. 
[image]

Please retry with proper reset pin connection.

mantriyogesh, 2021-12-02:
Hello @rforro,

Apologies for delayed response.

Is the problem resolved at your end?
If not, Is it possible to test the basic SPI working with some dummy data using any user space library like spidev, wiringPi etc?
If the user space library is running smooth, we can rule out SPI hardware/pin connection issues.

rforro, 2021-12-02:
@mantriyogesh I've tested SPI independently, it was working. After that I switched back to SDIO, which isn't without flaws as well. I cannot stretch this enough, but anybody who is replicating this solution it doesn't matter if SPI or SDIO (UART works anytime) has to be very patient. For example the latest raspbian with kernel 5.something cannot be used, I had to use kernel 4.19.97. Another thing is SDIO interface, because of its high frequencies it is very prone to interference (probably different cable length as well).

To sum it up for anybody who is struggling, be patient, try different kernel versions, check Pull-ups, etc.

closed 2021-12-02 by rforro

mantriyogesh, 2021-12-02:
Thanks for sharing your observations.
We keep the latest kernel supported ([which supported by raspberry PI OS](https://en.wikipedia.org/wiki/Raspberry_Pi_OS)).
As if now we have tested till 5.10.17, which tested with complete regression suite.

Regarding signal integrity of SDIO, Yes! Your observation is correct, SDIO protocol **by nature** is prone to higher wire lengths interference.

This is the reason, we suggest in [documentation](https://github.com/espressif/esp-hosted/blob/master/docs/Linux_based_host/Linux_based_readme.md#11-hardware-requirements) to use small wires for testing of SDIO.

In general when the demo works, customers prefer to use PCB to connect SDIO, which does not face any such signal integrity issues. Again, that is something not mandatory but better way of using SDIO.

Please do not hesitate to raise issues if you face any, we are open to bugs and try to resolve as much as fast possible.