espressif/esp-hosted #75: Memory leaks in host commands

drorgl, 2021-11-14:
Hi,

I've encountered memory leaks in [commands.c](https://github.com/espressif/esp-hosted/blob/master/host/host_common/commands.c) when every call to `esp_hosted_config_payload__unpack` and `resp` doesn't get free, I've used `protobuf_c_message_free_unpacked(resp, NULL);` to solve the problem locally.

problem example: [commands.c - esp_hosted_config_payload__unpack](https://github.com/espressif/esp-hosted/blob/84b0372c390efe6352d77db70711fee2a37d2aac/host/host_common/commands.c#L98)

Does the memory get free somewhere else and I'm using the library wrong?

ajita02, 2021-11-16:
Hi @drorgl ,
Thanks for reporting problem. We will look into it.

Ajita

ajita02, 2021-11-17:
Hi @drorgl ,
Memory leak in [comamnds.c](https://github.com/espressif/esp-hosted/blob/master/host/host_common/commands.c) got fixed in latest master.
Can you please close this issue?

Thanks,
Ajita

drorgl, 2021-11-17:
I think 2 are missing:
[474](https://github.com/espressif/esp-hosted/blob/b054add2f2c00a8b1799316719724e3e014dffd3/host/host_common/commands.c#L474)

[480](https://github.com/espressif/esp-hosted/blob/b054add2f2c00a8b1799316719724e3e014dffd3/host/host_common/commands.c#L480)

ajita02, 2021-11-18:
Hi @drorgl.
Its updated in latest master.  Can you please close this issue?

Thanks,
Ajita

closed 2021-11-18 by drorgl