espressif/esp-hosted #76: Increase OTA firwmware size

cliffordjefferies, 2021-11-18:
Hi all and thanks for your work on this project

The recent OTA addition specifies - Currently, new OTA image supported size is upto 1MB.
I'm trying to enhance esp-hosted with some custom functionality and I'm allowing for partitions up to 0x130000

I'm happy to do the work and generate a pull request if you can give some pointers as to where the 1MB limit is enforced. I'm seeing the following but it not clear how it relates.

# Please do not use chunk size bigger than 4000.
chunk_size = 4000

if (chunk_size>4000):
    chunk_size = 4000

Regards
Cliff

mantriyogesh, 2021-11-18:
Hello @cliffordjefferies 

I would take opportunity to explain chunk in OTA.
Consider binary of say 1MB.
While performing OTA, It is generally broken down into smaller pieces and every piece is called as 'chunk'.
Every chunk is written sequentially into flash till complete binary image is flashed.
You can relate this to downloading big binary into small parts and then collate into big binary.
In python, this exact operation is supported, i.e. download a chunk and burn on flash, then go to next chunk, so on.

In C, currently HTTP client is not embedded, so downloading chunk by chunk from HTTP server is not supported and it is left to user. As a difference, In C, just splits pre-downloaded binary (this is also referred as chunk, but not streamed from HTTP server), and burns similar way to flash sequentially till complete binary image is flashed.

In short, This chunk size (max of 4000) does not limit the maximum size of binary.

With respect to bigger binary size (>1MB), there is a way to do this, which @ajita02 will explain shortly.

ajita02, 2021-11-18:
Hi @cliffordjefferies,

For custom partition table, change partition table option as `menuconfig->Partition Table->Partition Table->Custom partition table CSV`. Create custom partition table according to binary size using [this link](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html#creating-custom-tables). Place newly created `partitions.csv` in `esp/esp_driver/network_adapter` folder. Followed by build, flash on ESP32 module.

Thanks,
Ajita

cliffordjefferies, 2021-11-18:
Thanks for the responses @ajita02 and @mantriyogesh.
I'm already using a custom partition table with suitable sizes so it appears I have nothing further to do.
May I suggest the comment relating to the 1MB limit be reworded to say the limit is set by the partition sizes.
Thanks again
Cliff

closed 2021-11-18 by cliffordjefferies

mantriyogesh, 2021-11-19:
@cliffordjefferies 
Sure, we will update the documentation.