espressif/esp-hosted #77: Why run RPI_ Init.sh error

zhichunlee, 2021-11-22:
Hi，
I used raspberry pi 3B+ and esp32 development board (esp32 devkitc V4),
following the ESP hosted instructions in GitHub:
(1) Complete hardware wiring (refer to 1.1 hardware requirements)
(2) Complete the software update and clone ESP hosted project of raspberry pi (refer to 1.2 host setup)
(3) Modify /boot/config.txt of raspberry pi and restart the system.
(4) However, an error occurred while performing the following operations.
[image]
[image]
What is the reason for this? How to deal with it?

ajita02, 2021-11-22:
Hi @zhichunlee ,

You have not installed linux kernel headers. Please follow this [link](https://github.com/espressif/esp-hosted/blob/master/docs/Linux_based_host/Linux_based_readme.md#12-host-setup).

Thanks,
Ajita

zhichunlee, 2021-11-22:
Hi,Ajita
Thank you for your reply！
According to your reminder, confirmed Linux Kernel Headers and tools again.
It seems that they are all the latest versions, but the RPI_ Init.sh error message still appears.
Please see as follows:
[image]
Where does the module 'esp32_spi.ko' come from, Download or compile?

Thanks!

ajita02, 2021-11-22:
Hi @zhichunlee,

'esp32_spi.ko' is gets compiled using linux kernel headers. Here the problem is `*** /lib/modules/5.10.17-v7+/build: No such file or directory`.  There is no `build` folder under kernel directory `5.10.17-v7+`. Please check `uname -a` output to confirm kernel version.

Thanks,
Ajita

zhichunlee, 2021-11-22:
Hi,Ajita
Thank you very much for your guidance.
The problem has been solved by updating the raspberry-pi kernel version.

Thinks!

ajita02, 2021-11-22:
Hi @zhichunlee,

Glad to help you. Can you please close this issue?

Thanks.
Ajita

closed 2021-11-23 by zhichunlee