espressif/esp-hosted #78: Why ESP-HOSTED SPI not work

zhichunlee, 2021-11-23:
Hi
I used RPI 3B+ and ESP32-DevKitC V4, followed 'Wi-Fi and BT/BLE connectivity Setup over SPI',
tried running ESP hosted SPI. The ESP firmware binaries downloaded from pre-built releases site.
esp_hosted_firmware_esp32_spi_v0.3 is used.
I have successfully completed all the settings, got /dev/esps0, but the final checking the setup for
SPI is different from the description information. In addition, the next ap_scan_list.py did not succeed.
[image]

[image]

[image]

[image]
Please help analyze the reasons and give some suggestions, Thanks!

mantriyogesh, 2021-11-23:
Hello @zhichunlee 

Thanks for all detailed logs while issue creation.

Actually, the SPI connection and SPI documentation is updated since release 0.3.
The commit where the release 0.3 points is https://github.com/espressif/esp-hosted/tree/d30942e7d863a78c1ea2d13c0f964529ff542fef

Can you please check at https://github.com/espressif/esp-hosted/blob/d30942e7d863a78c1ea2d13c0f964529ff542fef/docs/Linux_based_host/SPI_setup.md#11-hardware-setup and confirm the SPI setup is as per old connections?

~ Yogesh

zhichunlee, 2021-11-23:
Hi Yogesh
Thank you for your reply!
According to the new link you provided, re-connected the hardware and prepared the software.
[image]
But But the problem remains that the ap scan cannot be completed and there is no HCI interface.
[image]

Thanks!

mantriyogesh, 2021-11-23:
Can you please confirm:
1.  /boot/config.txt is correctly populated for SPI (from the documentation)
2. The kernel module is loaded after `./rpi_init.sh spi` correctly?
2. Are the connections firm ?

Actually the first transaction from ESP32 to host is not happening.
This seems more of setup problem.

mantriyogesh, 2021-12-01:
Does the dmesg show any difference?

mantriyogesh, 2021-12-02:
https://github.com/espressif/esp-hosted/blob/master/docs/Linux_based_host/SPI_setup.md#3-checking-the-setup-for-spi
The SPI log in dmesg should show 'INIT' event.

mantriyogesh, 2021-12-02:
Also in case you still face spi connection issue,
It will be better to check with spidev, a userspace linux spi driver, to verify if the basic spi hardware and connections are working fine.

mantriyogesh, 2022-09-16:
Hello @zhichunlee 
Did you get it working?