espressif/esp-hosted #79: Can't generate protobuf files - compile errors

cliffordjefferies, 2021-11-23:
When I try and generate the protobuf files with **protoc-c esp_hosted_config.proto --c_out=.** the outputs are very different and will not compile. Protoc-c has just been pulled and built
The main issue is that members like **has_mode** are not generated yet these are referenced throughout the esp-hosted source giving errors like
```
home/cliffordj/src/esp-hosted/esp/esp_driver/network_adapter/main/slave_commands.c: In function 'data_transfer_handler':
/home/cliffordj/src/esp-hosted/esp/esp_driver/network_adapter/main/slave_commands.c:1934:9: error: 'EspHostedConfigPayload' {aka 'struct _EspHostedConfigPayload'} has no member named 'has_msg'
     resp.has_msg = true;

```
The start of the diff is
```
--- a/common/include/esp_hosted_config.pb-c.h
+++ b/common/include/esp_hosted_config.pb-c.h
@@ -8,9 +8,9 @@
 
 PROTOBUF_C__BEGIN_DECLS
 
-#if PROTOBUF_C_VERSION_NUMBER < 1000000
+#if PROTOBUF_C_VERSION_NUMBER < 1003000
 -# error This file was generated by a newer version of protoc-c which is incompatible with your libprotobuf-c headers. Please update your headers.
-#elif 1002001 < PROTOBUF_C_MIN_COMPILER_VERSION
+#elif 1003003 < PROTOBUF_C_MIN_COMPILER_VERSION
 -# error This file was generated by an older version of protoc-c which is incompatible with your libprotobuf-c headers. Please regenerate this file with a newer version of protoc-c.
 #endif
 
@@ -138,25 +138,22 @@ typedef enum _EspHostedConfigMsgType {
 struct  _EspHostedCmdGetMacAddress
 {
   ProtobufCMessage base;
-  protobuf_c_boolean has_mode;
   int32_t mode;
 };

```

mantriyogesh, 2021-11-23:
Hello @cliffordjefferies 

We will check into this issue.

Could you please confirm the
a. OS distribution version
b. protobuf version
that you are using?

ajita02, 2021-11-23:
Hi @cliffordjefferies ,

We are using protoc-c 1.2.1 version. It seems protoc-c is recently updated. We will support latest version shortly.

```
---protoc-c --version
protobuf-c 1.2.1
libprotoc 3.0.0
```

Thanks,
Ajita

cliffordjefferies, 2021-11-23:
The current pull of that repo gives
```
protoc-c --version
protobuf-c 1.3.3
libprotoc 3.8.0
```

cliffordjefferies, 2021-11-24:
I wasted a couple of hours trying to match your version but was unsuccessful. After getting it back together I now have 
```
protoc-c --version
protobuf-c 1.4.0
libprotoc 3.19.1

```

ajita02, 2021-11-24:
hi @cliffordjefferies,

We will update latest protobuf changes soon.

Thanks,
Ajita