espressif/esp-hosted #80: "Failed to process rx data" error

sinanaltinsoy, 2021-11-23:
Hi,

I’m getting below error when I tried esp-hosted sdio with ESP-WROOM-32 and Raspberry Pi 4,

[image]

1. /boot/config.txt is correctly populated for SDIO
2. The kernel module is loaded correctly after “./rpi_init.sh sdio” 
3. Pinout connections are okay.

I checked SD Pull-up Requirements and I tried the below combinations and the result is always the same

[image]

1. /boot/config.txt is correctly populated for SDIO (from the documentation)
2. The kernel module is loaded correctly after “./rpi_init.sh sdio” 
3. Pinout connections are okay.

Could you please help me?
Thanks in advance

kian99, 2021-11-23:
I experienced a similar problem with a Rasperry Pi Compute Module 3+ and the ESP-WROOM-32. (See the first post in the link below).

The solution was to change the drive strength of the SDIO pins. I was able to do this via adding a custom device tree overlay.

You can see my issue and solution here - https://github.com/espressif/esp-hosted/issues/26#issuecomment-819344162

Hope it helps.

sinanaltinsoy, 2021-11-24:
Hi kian99, thanks for the comment. I tried your solution but it is not working for me. I'm still getting the same error. I'm new about raspberry pi, So maybe I missed something. So, I'm sharing my steps. 

I'm using Raspberry Pi 4 B and I edited the bcm2711-rpi-4-b.dtb file under the /boot folder as below. 

```
/dts-v1/;

/memreserve/	0x0000000000000000 0x0000000000001000;
/ {
	compatible = "raspberrypi,4-model-b\0brcm,bcm2711";
	model = "Raspberry Pi 4 Model B";
	#address-cells = <0x02>;
	#size-cells = <0x01>;
	interrupt-parent = <0x01>;

	aliases {
		serial0 = "/soc/serial@7e215040";
		serial1 = "/soc/serial@7e201000";
		emmc2bus = "/emmc2bus";
		ethernet0 = "/scb/ethernet@7d580000";
		pcie0 = "/scb/pcie@7d500000";
		blconfig = "/reserved-memory/nvram@0";
		audio = "/soc/mailbox@7e00b840/bcm2835_audio";
		aux = "/soc/aux@7e215000";
		sound = "/soc/sound";
		soc = "/soc";
		dma = "/soc/dma@7e007000";
		watchdog = "/soc/watchdog@7e100000";
		random = "/soc/rng@7e104000";
		mailbox = "/soc/mailbox@7e00b880";
		gpio = "/soc/gpio@7e200000";
		uart0 = "/soc/serial@7e201000";
		uart1 = "/soc/serial@7e215040";
		sdhost = "/soc/mmc@7e202000";
		mmc = "/soc/mmc@7e300000";
		mmc1 = "/soc/mmcnr@7e300000";
		mmc0 = "/emmc2bus/mmc@7e340000";
		i2s = "/soc/i2s@7e203000";
		i2c0 = "/soc/i2c0mux/i2c@0";
		i2c1 = "/soc/i2c@7e804000";
		i2c10 = "/soc/i2c0mux/i2c@1";
		spi0 = "/soc/spi@7e204000";
		spi1 = "/soc/spi@7e215080";
		spi2 = "/soc/spi@7e2150c0";
		usb = "/soc/usb@7e980000";
		leds = "/leds";
		fb = "/soc/fb";
		thermal = "/soc/avs-monitor@7d5d2000/thermal";
		axiperf = "/soc/axiperf";
		mmc2 = "/soc/mmc@7e202000";
		i2c3 = "/soc/i2c@7e205600";
		i2c4 = "/soc/i2c@7e205800";
		i2c5 = "/soc/i2c@7e205a00";
		i2c6 = "/soc/i2c@7e205c00";
		i2c20 = "/soc/i2c@7ef04500";
		i2c21 = "/soc/i2c@7ef09500";
		spi3 = "/soc/spi@7e204600";
		spi4 = "/soc/spi@7e204800";
		spi5 = "/soc/spi@7e204a00";
		spi6 = "/soc/spi@7e204c00";
	};

	chosen {
		bootargs = "coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1";
	};

	reserved-memory {
		#address-cells = <0x02>;
		#size-cells = <0x01>;
		ranges;
[... 5172 lines cut ...]
		regulator-min-microvolt = <0x1b7740>;
		regulator-max-microvolt = <0x325aa0>;
		regulator-always-on;
		regulator-name = "vdd-sd-io";
	};

	fixedregulator_5v0 {
		compatible = "regulator-fixed";
		phandle = <0xde>;
		regulator-min-microvolt = <0x4c4b40>;
		regulator-max-microvolt = <0x4c4b40>;
		regulator-always-on;
		regulator-name = "5v0";
	};

	axi {

		vc_mem {
			reg = <0x3ec00000 0x40000000 0xc0000000>;
		};
	};
};
```

If I did everything correct, what is the problem? Could someone help me?

MangeshMalusare, 2021-11-25:
Hi @sinanaltinsoy, Can you please help share pull up specification? Can you please confirm if following requirement is fulfilled?

> Attach external pull-ups by connecting each pin which requires a pull-up to VDD via a 10 kOhm resistor
Reference: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html#no-pull-ups

Also please confirm how you are connecting the GPIO pins? If you are using dupont wires, please make sure that those are 10cm or less in length.

Thanks,
Mangesh

kian99, 2021-11-25:
@sinanaltinsoy, Check your /boot/config.txt file as well.

You mentioned that you added the below lines,
```
#dtoverlay=sdio,poll_once=off,gpios_22_27
#dtoverlay=disable-bt
```

The # symbol should be remove, it indicates comments. So those overlays are not being enabled.

sinanaltinsoy, 2021-11-25:
> @sinanaltinsoy, Check your /boot/config.txt file as well.
> 
> You mentioned that you added the below lines,
> 
> ```
> #dtoverlay=sdio,poll_once=off,gpios_22_27
> #dtoverlay=disable-bt
> ```
> 
> The # symbol should be removed, it indicates comments. So those overlays are not being enabled.

In the original file there is no # symbol, I forget to delete it in the post. Sorry for the conflict.

mantriyogesh, 2021-12-02:
Hello @sinanaltinsoy,

Were you able to manage to get it working or still face issues?

sinanaltinsoy, 2021-12-02:
Hi @mantriyogesh,

I couldn't solve my issue, I'm waiting your help.

mantriyogesh, 2021-12-02:
We will have to make similar environment to yours to get the config feed using explicit device tree.
But this might take unusual time.

Is it possible to just mimic the steps using  installing normal raspberry Pi OS and the ESP-Hosted documentation?

we have even compiled the linux device trees for multiple Linux variants like yocto etc. The problem is, those all setups will not be exactly same, as SoC will have their device driver on top of Linux, which varies. And to debug the device tree issue, it generally takes some time even when you have hardware with you.

So, in summary, if feasible, can you please try to use raspberry pi OS directly as you are already on Raspberry Pi hardware only and just follow steps line by line, instead of customised way of device tree on Raspberry Pi?

This is easier as far as Raspberry Pi hardware is being used. This way we will be on same page and it will be easy to debug the issue.

If that is not at all possible, and I were in your place, I would have done:
1. Only enable selective sdio hardware in device tree
2. Rest of the hardware, at least which we are not using, should be removed/disabled from the device tree
3. SDIO could be enabled through 2 different device drivers (AFAIK), pure sdio driver and sdhci. In raspberry pi OS (default OS, default settings, without any Device tree addition/modification), we use sdhci.
4. Do not ignore any warnings while compiling dts. Why? -> dts compilation just checks syntax. Compilation does not detect any semantic errors.
5. Basically SDIO hardware should get to probe this way.
6. Now load the kernel driver and verify if the driver is working or not.

mantriyogesh, 2022-07-13:
@sinanaltinsoy 

Does the solution worked for you?
Overall, Raspberry Pi should work with ESP32 as per documentation.
Please make sure if you have smaller jumper wires (all equal and <6cm) and meet the pull up requirements. Pull up requirements depend upon the ESP32 you use: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html