espressif/esp-hosted #84: Is there no way to use UART2 for flashing ESP32? or use ESP32 without flashing ?

tushar-uwaterloo, 2021-12-12:
I am using a PMOD ESP32 : https://digilent.com/reference/pmod/pmodesp32/reference-manual
I have connected it to the Go board FPGA : https://www.nandland.com/goboard/introduction.html

My setup is like this: 
Host computer ----- (USB)----> FPGA (has USB to UART converter) ---  PMOD --- > ESP32 

The problem is, FPGA connects to PMOD ESP32 on UART2 port of ESP32 which cannot be used for flashing images. Is there no way that I can flash the ESP32 using the setup I have?

The only other option I can think of is to have another USB-UART converter and use that to flash ESP32 on UART0 port. OR is there a way that I can use ESP32 without flashing?

cyildirim23, 2022-03-21:
Hey, I'm currently trying to use this pmod with a project and I noticed the same thing. No USB port to program the board. Same setup. I see 4 pins on top, rx, tx, gnd, IO0. Assuming it has to be done with these pins?

mantriyogesh, 2022-03-22:
Hello @tushar-uwaterloo 
This looks like some general flashing requirement. Please close and raise this issue in ESP-IDF for this.

Anyway, one way of doing it is [ESP Serial Flasher - Raspberry Pi example](https://github.com/mantriyogesh/esp-serial-flasher/tree/master/examples/raspberry_example)
Here Raspberry Pi connects to [above ESP pins](https://github.com/mantriyogesh/esp-serial-flasher/tree/master/examples/raspberry_example#hardware-connection) and does flashing.

In case of flashing ESP-Hosted, Easy flashing could be done using pre-built binaries from release using [these](https://github.com/espressif/esp-hosted/blob/master/docs/Linux_based_host/SDIO_setup.md#221-load-pre-built-release-binaries) steps