espressif/esp-hosted #87: Is there any special limitation in ESP32 SPI Slave?

TreeNewbeer, 2021-12-16:
A few month ago I ported ESP-Hosted(commit: `0f7ce11`) MCU host driver on one MCU communicate with ESP32 and everything goes well. Recent days I updated my ESP-Hosted to latest master branch(commit: `07beeb9`) and found that the host MCU could not detected the `INIT_EVENT` from ESP32, the reason is ESP32 always send `0xff 0x0 0x0 0x0 ...` to host MCU after startup. And when I replaced the Slave MCU as ESP32-C3, everything goes well again.

In MCU host driver I found there some "hard delay"  inside `reset_slave()` function, and in the latest master branch the `sleep(1)` 
(before generate init event) appeared on Slave driver, so I am a little curious that is there any special limitation in ESP32 SPI Slave?

mantriyogesh, 2021-12-16:
Hello @TreeNewbeer,

We much appreciate for evaluating the project since start.

We will look into code for issue you faced and get back to you shortly.

Whitehouse112, 2022-08-09:
Hi everyone,
I read the issue posted by @TreeNewbeer and I would like to know is a solution has been found. I'm stuck at the same point: i'm working with an esp32-wroom-32e connected through SPI to a MCU (stm32u575) but i'm unable to get the `INIT_EVENT` after esp32 startup. I receive only `0xff 0x0 0x0 0x0...`. I'm working with esp-hosted (branch `ESP-Hosted_MCU_Host`, commit: `f209c91`) on MCU, while `esp_hosted_firmware_esp32_spi_v04.bin` has been flashed with esptool.py over ESP32.

Thanks

mantriyogesh, 2022-08-11:
As a matter of fact, ESP32 has limitation as per mentioned in
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_slave.html#restrictions-and-known-issues.

Any other chipsets like ESP32-C3, ESP32-S2, ESP32-S3 do not have these issues.

This is why ESP32 has differences in IOC files and code as well.

IOC:
'v1' relates to ESP32.
(https://github.com/espressif/esp-hosted/tree/ESP-Hosted_MCU_Host/host/stm32/proj)

code:
https://github.com/espressif/esp-hosted/blob/ESP-Hosted_MCU_Host/host/stm32/driver/spi/spi_drv.c#L45-L54

mantriyogesh, 2022-08-16:
As workaround for now, Issue for @Whitehouse112 is fixed under https://github.com/espressif/esp-hosted/issues/142 with release v0.3 firmware and STM code.
@TreeNewbeer Can you please try if that works for you?