espressif/esp-hosted #88: Performance decrease since 4848e87940567d97e51399056d5834b3e174548d

starkeeper82, 2021-12-21:
Hi,
recently I updated my ESP firmware to get the latest head changes. After fiddling around a bit I noticed a major performance drop compared to the old firmware I used.
I then crawled back in the commit history and found that with the changes made in this commit, the speed drecreased by 70%.
4848e87940567d97e51399056d5834b3e174548d

Is there anyone who can reproduce my results?

mantriyogesh, 2021-12-21:
Hello @starkeeper82,

Can you please let us know the transport (SPI/SDIO), ESP chipset (ESP32/ESP32S2/ESP32C3), kernel for both tests and data rate throughput at both tests?
With this info, we will address this issue quickly.

mantriyogesh, 2021-12-23:
Also, Can you please give us comparison in figures for this?

starkeeper82, 2021-12-23:
@mantriyogesh 
My normal Tests using iperf2 showed a solid 4,5MBit/s transfer rate. Using software after the mentioned commit, the data rate is in the 1 MBit/s area.

alireza3137, 2022-01-09:
@starkeeper82 
Hi
I know my question is not related to your issue. I was wondering if you port LWIP with MCU and esp_hosted with SPI protocol. 
If yes what is your data rate in TCP communication?

starkeeper82, 2022-01-11:
> Hi I know my question is not related to your issue. I was wondering if you port LWIP with MCU and esp_hosted with SPI protocol. If yes what is your data rate in TCP communication?

Just right above your question I have posted some numbers..

mantriyogesh, 2022-06-24:
Hello @starkeeper82

Apologize for late response.

We also observed similar issue with C3 with Linux.
We had addressed some issues in the latest master, queue size increase, meta queue over priority queue (so that tasks would not be polling) to address issues in this area.

Some STM32 specific options helped us to bump throughput as well.
Options shown in STM32CubeIDE as below. Other tools using STM32 can change as per their configurator.
1. FIFO word setting
[image]

2. cache enabling
[image]

Please note that above discussed both settings are not generic and might be available to some or latest STM32 platforms or SDKs.